ckb-next/ckb-next #167: Different macros for key press, hold and release.

Sur3, 2018-05-03:
Feature-request:
It would be nice to have more complex macros functions, like executing an macros once when pressing a key, then repeating a different macro continuesly while the button is hold and execute another different macro when the key is released.

My ckb-next version 0.2.9.

ZirconiumX, 2018-05-03:
First off, we did recently release 0.3.0, so consider upgrading.

For the macros themselves, it wouldn't be too hard to trigger an event based on keyup (i.e. key release). The key being held would be harder to detect for us, so that would be further off.

Sur3, 2018-05-03:
Well being held is just the time between press and release, if press and release is detected the hold is just between that, meaning the press is macro 1 and if the key is not released play macro 2 repeated until it is released, then play macro 3.
This could also be written in a single text field using some separator for the three individual parts, e.g. a semicolon with timings of the individual parts beginning at zero time each, i.e.:
```
+mouse4=0, -mouse4=10000;  //press: switch weapon
+mouse1=10000, -mouse1=20000;  //hold: shoot continuously
+mouse4=5000, -mouse4=10000  //release: switch weapon back

```
Yes 0.3.0 is not yet in the repository of my distro.

ZirconiumX, 2018-05-03:
>Well being held is just the time between press and release,
And measuring that time is the difficult bit, because what if somebody assigns a macro to every single key of their keyboard? Then we have to track how long 100+ keys have been held down for while still feeling fast and fluid (which means we need to check often). Assuming we poll every millisecond, that's going to be noticeable CPU usage spikes for macro keys, in addition to the overhead of using Qt to render our animations.

Sur3, 2018-05-03:
> how long 100+ keys have been held

I've never seen someone with 100 fingers, I suppose the maximum number of keys pressed simultaneously would be limited by ten fingers.. :smile:

ZirconiumX, 2018-05-03:
In contrast, we get an event for the key being pressed and released, so we can simply call the macro code when we receive that event.

And before you ask, no, we're not going to add an event for keypress, because it moves the CPU overhead to the daemon, and also causes syscall overhead, which is already crippling Skylake (see #97).

tatokis, 2018-05-03:
It should in theory be possible to run without measuring the time, until it gets interrupted by the keyup event.
Something like spinning up a thread (like we already do for macros anyway), having it loop endlessly until it gets destroyed

ZirconiumX, 2018-05-03:
>I've never seen someone with 100 fingers, I suppose the maximum number of keys pressed simultaneously would be limited by ten fingers..

Like it or not, it *is* a situation we have to deal with.

>Something like spinning up a thread (like we already do for macros anyway), having it loop endlessly until it gets destroyed

This would create more CPU overhead.

Sur3, 2018-05-03:
Well if macros use a special thread anyway I suppose there would be barely any overhead, because the thread creation is the major time consumption because of syscalls, the running of a thread pressing keys should not be really CPU intensive in times of ryzen..

> Assuming we poll every millisecond,

What do you wanna poll? I suppose a macro thread would use sleep events and not polling, it is just the key press events, then an infinite loop of hold events if existent and on thread destruction the key release events can be called. And between the key events are some sleep calls depending on the timings given.

tatokis, 2018-05-03:
@Sur3 Yeah, that is what I was thinking. A thread already starts in order to execute macros, so it should be a matter of it not exiting once the macro is done.
https://github.com/ckb-next/ckb-next/blob/master/src/daemon/input.c#L155

OvermindDL1, 2018-05-03:
I'd say don't have something for held down repeating, just have keyup/down.  With just keyup/down you can easily have a script that starts doing something repeated on keydown and another script that kills it on keyup, like here is one I use (not the most efficient at all, but it does the job, I should probably set a global mutex instead or a named pipe or something):
```sh
#!/bin/sh

t="$1"
s="$2"
f="$3"
b="${4-1}"

[ ! -e "$f" ] && touch "$f"

while [ -e "$f" ] && [ "$t" -ge 0 ]; do
        t="$(($t-1))"
        xdotool click "$b"
        sleep "$s"
done
```
And to kill it I just delete the filepath I passed in on keyup.  All this stuff is easy to do in userland.

Sur3, 2018-05-03:
But a shellscript is pretty inefficient, every xdotool click will spawn a subprocess, handling that in a single thread would be much more efficient, for my example code
```
+mouse4=0, -mouse4=10000;  //press: switch weapon
+mouse1=10000, -mouse1=20000;  //hold: shoot continuously
+mouse4=5000, -mouse4=15000  //release: switch weapon back
```

This could look like this in pseudocode:

```
macrothread(){
char release_not_pressed=true
signal_handler_on_signal_set_variable_to_false(&release_not_pressed)

mouse(4,1); sleep(10000); mouse(4,0);  //press: switch weapon

while(release_not_pressed){
 sleep(10000); mouse(1,1); sleep(10000); mouse(1,0);  //hold: shoot continuously
}

sleep(5000); mouse(4,1); sleep(10000); mouse(4,0);  //release: switch weapon back
}
```

Of course the sleep and mouse events would have to be parsed instead of directly calling them.

OvermindDL1, 2018-05-03:
> But a shellscript is pretty inefficient, every xdotool click will spawn a subprocess, handling that in a single thread would be much more efficient, for my example code

It does, but as stated it wasn't made to be efficient (it's usually a 1-click-a-second kind of thing), otherwise I'd whip something up in C++ or so, which would be perfectly efficient with only a single process then (and all of what, like 8 lines of code...).

Sur3, 2018-05-04:
Yeah but I want to use that for fps, so I need fast reaction, shell script spawning is therefore out of the question.. :smiley:

OvermindDL1, 2018-05-04:
Well there are native programs already built (or can build your own, they are super easy), or just have it send a signal via a named pipe to a constantly running process (would be hard to be faster than that), and you can do anything and everything that you'd want then.  :-)

I've been tempted to build out a full macro'ing system like that someday (like take idle clicker games, I find them fun not because of the game but due to finding the ways to automate them, and I've automated some exceptionally difficult to automate ones ^.^).  :-)

But been lazy, still just using shell scripts, spawning xdotool takes milliseconds anyway so it's latency has not been an issue as of yet, I can average 125 calls per second of it.

dibstern, 2018-05-10:
I've wanted this so I can map CMD+CTRL+SHIFT+OPTION to my CAPS key (you can't currently rebind a key to multiple keys). I used to do it with Karabiner-Elements, but it can't connect to my keyboard -> from Karabiner's log:
`[2018-05-10 14:17:30.455] [error] [grabber] IOHIDDeviceOpen error: kIOReturnExclusiveAccess (-536870203) Corsair Gaming K95 RGB PLATINUM Keyboard (device_id:2)`

I tried different delays on key-up and key-down in the Macro binding, as suggested above, but it doesn't work - no idea why.

ZirconiumX, 2018-05-10:
That's because ckb-next conflicts with Karabiner; they both try to get exclusive access to the device.

dibstern, 2018-05-10:
Yep, I assumed as much, still can't remap my caps lock to those key presses though. Would be ace if we could map keydowns and keyups, as discussed above.

embedisbae, 2020-03-15:
Is this a thing yet, or being worked on? Would be great to have this for Minecraft AFK fishing.

ZirconiumX, 2020-03-15:
No, but code contributions are always welcome.