ckb-next/ckb-next #168: Long shutdown/reboot time

eproxy, 2018-05-03:
Hi,

I have K65 rgb keyboard and I'm using ckb-next-git AUR package on Arch. Most of the time everything  works fine but recently I noticed shutdown/reboot taking much more time than usually  ~30s with systemd just waiting for ckb-next-daemon to exit:

```
00:38:29  systemd-logind[425]: System is powering down.
...
00:38:31  systemd[1]: Stopped Login Service.
00:39:06  ckb-next-daemon[424]: [I] Removed device path /dev/input/ckb1
00:39:06  kernel: usbhid 1-4:1.1: can't add hid device: -110
00:39:06  kernel: usbhid: probe of 1-4:1.1 failed with error -110
00:39:06  ckb-next-daemon[424]: [I] Closing root controller
00:39:06  ckb-next-daemon[424]: [I] Removed device path /dev/input/ckb0
00:39:06  systemd[1]: Stopped Corsair Keyboards and Mice Daemon.
00:39:06  systemd[1]: Stopped target Basic System.
00:39:06  systemd[1]: Stopped target Slices.
...
00:39:06  systemd-shutdown[1]: Sending SIGTERM to remaining processes...
00:39:06  systemd-journald[234]: Journal stopped
```

Any hints are much appreciated.

tatokis, 2018-05-03:
Can you bisect this? Can you also reproduce this if you run the daemon from a terminal manually and then send a SIGTERM? `sudo killall -15 ckb-next-daemon`

ZirconiumX, 2018-05-03:
This is a known issue, with a few possible causes which are hard to fix.

The main reason is that ckb-next-daemon is trying to talk to the keyboard
to switch it back to software mode so that usbhid can talk to it, but the
keyboard isn't replying.

It could be that the keyboard has hung, which gets reset by a power cycle.

It could equally be that the kernel is having problems talking to it.

eproxy, 2018-05-03:
@tatokis 

> Can you also reproduce this if you run the daemon from a terminal manually and then send a SIGTERM? sudo killall -15 ckb-next-daemon

Yes I'm able to reproduce it:
```
ckb-next: Corsair RGB driver 0.2.9-185-g8531096
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair Gaming K65 RGB RAPIDFIRE Keyboar at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[I] Setup finished for /dev/input/ckb1

[I] Caught signal 15
[I] Disconnecting /dev/input/ckb1
[I] Removed device path /dev/input/ckb1
[I] Closing root controller
[I] Removed device path /dev/input/ckb0
```
It took 36s between `[I] Disconnecting /dev/input/ckb1` and `[I] Removed device path /dev/input/ckb1`

eproxy, 2018-05-03:
Sorry the package was slightly oudated, now I'm running 7229ea4 but the issue remains.
[journal.log](https://github.com/ckb-next/ckb-next/files/1970527/journal.log)

tatokis, 2018-05-04:
Hmmm.
Get the latest code, 
`cmake .. -DCMAKE_BUILD_TYPE=Debug && make -j4 daemon && sudo gdb ./bin/ckb-next-daemon`
Enter `r` in the gdb prompt.

You'll need another keyboard for this, but `sudo killall ckb-next-daemon`, enter `c` in the gdb prompt then wait a bit until it hangs.
Once it does, Ctrl C the gdb prompt, `thread apply all bt full` and send the output.

eproxy, 2018-05-05:
Please find attached although I'm not sure if I have the right logs, ctrl-c didn't seem to be working. 
[image]
[1.log](https://github.com/ckb-next/ckb-next/files/1976653/1.log)
[2.log](https://github.com/ckb-next/ckb-next/files/1976654/2.log)
[3.log](https://github.com/ckb-next/ckb-next/files/1976655/3.log)

eproxy, 2018-05-09:
I cloned your repository on Saturday around 13 CET so that was already after your commits. I can see the changes you made in the source file but unfortunately it doesnt change anythin. Funny observation, every ~10-15th reboot is successful (without slowdown) :)

eproxy, 2018-05-12:
It seems to be working after applying the patch :+1:  All next 4 restarts were instant (ssd), will keep an eye on it for a while.

eproxy, 2018-05-20:
Just wanted to update that since last weekend everything has been working fine :)

tatokis, 2018-05-20:
Would you mind dropping by on IRC so that I can ask you to try a few other things?

eproxy, 2018-05-20:
Hi, sorry cant join today, perhaps tomorrow?

ZirconiumX, 2018-05-21:
Today is fine for me at least.

tatokis, 2018-05-21:
Worked around with https://github.com/ckb-next/ckb-next/commit/30796c05072a821c07064fe6ed2956216bb74921

closed 2018-05-21 by tatokis

ashtonx, 2018-06-15:
Affect k95 rgb as well, eu version at least, takes 36 seconds every time to shut down.

Affects 0.3.0-1 as well as git

```
cze 15 08:29:39 Uroboros ckb-next-daemon[1568]: [I] Caught signal 15
--
cze 15 08:29:39 Uroboros systemd[1]: Stopping Login Service...
cze 15 08:29:39 Uroboros ckb-next-daemon[1568]: [I] Disconnecting /dev/input/ckb1
--
cze 15 08:29:39 Uroboros systemd[1]: Stopped D-Bus System Message Bus.
cze 15 08:30:15 Uroboros ckb-next-daemon[1568]: [I] Removed device path /dev/input/ckb1
--
cze 15 08:30:15 Uroboros kernel: usbhid 3-4:1.1: can't add hid device: -110
cze 15 08:30:15 Uroboros kernel: usbhid: probe of 3-4:1.1 failed with error -110
cze 15 08:30:15 Uroboros ckb-next-daemon[1568]: [E] os_sendindicators (usb_linux.c:258): Bad file descriptor
cze 15 08:30:15 Uroboros ckb-next-daemon[1568]: [I] Closing root controller
cze 15 08:30:15 Uroboros ckb-next-daemon[1568]: [I] Removed device path /dev/input/ckb0
cze 15 08:30:15 Uroboros systemd[1]: Stopped Corsair Keyboards and Mice Daemon.
```

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2105009/ckb-next-dev-detect-report.gz)

ZirconiumX, 2018-06-15:
@ashtonx Can you apply the following patch and see if that fixes it?

```diff
diff --git a/src/daemon/usb.h b/src/daemon/usb.h
index 6677b03..80516da 100644
--- a/src/daemon/usb.h
+++ b/src/daemon/usb.h
@@ -167,7 +167,7 @@ const char* product_str(short product);

 /// Some devices cause usbhid to spend a long time initialising it. To work around this, we intentionally uncleanly
 /// deinitialise the device, skipping the usbhid handover.
-#define NEEDS_UNCLEAN_EXIT(kb)          ((kb)->product == P_K65_RFIRE || (kb)->product == P_K70_RFIRE || (kb)->product == P_K70_RFIRE_NRGB)
+#define NEEDS_UNCLEAN_EXIT(kb)          ((kb)->product == P_K65_RFIRE || (kb)->product == P_K70_RFIRE || (kb)->product == P_K70_RFIRE_NRGB || (kb)->product == P_K95)

 /// Used for new devices that come with V3 firmware endpoint configuration out of the factory, but have fwversion < 0x300.
 /// Note: only the RGB variant of the K68 needs a v3 override.

```

ZirconiumX, 2018-06-15:
We've examined this, and it's rather bizarre, because [it's fixed in the kernel](https://elixir.bootlin.com/linux/v4.17/source/drivers/hid/hid-quirks.c#L63), but it still doesn't work for you.

ashtonx, 2018-06-16:
Sadly seems like it aint' helping.. unless I messed it up (added it to grub) ;/ maybe something went off when I added kernel from aur.. or when I patched it for sbz...  

I'll prolly add it to needs_unclean_exit for now. I'm linux newbie (3 weeks only) so I already assumed I'm gonna make clean install once 4.8 is released  (sbz drivers) and I tried enough stuff and understood enough.

Anyway thanks for help.

ZirconiumX, 2018-06-16:
ckb-next runs in userspace, so you don't have to rebuild for new kernels.
Linus would shout at anybody who broke userspace.

ashtonx, 2018-06-17:
Yeah it look nice but I plan to clean up my hdd, I already decided to try all sorts of things and then proceed with cleaner setup. Atm I'm in middle of testing and there's some junk leftovers ;)

Anyway, kernel 4.18 rc1 got released and it has sbz patch I needed, so far I stuck to 4.17-1 since it was only one I knew patch worked for. Anyway I'll check settings before compiling new kernel, maybe quirks are somehow disabled in default aur config.