ckb-next/ckb-next #169: [BUG] ckb-next-daemon 100% CPU -- ALL animations disabled (v0.3.0)

k3mist, 2018-05-03:
# Bug report

## General information

100% CPU on 0.3.0 with all animations disabled. Screenshots provided below.

This issue does not exist on 0.2.9. I upgraded from 0.2.9 to 0.3.0 to get support for Polaris pad.

```
OS X Sierra 10.12.6 (16G1314)
MacBook Pro (Retina, 15-inch, Mid 2015)
2.5 GHz Intel Core i7
AMD Radeon R9 M370X 2048 MB
Intel Iris Pro 1536 MB
```

### Source
Pre built package

## Logs & crash reports

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1971669/ckb-next-dev-detect-report.gz)

[image]
[image]

## What is the program's current behavior?

100% CPU

## What is the expected behavior of the program?

Not 100% CPU

## Any other notes & comments?

Seems whatever changes were made between 0.2.9 and 0.3.0 have caused the issue.

Currently reverting back to 0.2.9

ZirconiumX, 2018-05-03:
Is your processor an Intel Skylake, by any chance?

k3mist, 2018-05-03:
```
sysctl -n machdep.cpu.brand_string
Intel(R) Core(TM) i7-4870HQ CPU @ 2.50GHz
```
https://ark.intel.com/products/codename/51802/Crystal-Well

Crystal Well?

I believe its the mobile version of Haswell from a quick google search

ZirconiumX, 2018-05-03:
Yeah, Haswell. Hmmm. That makes it the same symptoms, but a different cause compared to #97.

tatokis, 2018-05-03:
It's clearly a regression in the daemon. 
Looking at the logs the gui can't access the features file on one device and the daemon registers the devices twice at some point as ckb3 and ckb4. 

It's completely unrelated.

k3mist, 2018-05-03:
Please let me know if there is anything I can do on my end to help track down the cause. 

My schedule is pretty tight with work atm and have plans this weekend. I'll see if I can find some time to build from source and see what I get with that.

As a side note, is there any build flags that would help?

tatokis, 2018-05-04:
Hmmm, try this.
Upgrade to 0.3.0, unplug all your corsair devices and reboot. Plug in the scimitar and check the CPU usage. If it's normal, plug in the polaris and check it again.

If you still get crazy high cpu usage, then I can build you previous versions of the daemon for you to manually run and test.
Alternatively, you can get the source, compile the daemon, and bisect it.
It's probably an issue in the last few commits, but I really can't tell.

k3mist, 2018-05-10:
@tatokis Sorry for the delay getting back. Been very busy.

I used the uninstall instructions on the OS X uninstall page to remove 0.2.9.

I can confirm with 100% certainty on 0.3.0 plugging in the **polaris** is the cause of 100% cpu usage on the ckb-next-daemon. 

Simply plugging it in causes the issue and unplugging it will bring the cpu usage back to normal _without_ restarting the daemon. This behavior does not change with the scimitar plugged in or unplugged.

Hope this helps!

tatokis, 2018-05-10:
@k3mist Do the LEDs on the mousepad work while in this state? As in, can you control them using the GUI?

k3mist, 2018-05-14:
@tatokis It did. It accepted my "static" non-animated all white led's to test if the animations had anything to do with cpu usage, but did not change the usage.

tatokis, 2018-05-14:
Right. I can either send you a debug build of the daemon to run with lldb or you can build it yourself.
If you do build it yourself, make sure you do a debug build by passing -DCMAKE_BUILD_TYPE=Debug to cmake.
Check if the daemon is running, and if so, stop it with `sudo launchctl unload /Library/LaunchDaemons/org.ckb-next.daemon.plist`.
Then run `sudo lldb ./bin/ckb-next-daemon`, type `run` and once it starts 100%ing the CPU press Ctrl-C and type `bt all` in the lldb prompt.

Paste all the output in here.

jrbenny35, 2018-06-24:
I ran into this problem as well. Here is the log:
```
(lldb) target create "./bin/ckb-next-daemon"
Current executable set to './bin/ckb-next-daemon' (x86_64).
(lldb) run
Process 12979 launched: './bin/ckb-next-daemon' (x86_64)
    ckb-next: Corsair RGB driver 0.3.0-35-g03fcb34
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair Gaming K70 LUX RGB Keyboard  at /var/run/ckb1
[I] Connecting Corsair Gaming MM800 RGB POLARIS at /var/run/ckb2
[E] iterate_devices_hid (usb_mac.c:879): Failed to create device plugin: e00002be
[E] iterate_devices_hid (usb_mac.c:879): Failed to create device plugin: e00002be
[E] iterate_devices_hid (usb_mac.c:879): Failed to create device plugin: e00002be
[E] iterate_devices_hid (usb_mac.c:879): Failed to create device plugin: e00002be
[I] Starting input thread for /var/run/ckb1
[I] Starting input thread for /var/run/ckb2
[E] os_usbrecv (via firmware.c:24): Got return value 0xe000404f
[W] _start_dev (device.c:43): Unable to load firmware version/poll rate
[I] Setup finished for /var/run/ckb2
[I] Setup finished for /var/run/ckb1
[I] Registering EventTap for modifier keys.
Process 12979 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
    frame #0: 0x00007fff7203520a libsystem_kernel.dylib`mach_msg_trap + 10
libsystem_kernel.dylib`mach_msg_trap:
->  0x7fff7203520a <+10>: retq
    0x7fff7203520b <+11>: nop

libsystem_kernel.dylib`mach_msg_overwrite_trap:
    0x7fff7203520c <+0>:  movq   %rcx, %r10
    0x7fff7203520f <+3>:  movl   $0x1000020, %eax          ; imm = 0x1000020
Target 0: (ckb-next-daemon) stopped.
(lldb) bt all
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
  * frame #0: 0x00007fff7203520a libsystem_kernel.dylib`mach_msg_trap + 10
    frame #1: 0x00007fff72034724 libsystem_kernel.dylib`mach_msg + 60
    frame #2: 0x00007fff49bd57d5 CoreFoundation`__CFRunLoopServiceMachPort + 341
    frame #3: 0x00007fff49bd4b27 CoreFoundation`__CFRunLoopRun + 1783
    frame #4: 0x00007fff49bd41a3 CoreFoundation`CFRunLoopRunSpecific + 483
    frame #5: 0x00007fff49c12c33 CoreFoundation`CFRunLoopRun + 99
    frame #6: 0x0000000100010947 ckb-next-daemon`usbmain at usb_mac.c:1012 [opt]
    frame #7: 0x000000010000ba21 ckb-next-daemon`main(argc=<unavailable>, argv=0x00007ffeefbffa00) at main.c:224 [opt]
    frame #8: 0x00007fff71eee015 libdyld.dylib`start + 1
  thread #3
    frame #0: 0x00007fff7203f292 libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff72206009 libsystem_pthread.dylib`_pthread_wqthread + 1035
    frame #2: 0x00007fff72205be9 libsystem_pthread.dylib`start_wqthread + 13
  thread #4
    frame #0: 0x0000000000000000
  thread #5
    frame #0: 0x00007fff7204014a libsystem_kernel.dylib`read + 10
    frame #1: 0x0000000100004f7e ckb-next-daemon`readlines(fd=5, ctx=0x0000000100203b80, input=0x0000700008ee2e58) at devnode.c:383 [opt]
    frame #2: 0x000000010000efb1 ckb-next-daemon`devmain(kb=<unavailable>) at usb.c:218 [opt]
    frame #3: 0x000000010000e835 ckb-next-daemon`_setupusb(context=0x000000010001d278) at usb.c:442 [opt]
    frame #4: 0x00007fff72206661 libsystem_pthread.dylib`_pthread_body + 340
    frame #5: 0x00007fff7220650d libsystem_pthread.dylib`_pthread_start + 377
    frame #6: 0x00007fff72205bf9 libsystem_pthread.dylib`thread_start + 13
  thread #6
    frame #0: 0x00007fff7204014a libsystem_kernel.dylib`read + 10
    frame #1: 0x0000000100004f7e ckb-next-daemon`readlines(fd=3, ctx=0x0000000100400a50, input=0x0000700008f65e58) at devnode.c:383 [opt]
    frame #2: 0x000000010000efb1 ckb-next-daemon`devmain(kb=<unavailable>) at usb.c:218 [opt]
    frame #3: 0x000000010000e835 ckb-next-daemon`_setupusb(context=0x000000010001d460) at usb.c:442 [opt]
    frame #4: 0x00007fff72206661 libsystem_pthread.dylib`_pthread_body + 340
    frame #5: 0x00007fff7220650d libsystem_pthread.dylib`_pthread_start + 377
    frame #6: 0x00007fff72205bf9 libsystem_pthread.dylib`thread_start + 13
  thread #7
    frame #0: 0x00007fff7203520a libsystem_kernel.dylib`mach_msg_trap + 10
    frame #1: 0x00007fff72034724 libsystem_kernel.dylib`mach_msg + 60
    frame #2: 0x00007fff49bd57d5 CoreFoundation`__CFRunLoopServiceMachPort + 341
    frame #3: 0x00007fff49bd4b27 CoreFoundation`__CFRunLoopRun + 1783
    frame #4: 0x00007fff49bd41a3 CoreFoundation`CFRunLoopRunSpecific + 483
    frame #5: 0x00007fff49c12c33 CoreFoundation`CFRunLoopRun + 99
    frame #6: 0x000000010000fffe ckb-next-daemon`os_inputmain(context=0x000000010001d278) at usb_mac.c:477 [opt]
    frame #7: 0x00007fff72206661 libsystem_pthread.dylib`_pthread_body + 340
    frame #8: 0x00007fff7220650d libsystem_pthread.dylib`_pthread_start + 377
    frame #9: 0x00007fff72205bf9 libsystem_pthread.dylib`thread_start + 13
  thread #8
    frame #0: 0x00007fff72204875 libsystem_pthread.dylib`_pthread_mutex_unlock_slow + 297
    frame #1: 0x00007fff49bd421f CoreFoundation`CFRunLoopRunSpecific + 607
    frame #2: 0x00007fff49c12c33 CoreFoundation`CFRunLoopRun + 99
    frame #3: 0x000000010001002d ckb-next-daemon`os_inputmain(context=0x000000010001d460) at usb_mac.c:477 [opt]
    frame #4: 0x00007fff72206661 libsystem_pthread.dylib`_pthread_body + 340
    frame #5: 0x00007fff7220650d libsystem_pthread.dylib`_pthread_start + 377
    frame #6: 0x00007fff72205bf9 libsystem_pthread.dylib`thread_start + 13
(lldb)
```

tatokis, 2018-06-24:
```
diff --git a/src/daemon/usb_mac.c b/src/daemon/usb_mac.c
index 287a8ac..ac33d01 100644
--- a/src/daemon/usb_mac.c
+++ b/src/daemon/usb_mac.c
@@ -324,7 +324,7 @@ CGEventRef mouse_event_modifier_callback(CGEventTapProxy proxy, CGEventType type
                 kb = keyboard + i;
                 // Only care about the active keyboard for grabbing the modifier keys.
                 // Once found, can move on.
-                if (!IS_MOUSE_DEV(kb) && kb->active == 1) {
+                if (!(IS_MOUSE_DEV(kb) || IS_SINGLE_EP(kb)) && kb->active == 1) {
                     pthread_mutex_lock(imutex(kb));
                     CGEventSetFlags(event, (kCGEventFlagMaskNonCoalesced | kb->modifiers | existingFlags));
                     pthread_mutex_unlock(imutex(kb));
```
This is probably not it, but can you try this patch and see if it helps?

tatokis, 2018-06-24:
Apparently the above patch makes macOS hang for me. Interesting

jrbenny35, 2018-06-24:
Should I still try it?

tatokis, 2018-06-26:
Apparently the OS locks up for me every time the daemon starts, with and without that patch. So yes, please try that patch and see if it helps.

k3mist, 2018-06-26:
@tatokis I did some more testing on 0.3 and another thing I noticed on my mouse is the animations are "stuttered" with high cpu usage, not 100% like with the mouse pad plugged in. But a steady ~15% on a 3.5ghz 6core cpu. 

Hard to explain what I mean by stuttered, but I guess the best comparison would be frames per second. If I was to compare to fps I would guess the animations were running somewhere between 10-15fps. The cpu usage of the daemon did not change either when I disabled the animations.

I'm currently running 0.2.9 again right now without issue, same mouse. Of course there is no support for polaris on 0.2.9 so I can't compare that. CPU usage is around 2% or so on 0.2.9

tatokis, 2018-08-01:
A month later, sorry, but if you're still around, I fixed an issue that might be closely related to this.
Can you please try this build? 

http://ckb-next.info/mac/ckb-next_v0.3.1-4-g0a698cc_10.11-SDK-polaris.dmg

jrbenny35, 2018-08-01:
I totally forgot about this. I'll give it a go tomorrow.

jrbenny35, 2018-08-01:
Just installed it. Looks to be running good. Seeing about 30% usage from ckd with a k70 lux and Polaris!

ZirconiumX, 2018-08-01:
30% is still painfully high, but an improvement.

jrbenny35, 2018-08-01:
> 30% is still painfully high, but an improvement. Well I know the percentages aren't 100% accurate

ZirconiumX, 2018-08-01:
I see the daemon at about 3 to 4% with animations on Linux. Still, this is
progress.

jrbenny35, 2018-08-01:
Maybe it was because I had the editor open. I am seeing between 4-8 now

tatokis, 2018-08-01:
Yeah, the GUI does take a bit of cpu time to animate the preview. Should I close this, then?

jrbenny35, 2018-08-02:
1 day later and I am still between the percentages I listed. I think its good! Thanks for the fix!

tatokis, 2018-08-02:
For future reference, it was probably this https://github.com/ckb-next/ckb-next/commit/0a698ccce4cb9a6950a80032627a31c168c4a668

closed 2018-08-02 by tatokis