ckb-next/ckb-next #170: K68 RGB unresponsive at CKB launch.

PaRkThEcAr, 2018-05-06:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1977666/ckb-next-dev-detect-report.gz)

Version: 0.3.0

OS. Mac OSX 10.13.4 (build 17E202)

When CKB launches, the K68 RGB becomes completely unresponsive.

tatokis, 2018-05-06:
When you say `When CKB launches`, do you mean the user interface?

PaRkThEcAr, 2018-05-06:
Tatokis, that is corret

PaRkThEcAr, 2018-05-10:
@tatokis do you need me to completely uninstall CKB beforehand?

tatokis, 2018-05-10:
No need to. Just install the new one. You might need to reboot afterwards, but I doubt it.

PaRkThEcAr, 2018-05-10:
@tatokis that works splendid! however, it seems that changing animations doesnt work :( but at least modifier keys, and general input seems to work. do you need me to send you more data? is there anything i can run outside of this package that would be beneficial for the future?

tatokis, 2018-05-10:
Perhaps run dev detect again. It might contain more info.

PaRkThEcAr, 2018-05-10:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1989939/ckb-next-dev-detect-report.gz)
@tatokis as per your request

PaRkThEcAr, 2018-05-10:
im not an expert, but i am good at testing :) i dont know how many K68 RGB's you have in your group to work with.

tatokis, 2018-05-10:
Absolutely no one. The only person who had a K68 RGB, helped us add support for it was only tested on linux.
Since the K68 has some oddities and hasn't been tested on macOS, you are the first person dealing with this.

It does, however, look like ckb drops the keyboard because it never gets a response back from it.
It's 4am here, however if you do find the time, please join our IRC room at a sensible European time and I can try and debug this with you.

PaRkThEcAr, 2018-05-10:
Okay, I have joined the IRC and should be able to join in this weekend. I am in the US (about a 9 hour difference)  so scheduling may be difficult on weekdays. I should be in this saturday, if not the following weekend.

closed 2022-03-05 by tatokis