ckb-next/ckb-next #171: K63 Keyboard Wired or Wireless showing no device coneected

DavidChuks, 2018-05-09:
I am not able to find my keyboard on the ckb application. It is not finding it at all.

I have attached my .gz report. 

Some help will be appreciated. Thanks.

I have also read other issues and none seem to resolve my issue.

Running Mac OS 10.13.4

Thanks.

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1987788/ckb-next-dev-detect-report.gz)

tatokis, 2018-05-09:
K63 Wireless is not supported yet.
You can follow https://github.com/ckb-next/ckb-next/pull/158 for updates

closed 2018-05-09 by tatokis

DavidChuks, 2018-05-09:
But is K63 wired supported?

ZirconiumX, 2018-05-09:
The original wired-only K63 is supported, yes.

DavidChuks, 2018-05-09:
oh it's the wired only version that is supported? wow..thanks for that reply. I was oblivious to that fact

tatokis, 2018-05-09:
Yes, the wireless K63 has a different protocol. It doesn't matter how it's connected to the computer.

DavidChuks, 2018-05-09:
Please add that clarification to the list of supported devices so people like me don't get confused. Great job guys. Now I know :(

DavidChuks, 2018-05-11:
Just came back here to say all the best in your exams! I am having mine too and I know how tough it can be. And hey you haven't disappointed me...you are doing great! keep it up @ZirconiumX