ckb-next/ckb-next #175: failed to start ckb-next-daemon.service

qwFaTE, 2018-05-14:
Hi,
sorry if this is something obvious, but I'm quite new to linux as a whole and need ckb-next for my glaive to actually work. That's all I want it for really.

I'm running MX Linux, which I believe is debian based (but I'm not sure), and after installing the program and running it, it tells me that the daemon isn't running and gives me this command: "sudo systemctl start ckb-next-daemon" to start it once. I gave it a go and I get this message: 

"Failed to start ckb-next-daemon.service: Unknown unit: ckb-next-daemon.service
See system logs and 'systemctl status ckb-next-daemon.service' for details."

I'm guessing the daemon doesn't even exist on my system.

I'm a noob to this stuff and I'd appreciate any help given, all I want is for my mouse to work. Thanks!

ZirconiumX, 2018-05-14:
That sounds like it didn't install properly; how did you install it?

qwFaTE, 2018-05-14:
I built from source, following the Ubuntu 16.04+ method in the "Linux Installation" section of this github project. It was the only method I could see that seemed to work for my system.

ZirconiumX, 2018-05-14:
> Where would I find that?

If you ran `./quickinstall` from the terminal, then paste the output of the
script.

If you tried to do something like double-click the script, then it won't be
able to install the daemon and you'll need to run it from a terminal.

qwFaTE, 2018-05-14:
Oh right, in that case the build log is:
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.11.0") 
-- ckb-next version: 0.3.0-9-g33c256d (Non-release)
-- Performing Test C_COMPILER_SUPPORTS_-Og
-- Performing Test C_COMPILER_SUPPORTS_-Og - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Og
-- Performing Test CXX_COMPILER_SUPPORTS_-Og - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST - Failed
-- Found Iconv: /usr/lib/x86_64-linux-gnu/libc.so
-- Checking for module 'libudev'
--   Found libudev, version 232
-- Found UDEV: /lib/x86_64-linux-gnu/libudev.so  
-- Searching for running ckb and/or ckb-next GUI
-- NOTE:
    Privileged access is required for operations upon the daemon at
    configure time as it is owned by root. If this is unacceptable,
    disable SAFE_INSTALL and prepare the filesystem manually.
CMake Warning at src/daemon/CMakeLists.txt:406 (message):
  No supported system service detected.

      Supported services are: systemd, launchd, OpenRC, upstart.

-- Found QuaZip: /usr/lib/x86_64-linux-gnu/libquazip5.so  
-- Found PulseAudioSimple: /usr/lib/x86_64-linux-gnu/libpulse-simple.so  
-- Found PulseAudio: /usr/lib/x86_64-linux-gnu/libpulse.so  
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.8") 
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Generating binary.qrc
-- Importing ckb-next-key.gpg
-- Generating image.qrc
-- Importing ckb-next.png
-- Importing icon_blank.png
-- Importing icon_mode.png
-- Importing icon_mode1.png
-- Importing icon_mode1_hardware.png
-- Importing icon_mode2.png
[... 208 lines cut ...]
[ 46%] Built target rain
[ 48%] Built target mviz
[ 49%] Automatic MOC, UIC and RCC for target ckb-next
[ 49%] Built target ckb-next_autogen
[100%] Built target ckb-next
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/bin/ckb-next-daemon
-- Up-to-date: /etc/udev/rules.d/99-ckb-daemon.rules
-- Installing: /usr/lib/cmake/ckb-next/Animation/AnimationTargets.cmake
-- Up-to-date: /usr/lib/cmake/ckb-next/Animation/AnimationConfig.cmake
-- Installing: /usr/lib/cmake/ckb-next/Animation/AnimationConfigVersion.cmake
-- Installing: /usr/local/libexec/ckb-next-animations/gradient
-- Installing: /usr/local/libexec/ckb-next-animations/heat
-- Installing: /usr/local/libexec/ckb-next-animations/pinwheel
-- Installing: /usr/local/libexec/ckb-next-animations/rain
-- Installing: /usr/local/libexec/ckb-next-animations/random
-- Installing: /usr/local/libexec/ckb-next-animations/ripple
-- Installing: /usr/local/libexec/ckb-next-animations/wave
-- Installing: /usr/local/libexec/ckb-next-animations/mviz
-- Installing: /usr/local/bin/ckb-next
-- Installing: /usr/local/share/applications/ckb-next.desktop
-- Up-to-date: /usr/local/share/icons/hicolor/512x512/apps/ckb-next.png
Done.
Thank you for installing ckb-next!

ZirconiumX, 2018-05-14:
>CMake Warning at src/daemon/CMakeLists.txt:406 (message):

No supported system service detected.

>  Supported services are: systemd, launchd, OpenRC, upstart.

The script couldn't detect that you were using systemd.

This sounds like a distro quirk, I'll have to investigate.

qwFaTE, 2018-05-14:
Alright thanks man. :)

ZirconiumX, 2018-05-14:
As a temporary workaround, if you look in the `build/bin` folder, you'll
find `ckb-next-daemon`. Run `sudo nohup ./ckb-next-daemon`, and your Glaive
should work. The `nohup` is so that you can close the terminal without
breaking your mouse.

qwFaTE, 2018-05-14:
Awesome! Thanks for the help dude, I really appreciate it!

ZirconiumX, 2018-05-14:
We've been in your shoes, so we know the importance of being helpful. It's
the least we can do to thank you for using CKB.

tatokis, 2018-05-14:
Can you run `ls /run`, `ls /run/systemd` and post the output?
Also, please use markdown quotes for quote blocks.

qwFaTE, 2018-05-14:
Okay, so I got this from running ls /run:

> acpid.pid      console-setup       lightdm         rpcbind          sm-notify.pid
acpid.socket   crond.pid           lightdm.pid     rpcbind.lock     systemd
acpi_fakekey   crond.reboot        lock            rpcbind.pid      systemd-shim-state
agetty.reload  cups                motd.dynamic    rpcbind.sock     tlp
alsa           dbus                mount           rpc_pipefs       udev
atd.pid        dhclient6-eth0.pid  network         rpc.statd.pid    udisks2
avahi-daemon   dhclient-eth0.pid   NetworkManager  rsyslogd.pid     user
blkid          haveged.pid         ntpd.pid        samba            utmp
cgmanager      initctl             openvpn         sendsigs.omit.d
cgmanager.pid  initramfs           resolvconf      shm

And this from ls /run/systemd/:

> inhibit  seats  sessions  users

Hopefully that's helpful, though I'm not sure what markdown quotes are exactly, so sorry if I got that wrong.

ZirconiumX, 2018-05-14:
Markdown quotes use ` (backticks) around them.

GitHub has a guide here:
https://guides.github.com/features/mastering-markdown/

tatokis, 2018-05-14:
I pushed a commit that should fix this https://github.com/ckb-next/ckb-next/commit/7fe024e81cad4641abe2a9445a7d877c49a973c7 .

Can you try quickinstall with the new code, reboot, and then see if it works on boot?

qwFaTE, 2018-05-14:
It still doesn't seem to work.

ZirconiumX, 2018-05-14:
```
-- systemd detected
-- Generating and importing ckb-next-daemon.service (systemd)
Failed to retrieve unit: No such method 'GetUnit'
Failed to get unit file state for ckb-daemon.service: Unknown error 1349259488
Failed to retrieve unit: No such method 'GetUnit'
enabled
```
Now, ckb-next uses systemd to generate a config file, but MX Linux uses a fake systemd, which appears to not support this feature, so we need a different approach. It probably involves us writing a sysvinit script, which is going to be fun (it was deprecated for a reason).

qwFaTE, 2018-05-14:
Oh man, sorry about this! Didn't think it'd need this much work. D:

Thanks very much though. :)

OvermindDL1, 2018-05-14:
Huh, did some reading about mxlinux and I'm really not liking a few things about it...

However, it looks like you can setup real systemd in it, instructions at:  https://mxlinux.org/wiki/system/systemd

There only argument against it is that it is 'not the unix way', yeah sure but `init` being simple is what gave us the horrors of sysvinit, systemd is much faster, more capable, actually supports modern hardware things like, oh, hotswapping USB natively (which ckb uses via udev) among others, things that are *very* hard or impossible to tack-on to sysvinit...

I'm not a fan of this distro it seems...  >.>

tatokis, 2018-05-14:
@OvermindDL1 I don't think this is the right thread for this.
That said, I am working on a script.
Also, for a distro that has a manual that says "systemd installations are not officially supported", I wouldn't suggest doing something like this to someone who is pretty new to this.

tatokis, 2018-05-15:
@qwFaTE Please try with the latest code.

OvermindDL1, 2018-05-15:
Heh, sorry yeah, I was just curious why it seemed to have some systemd commands but was not systemd, it appears they made a systemd-shim(s?) thing, which seems completely point defeating to me...  But as I found out how to setup actual systemd I figured linking that might be useful (with a disclaimer about their disclaimer).  They state it is not officially supported because the dev's of that distro do not like systemd for non-technical reasons.

qwFaTE, 2018-05-15:
@tatokis Amazing man, works perfectly!

Thanks for this, I'm really sorry for the extra work you had to go through for this. It's really appreciated.

Have a great day! :)

ZirconiumX, 2018-05-15:
I'm so glad it works. We were dreading having to write an init script, but it doesn't seem to be too painful.

Take care, and have a good day yourself!

closed 2018-05-15 by ZirconiumX

robabby, 2019-05-11:
Apologies for reviving a dead issue, and happy to create a new ticket if you prefer, but I just came across this issue on Ubuntu 18.04.2.  I had this library working just fine until I had to re-install my OS due to upgrading my Video Card.  Now the install is failing with this error.  

I attempted the following the recommendation, but there is no `build/bin` folder, just `build/src`:
> As a temporary workaround, if you look in the `build/bin` folder, you'll find `ckb-next-daemon`. Run `sudo nohup ./ckb-next-daemon`, and your Glaive should work. The `nohup` is so that you can close the terminal without breaking your mouse.

Here is my install log:
```
➜  ckb-next git:(master) sudo ./quickinstall                                                                                                          
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- ckb-next version: 0.4.0-9-gfd509f5 (Non-release)
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST - Failed
-- Found Iconv: /usr/lib/x86_64-linux-gnu/libc.so
-- Checking for module 'libudev'
--   Found libudev, version 237
-- Found UDEV: /lib/x86_64-linux-gnu/libudev.so  
-- Searching for running ckb and/or ckb-next GUI
-- NOTE:
    Privileged access is required for operations upon the daemon at
    configure time as it is owned by root. If this is unacceptable,
    disable SAFE_INSTALL and prepare the filesystem manually.
-- systemd detected
-- Generating and importing ckb-next-daemon.service (systemd)
inactive
Failed to get unit file state for ckb-daemon.service: No such file or directory
inactive
Failed to get unit file state for ckb-next-daemon.service: No such file or directory
-- Found QuaZip: /usr/lib/x86_64-linux-gnu/libquazip5.so  
-- Found PulseAudioSimple: /usr/lib/x86_64-linux-gnu/libpulse-simple.so  
-- Found PulseAudio: /usr/lib/x86_64-linux-gnu/libpulse.so  
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfig.cmake:27 (message):
  The imported target "Qt5::Gui" references the file

     "/usr/lib/x86_64-linux-gnu/libEGL.so"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake:50 (_qt5_Gui_check_file_exists)
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake:72 (_qt5gui_find_extra_libs)
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfig.cmake:184 (include)
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Widgets/Qt5WidgetsConfig.cmake:101 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:28 (find_package)
  src/gui/CMakeLists.txt:72 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/rob/Workbench/ckb-next/build/CMakeFiles/CMakeOutput.log".
See also "/home/rob/Workbench/ckb-next/build/CMakeFiles/CMakeError.log".
An error occurred,
press enter to exit.
```

ZirconiumX, 2019-05-11:
By the looks of it you need to install EGL from wherever you're getting
your drivers.

robabby, 2019-05-11:
@ZirconiumX My video card drivers?  I installed the Nvidia drivers from source.

ZirconiumX, 2019-05-11:
Definitely do not do that on Ubuntu. Use one of the Ubuntu provided
packages instead.

robabby, 2019-05-11:
@ZirconiumX I tried that, but had a hell oif a time getting them to work with the newer RTX card.  It was a real pain the butt.

ZirconiumX, 2019-05-11:
Using the graphics-drivers PPA?

robabby, 2019-05-11:
@ZirconiumX yes.  Wound up having to follow this guide:
https://medium.com/@avinchintha/how-to-install-nvidia-drivers-and-cuda-10-0-for-rtx-2080-ti-gpu-on-ubuntu-16-04-18-04-ce32e4edf1c0

ZirconiumX, 2019-05-11:
Honestly Ubuntu problems are not my forté, that's @tatokis's department,
but he's currently asleep. Either way the build is failing because your
graphics drivers are not properly installed, not because of any issue on
our end.

robabby, 2019-05-11:
@ZirconiumX Whelp, that is good to know that the problem is straightforward.  Thank you for your time.

Given the hellish time I had getting this new card working, I did not install the CUDA package for the card because frankly I was pretty tired fo the graphics issues and was happy enough when things started working and looking correct.  Would installing the CUDA packages resolve this issue?

ZirconiumX, 2019-05-11:
Well, the main use of EGL is actually Wayland, so that's what you'll be
looking for.

robabby, 2019-05-18:
In case anyone comes here after me, I was able to get the Nvidia PPA drivers working following the **How to Install the Latest Version Of Nvidia Drivers** of this article:

https://www.linuxbabe.com/ubuntu/install-nvidia-driver-ubuntu-18-04

`ckb-next` is up and running just fine :+1:

phpbbireland, 2019-05-22:
After following the install guidelines using git (22nd May 2019) my MX build appears to work fine with one small little issue... 

***Booting/Shutdown:***  A start job (and stop job) for Corsair Keyboard / Mouse timed out... takes an extra minute and a half to boot and shut down due to time out but everything else appears to work perfectly...

**Log:** Timed out waiting for device /dev/disk/by-uuid/edf401fd-blahblah ...

gmgrigsby, 2024-07-18:
Apologies for posting in this old thread, but facing the same issue on my wife's computer running MX Linux 23.3. Was the final solution to enable Systemd for qwFaTE or is there now a script available that will work with SysV init?

Thanks in advance.