ckb-next/ckb-next #176: Native keyboard press repeat/delay statically fixed when Corsair keyboard is removed

AaronMartone, 2018-05-14:
With this software installed and running, and my Corsair Strafe RGB plugged in, all seems right. They keyboard is responsive and if I click and hold a button down there is a very short delay before it begins to repeat quickly (the way I like it)

However, now that I remove my keyboard (and whether I have the daemon running or not), my Macbook Pro's built-in keyboard's hold/repeat seems statically set. The delay before repeat is about the same, but when it does repeat, it's outputting approximately 3 characters per second (insanely slow) compared to how the system works when my external keyboard is plugged in.

I checked system settings, and Key Repeat is set to Fast while Delay Until Repeat is set to Short, but that isn't the performance I've had prior to installation of the drivers.

I'd love to help troubleshoot however I can. And I want to also thank the developer for this project; picking up the ball where Corsair dropped it.

ZirconiumX, 2018-05-14:
This is a curious issue. It's going to be down to @tatokis to attempt to
debug this, because Mac OS X is something of a second-class support
(neither me nor tatokis have Macs because we're poor students), but we try
our best.

tatokis, 2018-05-14:
I know the daemon reads the repeat settings from the OS so that it replicates them. I don't see why it'd affect other keyboards, but it's macOS, soooo. 

That said, all of this will be unnecessary once we switch over to the Karabiner input kext.
I can get you a fixed build to try and see if it solves your issue, but since the code is still in early stages, it may have a few bugs.

AaronMartone, 2018-05-14:
Thank you for the offer. May I ask, do you know of a way to display those related values via the terminal so I can actively inspect them? Would there be a possibility that they are not synced with what the OS shows, and/or maybe have been set in such a way where the powers that be no longer have rights to alter their value (and as such could be erroring out in some log file I'm unaware of silently)?

For clarification, the affected keyboard is the one built into the Macbook Pro. Only when I use that do I see this discrepancy between what the OS shows and the effect in apps.

I can be patient for this Karabiner input feature to get integrated and further tested. Luckily for me, using the builtin keyboard is rare. But when long coding stints are being done, the noticeable lack of hold+repeat becomes more apparent.

Also, again, EXCELLENT work on this tool. I'm very happy I can use a keyboard I love on my Mac.

tatokis, 2018-05-15:
I did understand what you meant, and that is the puzzling part.
I don't think there is a command line application for you to check the values.

I do recently recall replacing a deprecated function with a new one though, but it appeared that they had the same functionality. (I say appeared because I couldn't find any documentation.)

If you want to check the values in a terminal, you'll need to write a small program that does what this function does.
https://github.com/ckb-next/ckb-next/blob/master/src/daemon/input_mac.c#L181

tatokis, 2018-07-19:
Closing due to karabiner backend

closed 2018-07-19 by tatokis