ckb-next/ckb-next #177: How to upgrade to ckb-next daemon?

octillery143, 2018-05-16:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

### Source

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

tatokis, 2018-05-16:
Sorry, what?

octillery143, 2018-05-16:
im new, but it says to upgrade ckb-next-daemon with a warning about a driver mismatch.

octillery143, 2018-05-16:
(beta-0.2.6)

tatokis, 2018-05-16:
Which operating system are you running?
Which OS version?
How did you install ckb?
How did you install ckb-next?
Have you tried rebooting?

Next time please describe your issue better, your setup, and what steps you have taken.

octillery143, 2018-05-16:
im running macOS high sierra 10.13.4
i did not installed ckb, i installed ckb-next with the pre-made package that was provided
and i did try rebooting

octillery143, 2018-05-16:
ill try again.

tatokis, 2018-05-16:
The latest version is 0.3.0, so uninstall any old versions you had installed. And install https://github.com/ckb-next/ckb-next/releases/download/v0.3.0/ckb-next_v0.3.0.dmg

octillery143, 2018-05-16:
Thank you, i did and rebooted and now its working.

closed 2018-05-16 by ZirconiumX

MordicusEtCubitus, 2020-03-22:
Here is my file to support my new device : ironclaw mouse wireless, connected on usb port using cable at the moment of building the report

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/4365243/ckb-next-dev-detect-report.gz)