ckb-next/ckb-next #178: Fedora 27 issue with ckb daemon not starting

AngelusDulcis, 2018-05-17:
Logs are attached, I think I saw a closed issue with this earlier but the latest version doesn't resolve it for me. 

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2013066/ckb-next-dev-detect-report.gz)

[image]

tatokis, 2018-05-17:
Try `sudo rm /etc/init.d/ckb-next-daemon.sh` and see if that makes it start

AngelusDulcis, 2018-05-17:
systemctl start ckb-next-daemon
Failed to start ckb-next-daemon.service: Access denied
See system logs and 'systemctl status ckb-next-daemon.service' for details.

systemctl status ckb-next-daemon.service
Failed to get properties: Access denied

enable ckb-next-daemon
Failed to enable unit: Unit /run/systemd/generator.late/ckb-next-daemon.service is transient or generated.

tatokis, 2018-05-17:
Use sudo systemctl if you aren't doing so. Need to be root.

AngelusDulcis, 2018-05-17:
I'm in a root session, I'm sorry I should have clarified that.

tatokis, 2018-05-17:
Try the latest code from master with quickinstall. That should fix it.
Make sure /etc/init.d/ckb-next-daemon.sh doesn't exist both before and after.

AngelusDulcis, 2018-05-17:
There was the magic, installed it again rebooted and we're up right as rain. Thank you @tatokis !

closed 2018-05-17 by AngelusDulcis