ckb-next/ckb-next #181: Unable to stop macro record

Perflyst, 2018-05-19:
# Bugreport

### Source
  * built manually using quickinstall script

## What is the program's current behavior?
Click on stop on macro record does not stop recording.
## What is the expected behavior of the program?
Click on stop should stop macro record.

---

# Something else
[image]

Kernel: 4.15.0-20-generic #21~16.04.1-Ubuntu
ckb-next version: ckb-next 0.3.0-16

tatokis, 2018-05-19:
Should be fixed now. Thanks for reporting

closed 2018-05-19 by tatokis

Perflyst, 2018-05-20:
Thanks, works now. One question, if I record a macro that takes the click was on which position my cursor is? When I do that he clicks where I'm currently with the mouse and not where I clicked in the macro recording.

ZirconiumX, 2018-05-20:
I don't think the macro system supports that, because mouse movement is
relative, not absolute.

So if you record the relative motions, but start on a different pixel, you
will end on a different pixel.

It *might* be possible to ask the window system where the absolute cursor
position is, but that's going to be platform dependent.

commit 7feb7ac1 (2018-05-19): Don't steal focus when recording mouse1 macros

Closes #181