ckb-next/ckb-next #182: Sometimes the keyboard is unresponsive at login screen

perqin, 2018-05-20:
# Devices information

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2020285/ckb-next-dev-detect-report.gz)

---

# Bug report

## General information

### Source

ckb-next package in AUR

## Logs & crash reports

ckb-next-dev-detect-report.gz is uploaded above.

## What is the program's current behavior?

When booting up to display manager login screen, the keyboard cannot send any input. But the RGB lightings are on. I have to restart the system using mouse. The keyboard is working when selecting boot options in grub screen.

This issue happens randomly, so restarting system helps.

## What is the expected behavior of the program?

Working well at boot.

## Any other notes & comments?

---

# Something else

tatokis, 2018-05-20:
Can you also try adding this to your kernel cmdline?
` usbcore.quirks=1b1c:1b11:g`

perqin, 2018-05-20:
@tatokis Already added:

```
perqin@archlinux $ cat /etc/default/grub | grep CMDLINE
GRUB_CMDLINE_LINUX_DEFAULT="quiet usbhid.quirks=0x1B1C:0x1B11:0x20000408"
GRUB_CMDLINE_LINUX=""
```

I am using K95 RGB on Arch Linux.

tatokis, 2018-05-20:
Just to clarify, I asked for you to try this
```
GRUB_CMDLINE_LINUX_DEFAULT="quiet usbhid.quirks=0x1B1C:0x1B11:0x20000408 usbcore.quirks=1b1c:1b11:g"
```