ckb-next/ckb-next #183: Mute indicator not working in v0.3.0

sjthespian, 2018-05-24:
# Bug report

## General information

### Source
  Downloaded from GitHub releases

## Logs & crash reports

`ckb-next-dev-detect-report.gz` attached

## What is the program's current behavior?
After upgrading from v0.2.9 to v0.3.0, the mute indicator no longer indicates the mute status on my K70 LUX RGB keyboard. Once I set a color for "Unknown" I see that it is always indicating an unknown state.

## What is the expected behavior of the program?
Expect the mute button to indicate the mute status as shown in the Performance settings for my keyboard profile.

sjthespian, 2018-05-24:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2033330/ckb-next-dev-detect-report.gz)

tatokis, 2018-05-24:
I didn't realise it was working in 0.2.9. I thought it was broken from the beginning.
I am currently working on other major mac fixes, so this might take a bit, but I am going to look into it asap.

tatokis, 2018-05-24:
@sjthespian I have reverted the media code to the old one, and that seems to fix the issue for now.
https://github.com/ckb-next/ckb-next/commit/80f672f26c94e2bef54ae2d12a1c22080ee97bb9

Here's a build for you to use:
```
fb5a9c53292c3c3a6b2fdaab4e56b0beff1871af0e3faabc6fdf303a640bf55a  ckb-next_v0.3.0-19-g80f672f-old-media-mac.dmg
```
http://ckb-next.info/mac/ckb-next_v0.3.0-19-g80f672f-old-media-mac.dmg

sjthespian, 2018-05-24:
Wow, thanks for the amazingly fast turnaround! I'm used to GitHub issues that have to celebrate a birthday or two before anyone notices them.  (some on projects I've contributed to :-) )

Thanks!!!

tatokis, 2018-05-24:
Will be fixed properly in the next release.
Thanks for reporting

commit 93f34db7 (2018-05-24): Switch back to AudioObject with correct data type

Turns out, AudioObjectGetPropertyData failed with error 561211770
meaning "The buffer provided for the property is too small."

Switching from bool to int fixes that.

Closes #183
closed 2018-05-24 by tatokis