ckb-next/ckb-next #184: "No Devices connected" everytime the computer gets put into sleep.

BillBurns1, 2018-05-30:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

### Source

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

So first off, when this works it is a fantastic program!! Thank you. The problem i am running into is whenever i put my Mac Mini running OS 10.13.4 into sleep mode, ckb-next stops, working telling me there are no devices. i have tried restarting and reinstalling. Reinstall works until computer goes into sleep mode, at least i think sleep mode is the problem. I am using a Corsair Strafe RGB. When it says no devoce connected my WASD and left, right, up and down arrows are white and the rest of the keyboard is red, the corsair logo is yellow, the num lock, caps lock and scroll lock are off and will not come on by pressing however still work. Hope this is all you guys need to sus out whats going on, thanks in advance,
Bill

tatokis, 2018-05-30:
Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Then upload `ckb-next-dev-detect-report.gz`

BillBurns1, 2018-06-01:
I'm really sorry but i don't know how to do all that? could you elaborate?

BillBurns1, 2018-06-01:
Never mind, googles a good thing!! here you go.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2064254/ckb-next-dev-detect-report.gz)

tatokis, 2018-06-02:
Can you install this version, reboot, and see if the problem persists?
http://ckb-next.info/mac/ckb-next_v0.3.0-19-g80f672f-old-media-mac.dmg

breckenedge, 2018-06-06:
I also had this problem. The updated .dmg worked. Here's my device report: [ckb-next-dev-detect-report.txt](https://github.com/ckb-next/ckb-next/files/2077315/ckb-next-dev-detect-report.txt)

ark4peace, 2018-06-06:
@tatokis I had the same problem (program "loosing" the devices), installed the dmg, and it works now.

Xalior, 2018-06-12:
I can confirm I had this problem with a K70 Lux under Sierra, the build above fixed it, but introduced other issues (random mouseclicks not noticed by finder) but the build in https://github.com/ckb-next/ckb-next/issues/186#issuecomment-394566227 seems to have solved both that, and this bug.

I still, sometimes, have to reboot to get my animations & input back, but that has only happened to me once on build 31 so far...

-Dx

ZirconiumX, 2018-06-12:
> by finder) but the build in #186 (comment)

We have plans to release 0.3.1 soon after exams. Build 31 uses Karabiner as
its input backend, because Mac OS X has become a difficult platform for
userspace input.

ZirconiumX, 2020-05-14:
Since 0.3.1 has been out for a year, closing as stale.

closed 2020-05-14 by ZirconiumX