ckb-next/ckb-next #185: MM800 Polaris not responding to commands after boot-up/restart

ExtremeProgrammerPro, 2018-06-03:
# Devices information
The report is attached

# Bug report

## General information
### Problem Description
After booting up, the MM800 just continues to have its stock rainbow pinwheel and any changes in the software are not reflected by the hardware. Also, the brightness in the software is set to 33% even though it was previously set to 100%. I am using the latest firmware version - 1.02. On an unrelated note, the CPU usage jumps by about 15% when the MM800 is plugged in versus when it isn't. #169 .

### Solution
Only after unplugging and plugging back in does it start to work.

### Source
  * pre-built pkg for macOS - latest release 0.3.0

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2066397/ckb-next-dev-detect-report.gz)

tatokis, 2018-06-03:
#169 is for 100% CPU on the daemon. Make sure it's the same issue first.
Also, regarding the unplug issue, try this and see if it helps http://ckb-next.info/mac/ckb-next_v0.3.0-19-g80f672f-old-media-mac.dmg
Be sure to reboot once you install that.

ExtremeProgrammerPro, 2018-06-03:
Unfortunately, even after uninstalling (following the uninstall mac procedure in the wiki) and installing the version provided in the previous post, the mousepad is completely ignoring the commands sent to it. As for the #169 issue, it probably isn't the same since i am only facing a slight increase in cpu usage. Thanks for the quick reply btw. I am ready to provide the needed information so we can sort this issue out (hopefully).

tatokis, 2018-06-03:
When you ran the script to generate the report, was the mousepad not accepting commands?
If not, reboot and run the script while the mousepad isn't behaving properly.

tatokis, 2018-06-03:
Can you run these in a terminal and see if they help?
```
echo active > /var/run/ckb1/cmd
echo active > /var/run/ckb2/cmd
```
If that doesn't work, unplug the mousepad and plug it back in, and after confirming that it works, generate and upload a new report.

ExtremeProgrammerPro, 2018-06-03:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2066423/ckb-next-dev-detect-report.gz)

Unfortunately, the commands had no effect. I did as you said and after plugging the mousepad back in, it started working so I generated this report.

SethBarberee, 2018-06-04:
For my Polaris, I usually close the GUI and reopen it. That usually fixes the issue.

tatokis, 2018-06-04:
@SethBarberee Sadly this seems to be a timing issue with the Apple HID driver not being up at the right time.
From the reports above, the daemon claims/attaches to different interfaces depending on if the system was booted with the device or if the device was plugged in later on.

After the system has booted up and the HID driver starts, we can not get direct control of the mousepad to send Interrupts, so we have to send control reports through the HID driver.
My guess is the daemon starts before the HID driver and since that rarely happens, we've never really debugged/set up the responsible code to handle a polaris specifically on such a situation.

I'll make a debug build tomorrow to get more info as to what is going on during boot and go from there.
I'll also have to update the dev detect script to try and scan the HID IORegistry entries for devices, not just IOUSB.

ExtremeProgrammerPro, 2018-06-06:
@SethBarberee Thanks for your response. I tried this prior to posting here and sadly it has no effect whatsoever and I'm guessing that's due to the daemon running on a separate process/thread from the GUI, so it's logical that there's no effect. Might be wrong of course.

tatokis, 2018-06-06:
@ExtremeProgrammerPro there's a different issue on linux where the device will reset but not switch back to software mode. And a way to do that is by restarting the gui, but in this case it's more of an issue of the daemon not being able to talk to the device.

Either way, unfortunately you will need to wait until next week, as I am currently busy with exams. Apologies.

ExtremeProgrammerPro, 2018-06-06:
@tatokis Thanks, I'm not in any rush and would be happy to help in any way I can. I guessed it might be a daemon issue.

tatokis, 2018-06-17:
As promised, here is the build.
http://ckb-next.info/mac/ckb-next-daemon-inputdbg
```
62d76b94c8836a1f238295401e722202813401f77d7ba9de22277b2855ddb7e1  ckb-next-daemon-inputdbg
```

Download the daemon binary, verify the sha256sum matches (can be done with `shasum -a256 ckb-next-daemon-inputdbg`), then run this command in the same directory as where you downloaded the daemon.
```
sudo cp ckb-next-daemon-inputdbg /Library/Application\ Support/ckb-next-daemon
``` 

After that, reboot with the polaris plugged in.
Make sure the polaris is not functioning, wait for a few seconds, unplug it and plug it back in.
Check that it is now working.

Once you do that, re-run ckb-next-dev-detect and upload it.
I will most likely be able to fix this after receiving that report.

ExtremeProgrammerPro, 2018-06-17:
Hi there, I have removed ckb-next completely. Do I reinstall it before performing the given steps? Thank you.

tatokis, 2018-06-17:
Yes, please.

ExtremeProgrammerPro, 2018-06-17:
Unfortunately, it doesn't work at all with this daemon. I have followed your instructions exactly and have tried them twice. When I install the released dmg daemon, it works. But with the attached one it doesn't. The checksum matches the one that you provided.

tatokis, 2018-06-17:
Can you generate and upload a report?

ExtremeProgrammerPro, 2018-06-17:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2109333/ckb-next-dev-detect-report.gz)

ExtremeProgrammerPro, 2018-06-17:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2109360/ckb-next-dev-detect-report.gz)
Followed the same procedure. Now it works after plugging out and in but not on boot unfortunately. Thanks for taking your time to troubleshoot this btw.

tatokis, 2018-06-17:
http://ckb-next.info/mac/ckb-next-daemon-inputdbg3
I fixed a few issues. Hopefully this will work.
If not, it will provide the right debugging information to fix it.

```
5912a54aaa6d8a1aaeb4bef0b8d435369429c844c7a54670f01678855b30b0bc   ckb-next-daemon-inputdbg3
```

ExtremeProgrammerPro, 2018-06-17:
Great news! It works on boot - rebooted twice and it starts each time. Thank you so much! I have one last issue though. All of the debug daemons seem to make the animations slow and choppy. The release version works flawlessly. Is this normal for a debug version?

ExtremeProgrammerPro, 2018-06-17:
One last issue that I have is that all versions (just tested release) seem to sometimes run very choppy (debug and release versions) on the animations. It used to work just fine and I have no idea what happened. No matter which version I try, this happens.

tatokis, 2018-06-17:
No idea, but run the commands from the first block to completely uninstall ( https://github.com/ckb-next/ckb-next/wiki/macOS-Installation#uninstallation ), then try this new release build to see if it will help.

http://ckb-next.info/mac/ckb-next_v0.3.0-40-g572c7dd_polarisusb.dmg

```
7727c4fdb2161ceec74c153f327ad08894350352c6f3e6ee5ca4b40dde89a104  ckb-next_v0.3.0-40-g572c7dd_polarisusb.dmg
```

Make sure to reboot after installing it. Do check CPU usage too, to see if it is higher than usual.

ExtremeProgrammerPro, 2018-06-17:
This build works great, thanks! The animations are still choppy though. The CPU usage is only around 5-6% and the poll rate is 8ms according to the ckb-next GUI. Maybe I need to open a separate issue for this, but it's weird that it used to have flawless animations with the release version but now none of the provide that. I will investigate further but thanks again for fixing the boot issue!

tatokis, 2018-06-17:
Sure, but I have no idea what could be causing it.
Perhaps some power saving features? I know the GUI turns off App Nap.

Maybe try renicing/changing the priority? There were some code changes in all the new daemons I sent, but they weren't in the 0.3.0 release, so I have no idea what could be going on.

closed 2018-06-17 by tatokis

ExtremeProgrammerPro, 2018-06-17:
I am sorry to bother you again @tatokis, I think that the animation performance issue is linked with this issue. The animations used to work perfectly before because they would only work after unplugging and plugging back in. If I boot now, the animations start but are choppy. Then if I quit ckb-next (from the "Quit" button in the GUI) and run the app again, then animations are flawless. So it may be that the daemon has some trouble sending packets initially. But when the daemon is restared, it works perfectly?

tatokis, 2018-06-17:
>Then if I quit ckb-next (from the "Quit" button in the GUI) and run the app again, then animations are flawless.

The daemon doesn't stop when you Quit the GUI.
Untick `Start ckb-next at login`, and then start it manually. See if the issue persists.

ExtremeProgrammerPro, 2018-06-17:
I tried it now. The animations don't start until I manually start the GUI. The issue is not present in this case - when starting the GUI manually - the animations are perfect. How do I make it start on boot without stutter?

tatokis, 2018-06-17:
Are the animations still smooth if you quit the GUI and then run it manually from a terminal with this?
```
/Applications/ckb-next.app/Contents/MacOS/ckb-next --background
```

ExtremeProgrammerPro, 2018-06-17:
Yes it does, thanks a lot! I will set it up as an Automator task but if there is anyway of fixing this within the app code it would be highly appreciated.

tatokis, 2018-06-17:
Only thing I can think of is adding a startup delay, as I can't reproduce this issue, and from what I can see you are the only person who ever had this problem.

Just as a test, re-enable start on login in ckb.
Then create a file in your home directory called `startckb.sh` with the following contents.
```
#!/bin/sh
sleep 5
/Applications/ckb-next.app/Contents/MacOS/ckb-next --background
```
Then run 
```
chmod +x startckb.sh
```

Once you've done that, open `/Users/youruser/Library/LaunchAgents/org.ckb-next.plist` with a text editor. (Where youruser is your username. You can `echo $USER` to get it.)
Near the middle, there's a line with `<string>/Applications/ckb-next.app/Contents/MacOS/ckb-next</string>`

Replace it with `<string>/Users/youruser/startckb.sh</string>`, save the file, and reboot.

If it does work, I'll have to find a more permanent way to add a delay, or somehow make sure the gui runs after some condition.

ExtremeProgrammerPro, 2018-06-17:
Thanks for your help :) Working perfectly! Don't worry about making more permanent changes. I am happy with this solution and since it's only me, it's probably not worth making them.

tatokis, 2018-06-27:
https://github.com/ckb-next/ckb-next/pull/195
Hopefully this helps in the next release