ckb-next/ckb-next #186: K70 LUX RGB and M65 Pro RGB are both bricked after macOS Mojave update

robertpan226, 2018-06-05:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash

```

Upload `ckb-next-dev-detect-report.gz` with this issue.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2070584/ckb-next-dev-detect-report.gz)

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?
## What would you like it to do instead?
## Any other notes & comments?

---

# Bug report

## General information
Updated macOS from High Sierra to Mojava (10.14). Both my K70 LUX RGB and M65 Pro RGB are detected by ckb-next however, both don't function like normal. The K70 works only during logging in after a restart, and after that, no inputs are processed. The M65's cursor moves but left + right click and scroll wheel all don't work.

### Source

Using cbk-next 0.3.0-13-g53dd0c4 from .dmg file from one of the responses posted in the issues section

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?
https://guides.github.com/features/mastering-markdown/
## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

tatokis, 2018-06-05:
I am going to work on this after I am done with my exams.

Until then, you can try installing this version https://github.com/ckb-next/ckb-next/issues/162#issuecomment-387854019 and hope it works.

Make sure to allow this software (possibly twice) in the security/privacy settings.
Once for ckb-next itself, and once for `Takayama Fumihiko`.

Do reboot afterwards too.

tatokis, 2018-06-05:
Actually, I think that version has mouse issues. (Cursor moving when clicking.)
I'll try and get a more up to date build for you soon.

tatokis, 2018-06-05:
Another update. This build doesn't have the issue mentioned above.
http://ckb-next.info/mac/ckb-next_v0.3.0-31-g2ad3d33_10.11-SDK-karabiner.dmg

robertpan226, 2018-06-05:
Ok, it seems to work now with your previous comment. Side buttons for the M65 still don't work right?

ZirconiumX, 2018-06-05:
Side buttons for the M65 should work fine; I have and use one.

I also took the liberty of renaming the title, because CKB isn't bricking your devices at all. It's not actually possible to brick Corsair devices; the firmware has a fallback bootloader that mounts itself as a USB stick that you can write the firmware images to. Trust me, we've unintentionally broke the firmware several times, and the device has never been unrecoverably broken.

tatokis, 2018-06-05:
I should add that I never tested the buttons with the above build, mostly because I don't think they did anything in general.
Can you give me an example of the buttons being used so that I can fix them?

jonbloom, 2018-06-05:
I'm experiencing the same issue. I tried installing the build from [this](https://github.com/ckb-next/ckb-next/issues/186#issuecomment-394566227) but it seems that the keyboard still doesn't respond. 

It's also showing ckb-next v0.2.9 in the window header but a 
**Warning: ** Driver version mismatch (0.3.0-31-g2ad3d33). Please upgrade ckb. If the problem persists, try rebooting.

Is there something else I have to do to update besides running the package in the above comment?

ZirconiumX, 2018-06-05:
You need to reboot. The driver part is loaded on startup.

jonbloom, 2018-06-05:
I did reboot. The "shimmer" effect seems to be frozen on my keyboard, and the keyboard is unresponsive. I'm going to pull ckb completely out of my system and reinstall from that package, reboot, and report back. 

I'm using a K95 Platinum, by the way.

ZirconiumX, 2018-06-05:
Did you enable the kext? If not, that would explain it.

tatokis, 2018-06-05:
Use the commands in the first one to Uninstall.
https://github.com/ckb-next/ckb-next/wiki/macOS-Installation#uninstallation

Also, again, make sure that you have allowed the software in privacy and settings after the installation.

jonbloom, 2018-06-05:
I used the uninstall commands to uninstall, rebooted, installed the build from the comment, rebooted, and all is working now. Thanks so much!

robertpan226, 2018-06-05:
How did you map commands to your side buttons?

tatokis, 2018-06-05:
I didn't. They work out of the box on linux, and I do not use macOS.
They don't do absolutely anything on macOS from what I can see, so I am asking you how you have them set up so that I can test them.

robertpan226, 2018-06-05:
[image]

Pressing the button won't launch Mission Control.

tatokis, 2018-06-08:
Can you temporarily rebind that button to something else? Perhaps a keyboard key, and see if it works then?

robertpan226, 2018-06-08:
Mapping that button to a keyboard key works.

ZirconiumX, 2018-06-08:
I shall celebrate this by creating a new tag.

ZirconiumX, 2018-06-08:
OK to close?

robertpan226, 2018-06-08:
Yup.

tatokis, 2018-06-08:
Leave this open for others that updated to 10.14 to find the links until we get the karabiner release out.
Also, can you open another application perhaps? Could be that apple broke the way the GUI runs apps.

Xalior, 2018-06-12:
Yay, and thanks, Installing build31 fixed my hung input bugs on Sierra too, both on resume and on hotplug.

But did leave me really REALLY confused why my Karabina custom keybindings were not working any more (default: swap \ and `  ...  welcome to the UK ... have a lovely Brexit ;-) )

After @ZirconiumX 's comments on https://github.com/ckb-next/ckb-next/issues/184#issuecomment-396516158 I tried uninstalling Karabina Elements, and switching my rebinds to the Binding tab in ckb, and that fixed everything.

So, whilst I don't actually need Karabina Elements any more I did find it terribly handy to keep Karabina EventViewer around during the experimental setup and keybind phase. 

Potentially having that included in the 0.3.1 cbk release in some way would be useful, and save you writing your own EventSpy? ;-)

-Dx

tatokis, 2018-06-12:
@Xalior ckb-next has always conflicted with karabiner one way or another, as it tries to get exclusive control over the devices like ckb-next does. Only difference is we now use its kernel driver to send input to the OS.

We need to find a way to prevent such conflicts, if not alert the user, or maybe code in an option to karabiner to exclude corsair devices when ckb-next is present.
There are a few more important issues at the moment though, so you'll need to wait.

As a sidenode, if you delete the config file that macOS uses to store keyboard layout preferences and reboot, you should be able to pick the right layout for your keyboard if that's being an issue.

ZirconiumX, 2018-06-12:
On Tue, 12 Jun 2018, 12:13 D. Rimron-Soutter, <notifications@github.com>
wrote:

> (default: swap \ and ` ... welcome to the UK ... have a lovely Brexit ;-) )
>

We can commiserate together.

>

kfirufk, 2018-06-19:
when is the fix from http://ckb-next.info/mac/ckb-next_v0.3.0-31-g2ad3d33_10.11-SDK-karabiner.dmg will be merged into master ? :)

tatokis, 2018-07-04:
@kfirufk Soon hopefully. A few minor issues remain, but for the most part it's okay. Need to also integrate the driver installation and handling on the build system itself.

joeysugay, 2018-07-20:
Thanks for the OP.  After doing a clean install and security accept and reboot, my K70 RGB is fully working back to life again.  thanks

I just bought a AZIO MAC keyboard out of frustration since mojave beta 1 to 4,  then this came.   So thanks to this group of people wo created this patch !!!  cheers

closed 2018-08-16 by tatokis