ckb-next/ckb-next #187: K68 NRGB Brightness, Windows Lock and Mute LEDs doesn't work

mauriciofauth, 2018-06-12:
I upgraded to the last commit (b646691dbdcd547bbdf80df778fbbc9d4cd1599f) and noticed that the volume up and volume down LEDs are now working.
However, the other LEDs on the other buttons on the first row still doesn't work.

The Brightness LED is always on, the Windows Lock and Mute LEDs are always off.
The volume up and volume down LEDs are working as expected.

ZirconiumX, 2018-06-12:
Can you hop on IRC, please?

mauriciofauth, 2018-06-12:
Sure!

ZirconiumX, 2018-06-13:
Can you also test it on firmware v1, to make sure the fix works on the old firmware too?

As context for anybody else reading this, it turns out that the K68 uses a new bit of protocol just for turning the winlock key on and off.

mauriciofauth, 2018-06-13:
@ZirconiumX The Windows Lock LED is also working on firmware v1. Thanks!

The Brightness and Mute LEDs still need to be fixed however.

mauriciofauth, 2018-06-15:
> I believe it's reliant on the current Pulseaudio status. If pulse isn't
> running, it won't do anything.

I use Linux Mint 18, and I didn't change anything related to Pulseaudio.

> I would have to investigate later (i.e. tomorrow), but if memory serves me
> correctly, the k68 uses the Japanese Ro LED code for its brightness, which
> will require you testing a keymap patch.

Only the LED brightness of the Brightness button does not work, the LEDs of the other keys work.

mauriciofauth, 2018-06-18:
@tatokis nothing returns from ```ldd `which ckb-next` | grep pulse```

mauriciofauth, 2018-06-18:
Okay, I installed the libpulse-dev package (it's been a while since I installed the packages from the dependency list. Could not the build system have identified this missing dependency?), and now the Mute key works as you described it, although I find this behavior odd and prefer it disabled.

tatokis, 2018-06-18:
>Could not the build system have identified this missing dependency?

```
-- Could NOT find PulseAudioSimple (missing: PULSEAUDIOSIMPLE_LIBRARY PULSEAUDIOSIMPLE_INCLUDE_DIR) 
-- Could NOT find PulseAudio (missing: PULSEAUDIO_LIBRARY PULSEAUDIO_INCLUDE_DIR) 
CMake Warning at src/animations/mviz/CMakeLists.txt:30 (message):
  Missing PulseAudio Simple API.  Not building mviz.

-- Could NOT find PulseAudioSimple (missing: PULSEAUDIOSIMPLE_LIBRARY PULSEAUDIOSIMPLE_INCLUDE_DIR) 
-- Could NOT find PulseAudio (missing: PULSEAUDIO_LIBRARY PULSEAUDIO_INCLUDE_DIR) 
```

ZirconiumX, 2018-06-23:
So, mute works, winlock works, now just the brightness LED, right?

mauriciofauth, 2018-07-04:
@tatokis If the _libpulse-dev_ package is optional, the GUI should not assume that it was compiled with this library, shouldn't it? For example, the 'Indicate mute' checkbox should be disabled.
Anyway, this is beyond the scope of this issue.

closed 2018-10-09 by ZirconiumX