ckb-next/ckb-next #188: Trouble building 0.3.0 on CentOS

hevanaa, 2018-06-16:
I'm trying to build latest release 0.3.0 on CentOS. I'm trying to create an rpm. This was no problem with 0.2.9. And building in Fedora versions 26-28 works fine with 0.3.0 after some fiddling with cmake (repo at https://copr.fedorainfracloud.org/coprs/johanh/ckb/).

CentOS needs cmake3 (/usr/bin/cmake3) instead of cmake (which is version 2 and too old).

I'm using the following commands:

cmake3 -H. -Bbuild -DCMAKE_BUILD_TYPE=Release -DSAFE_INSTALL=OFF -DSAFE_UNINSTALL=OFF
cmake3 --build build --target all -- -j %{?_smp_mflags}

(then %make_build in rpm spec)

Everything else works fine, but "build/lib/libKissFFT.so" is missing from the output. Build log says:

...
Scanning dependencies of target KissFFT
Scanning dependencies of target quazip_autogen
Scanning dependencies of target ckb-next-daemon
Scanning dependencies of target gradient
[  1%] Automatic MOC and UIC for target quazip
[  1%] Building C object src/libs/kissfft/CMakeFiles/KissFFT.dir/kiss_fft.c.o
[  2%] Building C object src/animations/CMakeFiles/gradient.dir/gradient/main.c.o
[  3%] Building C object src/daemon/CMakeFiles/ckb-next-daemon.dir/command.c.o
[  3%] Built target quazip_autogen
Scanning dependencies of target heat
[  4%] Building C object src/libs/kissfft/CMakeFiles/KissFFT.dir/kiss_fftr.c.o
[  5%] Linking C executable ../../bin/gradient
...
...
[  2%] Built target KissFFT

The only files found relating to KissFFT are: 
../BUILD/ckb-next-0.3.0/build/src/libs/kissfft/libKissFFT.a
../BUILD/ckb-next-0.3.0/build/src/libs/kissfft/CMakeFiles/KissFFT.dir

Any hints how to figure out what goes wrong? I'm not familiar with cmake...

ZirconiumX, 2018-06-16:
Actually that's to be expected; we build kissfft as a static library (the
.a file), not a dynamic library (which would be the .so you're expecting).

hevanaa, 2018-06-16:
Thanks for the explanation. I wonder why there is a libKissFFT.so library when I build on Fedora then? I use the same build arguments.

hevanaa, 2018-06-16:
And on Fedora, unless I included the libKissFFT.so file in the rpm and installed it, the rpm couldn't be installed (because rpm detected a dependency on libKissFFT.so that wasn't available).

ZirconiumX, 2018-06-16:
In a nutshell: ¯\_(ツ)_/¯

hevanaa, 2018-06-16:
Well, guess I have to add yet another %if 0%{?rhel} / %if 0%{?fedora} to the rpm spec.

ZirconiumX, 2018-06-16:
I add devices to the daemon, not deal with packaging issues. That used to
be l2y's job, but he left and so we're very short on people.

closed 2018-06-16 by hevanaa