ckb-next/ckb-next #196: Strafe/K70 MK.2

ZirconiumX, 2018-06-23:
We caught these well in advance, and we have the USB IDs already, but it seems these are [finally rolling out to market](https://www.kitguru.net/peripherals/briony-hannam/corsair-strafe-rgb-mk-2-keyboard-review/).

With that in mind:
- The Strafe MK.2 can probably use the existing Strafe GUI image.
- The K70 MK.2 will need a new GUI config, or possibly we can hack one as needed.

And, as always: **we need testers**.

drahenshaw, 2018-06-27:
As someone who is interested in using the new MK.2 for Mac, what is the best way to help / test for these projects?

tatokis, 2018-06-27:
You can start off by running ckb-next-dev-detect, attaching the report, and joining us on our IRC room.

platypython, 2018-07-27:
I have the K70 MK.2. Attached is a report.
[ckb-next-report-1532721718.gz](https://github.com/ckb-next/ckb-next/files/2237094/ckb-next-report-1532721718.gz)

tatokis, 2018-07-27:
@platypython Can you also run USB Prober and send the report? Once you do, I will add support for the device and give you code to test.

You can follow this. https://superuser.com/questions/781982/how-can-i-install-usb-prober-from-the-developer-sdk-on-mac-os-x#814339

As a sidenote, do NOT install any of the kexts included in usb prober as that will ruin your system.

tatokis, 2018-07-28:
I have added support for the K70 RGB MK2 (and possibly the SE) in the mk2 branch.
It would be appreciated if someone could test it and make sure both logo lights function normally.

ZirconiumX, 2018-07-30:
@k70rgbmk2 conveniently happens to have just that keyboard. Now we just need to check a Strafe.

platypython, 2018-08-02:
I see the two new logo lights in the keyboard layout, but they aren't working for me to toggle.

tatokis, 2018-08-02:
Is this from the master branch?

platypython, 2018-08-02:
mk2 branch

tatokis, 2018-08-02:
Try creating a new profile in the GUI

tatokis, 2018-08-02:
Do switch to master too, mk2 is out of date.

platypython, 2018-08-02:
All fixed. Works great, K70 RGB MK2 fully functional.

tatokis, 2018-08-02:
CC @k70rgbmk2. Logo should be working now. You probably need to create new profiles. Sorry

Wisssard, 2018-08-04:
I have a corsair strafe mk 2. Here's the file
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2259805/ckb-next-dev-detect-report.gz)

ZirconiumX, 2018-08-04:
@Wisssard Can you try again with the `mk2` branch?

Wisssard, 2018-08-04:
still dosn't work

[ckb-next-report-1533419486.gz](https://github.com/ckb-next/ckb-next/files/2259842/ckb-next-report-1533419486.gz)

ZirconiumX, 2018-08-04:
Actually, despite what you might think, it is almost working:
```
Aug 04 23:50:38.724265 wiz-pc ckb-next-daemon[28233]:     ckb-next: Corsair RGB driver 0.3.1-unknown
Aug 04 23:50:38.724515 wiz-pc ckb-next-daemon[28233]: [I] Root controller ready at /dev/input/ckb0
Aug 04 23:50:38.829761 wiz-pc ckb-next-daemon[28233]: [I] Connecting CORSAIR STRAFE RGB MK.2 Mechanical Gaming Keyboard at /dev/input/ckb1
Aug 04 23:50:38.989685 wiz-pc ckb-next-daemon[28233]: [I] Starting input thread for /dev/input/ckb1
Aug 04 23:50:39.105300 wiz-pc ckb-next-daemon[28233]: [E] getfwversion (firmware.c:27): Bad input header
Aug 04 23:50:39.105300 wiz-pc ckb-next-daemon[28233]: [W] _start_dev (device.c:43): Unable to load firmware version/poll rate
Aug 04 23:50:39.105389 wiz-pc ckb-next-daemon[28233]: [I] Setup finished for /dev/input/ckb1
```
But the GUI doesn't currently have a layout for the Strafe Mk.2, so the GUI comes up as empty.

I think I need to add it to `IS_V3_OVERRIDE` for the daemon to work properly, which is a five-minute fix.

Wisssard, 2018-08-04:
cool thanks

Wisssard, 2018-08-04:
[image]
Is it supposed to be like this?

ZirconiumX, 2018-08-04:
It will be like that for the moment. The Strafe Mk.2 needs its own unique
GUI layout to use all of the lights. As a compromise, if you run `sed -i
"s/strafe_mk2/k70mk2/" /dev/input/ckb1/features` then the GUI will show the
keyboard as a K70 Mk.2, which is close enough for now.

Would you mind taking another ckb-next-dev-detect?

Wisssard, 2018-08-04:
i did the command and after trying to change the lightning it no longer responds to key presses.

Wisssard, 2018-08-04:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2259875/ckb-next-dev-detect-report.gz)

ZirconiumX, 2018-08-05:
So, it appears that Corsair are somehow seriously fucking over CKB. This is
going to require you to come join us in IRC, because something is seriously
wrong here.

ZirconiumX, 2018-08-05:
> How do i join the irc?

At the top of the readme is a blue button that says "#irc: freenode". Click
it and it'll put you on a Web client that connects to our IRC channel. Note
there's a short delay of about 30 seconds before you can talk because we've
been under a spambot attack.

Wisssard, 2018-08-05:
ok im in

tatokis, 2018-08-12:
Sure!

closed 2018-08-12 by tatokis