ckb-next/ckb-next #204: sabre mouse

deanbolt, 2018-07-03:
i downloaded latest version and i played around with the settings now and my left click will only work sometimes, won't switch tabs etc how do i hard reset it back to the way it came out the box, i reset everything on ckb-next and it won't fix ive tried unplugging plugging back in with both clicks pushed in and still won't fix it. please help

ZirconiumX, 2018-07-03:
Did it do this before you installed ckb?

ZirconiumX, 2018-07-03:
Also, what OS are you on?

deanbolt, 2018-07-03:
high seria and no it didn't do it before hand

deanbolt, 2018-07-03:
ive unistalled ckb and its still doing it.

ZirconiumX, 2018-07-03:
Did you reboot?

tatokis, 2018-07-03:
Try installing this version, make sure you **allow system software**  in Security and Privacy **after** installation, and then reboot.

http://ckb-next.info/mac/ckb-next_v0.3.0-62-g68fedda_10.11-SDK.dmg

deanbolt, 2018-07-03:
ok ill try that now

deanbolt, 2018-07-03:
thank u so much tatokis my mouse seems to be working now, i will try and set colors etc later and ill let u know

deanbolt, 2018-07-03:
my mouse is currently all black, does save to hardware push the change?

tatokis, 2018-07-03:
Save to hardware is for when ckb-next isn't running. All changes should be reflected instantly without saving.
Assuming brightness is set to 100%, you might want to run 
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```
in a terminal and upload the report to see why it's not working.

deanbolt, 2018-07-03:
okay ive set the colors we looking good!

tatokis, 2018-07-03:
Glad to hear it

closed 2018-07-03 by tatokis