ckb-next/ckb-next #205: [Feature] Allow animation binding simultaneously with another binding

HactarCE, 2018-07-08:
# Feature request

## What is the program's current behavior?

I would like to make a single letter key cause a ripple effect throughout a larger portion of the keyboard using the "Animation" tab under the "Binding" tab. Currently, configuring the effect this way disables the normal operation of the key.

## What would you like it to do instead?

Make "Start animation on keypress" a checkbox, allowing it to be used simultaneously with any other binding.

## Any other notes & comments?

If this is implemented, it might be more intuitive to move this setting out of a tab, since it wouldn't be a radio button like those contained by the other tabs.

"Switch to mode" may also be useful simultaneously with other keys, though I can't think of a good example.

owendelong, 2020-05-29:
I would very much like to have this feature as well. One of the things I am hoping to do is create an effect where pressing a key causes the key to change color while pressed and then fade to the original "background" color over a bout a 0.25 second decay time. Ideally, I want the surrounding keys to "dim out" during this same time period so as to make it obvious to an observer which key was received by the system.

owendelong, 2020-05-29:
I realize after playing with the UI a little more that this feature is already present, but it's not on the binding page. It can be accomplished by adding animations on the bottom of the lighting page. Took some experimentation to find exactly what I was looking for and discover start with mode vs. start with key press as well as the interactions of multiple animations, but once I figured it out, it was quite functional.

owendelong, 2020-05-29:
As far as I am concerned, this request can be closed as implemented.

ecen, 2022-05-15:
I am interested in this feature as well. I do not consider it already implemented since it does not seem possible to start an animation on a set of keys by pressing a key that is not included in that set.