ckb-next/ckb-next #208: k95w ` and ~ turned to § and ±

xuguangyuansh, 2018-07-16:
k95w use ckb-next v0.3.0，the keyboard ` and ~ turned to § and ±，there is no problem when k95w on windows 10.

royalchan-okta, 2018-07-16:
I get this too, for some reason the automatic layout detection is wrong and sets the Layout to European on OSX on startup regardless of what you set it to. It also does not respect the layout from imported profiles.

You can manually fix this every time you start the app by manually changing the Layout under Settings->Binding.

ZirconiumX, 2018-07-16:
Unfortunately, automatic layout detection sometimes doesn't work on Mac
because we can't read from the keyboard.

xuguangyuansh, 2018-07-17:
@royalchan-okta Thank you very much, use your method, it has been solved. In this way, the bug does not need to be fixed, the k95w is only used temporarily, the rest of time is spent on k95 RGB (18g), and the k95 RGB with ckb-next v0.3.0 is perfect.

xuguangyuansh, 2018-07-17:
@ZirconiumX Thank you for your reply, the bug does not need to be fixed, the k95w is only used temporarily, the rest of time is spent on k95 RGB (18g), and the k95 RGB with ckb-next v0.3.0 is perfect.

It's really a pleasure to  working on Mac, now I am doing block chain. Later I will also study Qt and   join into the ckb-next project.

closed 2018-07-17 by xuguangyuansh