ckb-next/ckb-next #209: [MAC] cmake error, failed to install

snul2, 2018-07-20:
I get an error when installing the ckb-next software on my mac. I have tried it with all different installation methods from the installation guide for mac and get the same error every time. 

The error:  
command: cmake -H. -Bbuild -DCMAKE_BUILD_TYPE=Release -DSAFE_INSTALL=ON -DSAFE_UNINSTALL=ON
-- ckb-next version: 0.3.0-58-g87aca21 (Non-release)
-- CoreFoundation found: /System/Library/Frameworks/CoreFoundation.framework
-- CoreGraphics found:   /System/Library/Frameworks/CoreGraphics.framework
-- IOKit found:          /System/Library/Frameworks/IOKit.framework
-- Searching for running ckb and/or ckb-next GUI
-- NOTE:
    Privileged access is required for operations upon the daemon at
    configure time as it is owned by root. If this is unacceptable,
    disable SAFE_INSTALL and prepare the filesystem manually.
-- launchd detected
-- Importing org.ckb-next.daemon.plist (launchd)
qt 5.11.1
quazip 0.7.3_1
-- Foundation found:   /System/Library/Frameworks/Foundation.framework
-- CoreAudio found:    /System/Library/Frameworks/CoreAudio.framework
-- AudioToolbox found: /System/Library/Frameworks/AudioToolbox.framework
-- Generating and importing org.ckb-next.info.plist
-- Importing org.ckb-next.plist
CMake Error at src/gui/CMakeLists.txt:488 (if):
  if given arguments:

    "Qt5Core_VERSION" "VERSION_GREATER_EQUAL" "5.10.0"

  Unknown arguments specified

-- Configuring incomplete, errors occurred!

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2215547/ckb-next-dev-detect-report.gz)
[CMakeError.log](https://github.com/ckb-next/ckb-next/files/2215548/CMakeError.log)
[CMakeOutput.log](https://github.com/ckb-next/ckb-next/files/2215549/CMakeOutput.log)

ZirconiumX, 2018-07-24:
Hi, sorry for the delay in responding.

Can you try the `cmake-fix` branch, which should solve this issue?

closed 2018-07-29 by snul2

tatokis, 2018-07-29:
Is this fixed, then? Should we merge the branch?

snul2, 2018-07-29:
Yeah but then I had issues with macOS Mojave (https://github.com/ckb-next/ckb-next/issues/186). I reinstalled everything but without the cmake-fix branch and everything worked fine. 
So I really do not know if the cmake-fix branch solved the issue...