ckb-next/ckb-next #210: K65 rapidfire doesnt function with driver installed.

holytoast, 2018-07-24:
# Bug report

## General information

### Source

# Something else
The keyboard instantly stops working when you install this driver on my macbook pro 13" 2017 touchbar even with reboots replugs ect the LEDs do work however and i can change them then save them to the kyboards memory then uninstall the software and then the keyboard works again just fine.

ZirconiumX, 2018-07-24:
# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

Can you please run this? It's part of the bug script.

Oschly, 2018-08-31:
I have a similar problem but with K70 RGB Rapidfire (mx Speed) on Mojave so I've uninstalled ckb-next, but I put here this report.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2341815/ckb-next-dev-detect-report.gz)

tatokis, 2018-09-06:
@Oschly Install the latest version and generate a report while the problem is happening. You might have to use another keyboard to do it.

closed 2022-03-05 by tatokis