ckb-next/ckb-next #212: link not created in /dev/input/by-id

m5x, 2018-07-30:
For evdev passthrough purposes it would be very handy if link to the keyboard device got created in /dev/input/by-id as usual for input devices. I am on Solus system where it does not get created, should I report this to Solus package maintainers or ckb-next does not do it yet?

tatokis, 2018-07-30:
Apparently I never linked this to the wiki, but https://github.com/ckb-next/ckb-next/issues/123#issuecomment-399744172

This should get you going. That said, I'm not sure what's responsible for creating these symlinks, but it's clearly not creating symlinks for uinput devices.
I'm not sure there's much we can do about it.

m5x, 2018-07-30:
I thought it gets created by udev rules file, usually called something like `60-persistent-input.rules`. There is already rules file in ckb-next repository, maybe the link creation could be added to it?

tatokis, 2018-07-31:
udev doesn't have enough data to create an identifiable by-path or a by-id symlink.

```
$ udevadm info /dev/input/event9
P: /devices/virtual/input/input469/event9
N: input/event9
E: BACKSPACE=guess
E: DEVNAME=/dev/input/event9
E: DEVPATH=/devices/virtual/input/input469/event9
E: ID_INPUT=1
E: ID_INPUT_KEY=1
E: ID_INPUT_KEYBOARD=1
E: ID_INPUT_MOUSE=1
E: ID_SERIAL=noserial
E: MAJOR=13
E: MINOR=73
E: SUBSYSTEM=input
E: USEC_INITIALIZED=349442600887
E: XKBLAYOUT=us,gr
E: XKBMODEL=pc105
E: XKBOPTIONS=grp:alt_shift_toggle,grp_led:scroll
E: XKBVARIANT=,

$ udevadm info --attribute-walk /dev/input/event9

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/virtual/input/input469/event9':
    KERNEL=="event9"
    SUBSYSTEM=="input"
    DRIVER==""

  looking at parent device '/devices/virtual/input/input469':
    KERNELS=="input469"
    SUBSYSTEMS=="input"
    DRIVERS==""
    ATTRS{name}=="ckb1: Corsair K70 RGB Gaming Keyboard"
    ATTRS{phys}==""
    ATTRS{properties}=="0"
    ATTRS{uniq}==""
```
Looks like it would be typically created by the following rule
```
SYMLINK+="input/by-id/$env{ID_BUS}-$env{ID_SERIAL}-event-$env{.INPUT_CLASS}"
```
but ID_BUS and ID_SERIAL are empty.

Best we can do is use the name attribute to create a symlink in `/dev/input/`

Replace the contents of `/etc/udev/rules.d/99-ckb-daemon.rules` with
```
ACTION=="add|change", SUBSYSTEM=="input", ATTRS{name}=="ckb[0-9]: [A-Za-z0-9]*", ENV{ID_INPUT_JOYSTICK}="", SYMLINK+="input/$attr{name}-event-$env{.INPUT_CLASS}" 
```
Run `sudo udevadm trigger` and you should now have similar symlinks in `/dev/input/`
```
drwxr-xr-x  2 root root     100 Jul 30 20:57 ckb0
drwxr-xr-x  2 root root     220 Jul 30 21:00 ckb1
lrwxrwxrwx  1 root root       6 Jul 31 03:58 ckb1:_Corsair_K70_RGB_Gaming_Keyboard-event-kbd -> event6
lrwxrwxrwx  1 root root       6 Jul 31 03:58 ckb1:_Corsair_K70_RGB_Gaming_Keyboard-event-mouse -> event9
drwxr-xr-x  2 root root     220 Jul 30 21:00 ckb2
lrwxrwxrwx  1 root root       6 Jul 31 03:58 ckb2:_Corsair_Gaming_M65_Pro_RGB_Mouse-event-kbd -> event4
lrwxrwxrwx  1 root root       6 Jul 31 03:58 ckb2:_Corsair_Gaming_M65_Pro_RGB_Mouse-event-mouse -> event5
```

If you think that's satisfactory, I will go ahead and commit the change.

m5x, 2018-07-31:
Really nice and fast! :+1: 

It was my understanding that the links in `/dev/input/by-id` get created to be able to use device paths that does not change between boots. The links there seem to follow some naming rule so it would be great if it was possible to mimic it as closely as possible but logically it seems to belong there more than to `/dev/input`.

I am no bash guru, would it be possible to modify the rule further to parse the name attribute value to come up with names like this?

```
lrwxrwxrwx  1 root root       6 Jul 31 03:58 usb-Corsair_K70_RGB_Gaming_Keyboard-ckb1-event-kbd -> event6
lrwxrwxrwx  1 root root       6 Jul 31 03:58 usb-Corsair_K70_RGB_Gaming_Keyboard-ckb1-event-mouse -> event9
lrwxrwxrwx  1 root root       6 Jul 31 03:58 usb-Corsair_Gaming_M65_Pro_RGB_Mouse-ckb2-event-kbd -> event4
lrwxrwxrwx  1 root root       6 Jul 31 03:58 usb-Corsair_Gaming_M65_Pro_RGB_Mouse-ckb2-event-mouse -> event5
```
The `usb` prefix would have to be hard-coded but if `ckb` supports only usb devices it should  always be correct. I hope that does not break some Linux rule about naming links in `/dev/input/by-id`. I briefly searched for some documentation but did not find any.

tatokis, 2018-08-03:
@m5x any more customization requires launching either bash from udev, or making the daemon itself return a formatted string.

The latter is difficult and the former brings in more dependencies.
I'm not sure which one we should choose.
Another issue I can see with this is ckbX are not always the same.

Sometimes the daemon might recognise another device before another one, so they ckbX id changes.
If anything, if we want it to be "permanent", we need to edit the id out.

urmamasllama, 2018-08-03:
I have a tentative rules file that "works" it does give a unique consistent symlink for both parts of each ckb device however, its also creating symlinks for some non ckb devices as well I'll post it hopefully tonight.

urmamasllama, 2018-08-03:
In regards to my previous statement I have an important question that would help to make it look cleaner and possibly fix the unwanted symlinks. Is the ev value for the mouse and keyboard devices in ckb consistently 7 and 120003 respectively? if it is that will help a bunch so far I know that this is true for the m65, scimitar pro, k95w, and the strafe.

urmamasllama, 2018-08-03:
a bunch of other stuff should probably get fixed so that this isn't needed but this should work for now
`# Create base ckb-next devices
ACTION=="add|change", SUBSYSTEM=="input", ATTRS{name}=="ckb[0-9]: [A-Za-z0-9]*", ENV{ID_INPUT_JOYSTICK}=""

# If the device isn't an event input device with a recognised
# class, do nothing
KERNEL!="event*", GOTO="input_ckb_ids_end"
SUBSYSTEM!="input", GOTO="input_ckb_ids_end"
ENV{.INPUT_CLASS}!="?*", GOTO="input_ckb_ids_end"

# Store name in .INPUT_NAME, replace spaces with underscores
PROGRAM="/usr/bin/sed 's/ /_/g;s/.//6;s/./-/5;s/.//4' /sys/class/input/%k/device/name", ENV{.INPUT_NAME}="%c"
# Separate into both emulated devices
PROGRAM="/usr/bin/cat /sys/class/input/%k/device/capabilities/ev", ENV{.INPUT_EMUNUM}="%c"
# Filter to only match ckb-next supported devices
ENV{.INPUT_NAME}!="ckb*", GOTO="input_ckb_ids_end"

# Add a by-id symlink, with class to separate type:
SYMLINK+="/input/by-id/event-%E{.INPUT_NAME}-%E{.INPUT_EMUNUM}"

# Add a systemd .device service unit:
TAG+="systemd"
# Alternatively we can add additional paths for the .device units
# ENV{SYSTEMD_ALIAS}+="/input/by-name/%E{.INPUT_NAME}"

# Exit
LABEL="input_ckb_ids_end"
`

tatokis, 2018-08-05:
Right, since there's a PR open now, there's no need for this issue to remain open, as the PR will get merged eventually.

Anyone interested please track #217.

closed 2018-08-05 by tatokis