ckb-next/ckb-next #215: K55 wake from sleep issue  | Arch Linux - ckb-next / ckb-next-git (AUR)

RoroTiti, 2018-08-03:
Hello,

First of all, thank you very much for this awesome software. It's really nice to see people porting Corsair feature on Linux :+1: 

**Here is my problem :** 

I recently installed Arch Linux with GNOME on my computer (but don't worry, I'm not a completely newbie haha, I installed it for development purposes) and I noticed that my K55 keyboard doesn't respond when my computer wake from sleep. My Sabre mouse works perfectly, but not my K55.

_I notice that the lightning profile is executed on GDM screen, but can't use the keyboard._

To help, I installed CKB-next on Manjaro 17.11.1 and Fedora 28 as well, same behavior. So I think it's software related more than specific installation related. I also tested with USBHID Quirks and/or 
USBCore Quirks, no more luck.

If you need any other information, such as logs, feel free to ask them to me, I would be very happy to help you solving this issue.

I join to this report the ckb-next-dev-detect-report.gz, as asked.

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2256816/ckb-next-dev-detect-report.gz)

ZirconiumX, 2018-08-03:
As a temporary measure, could you test with the usbcore quirks b and m
(separately) to see if you can reproduce the issue with those? It's
entirely possible that it is a software issue, but we've had a lot of
problems with suspend on both Mac and Linux because it's hard to detect.

RoroTiti, 2018-08-03:
Okay, so finally, I made some tests... Here are the results : 

**USBCore Quirks** only : not working
**USBHID Quirks 0x20000408** : not working
**USBHID Quirks 0x4** : not working
**USBCore Quirks** and **USBHID Quirks 0x20000408** : not working
**USBCore Quirks** and **USBHID Quirks 0x4** : not working

So none of the kernel parameters have effect on issue... 

By the way, I notice that the keyboard works properly when CKB-next is closed/off.

Feel free to ask me other tests to do, I would be happy to help you debugging this issue !

ZirconiumX, 2018-08-03:
Which USBCore quirks?

RoroTiti, 2018-08-03:
These ones : https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#usbcore-quirks

tatokis, 2018-08-03:
As a workaround, try just the usbcore "b" quirk and see if it helps.

Usually I'd say to try echo idle > /dev/input/ckb*/cmd followed by echo idle > /dev/input/ckb*/cmd
but if input is broken, we are probably dealing with something else.

Somehow the logs also don't indicate anything.

RoroTiti, 2018-08-03:
Okay, here is the message I obtain : `bash: /dev/input/ckb*/cmd : ambiguous redirection`

Tested either as normal user and as root.

tatokis, 2018-08-03:
@RoroTiti sorry I should've specified. Replace the `*` with 1 in your case.
From your logs,
```
Aug 03 13:18:03.459881 archlinux ckb-next-daemon[335]: [I] Connecting Corsair Gaming K55 RGB Keyboard at /dev/input/ckb1
```

RoroTiti, 2018-08-03:
Okay, so I have to execute these commands when my computer wake from sleep, isn't it ?

tatokis, 2018-08-03:
Yeah, try it when the keyboard is in a non working state and see if it helps.

RoroTiti, 2018-08-03:
Okay, let me try to create a bash script, thank you very much for your help !

RoroTiti, 2018-08-03:
I did a systemd script executable at wake from sleep, but sadly, it doesn't help with the issue... Keyboard still unusable at wake from sleep...

RoroTiti, 2018-08-03:
OKAY, changed ckb1 to ckb2 (since the number change among reboots...) and now the keyboard works at wake from sleep ! Thank you very much for you help !

You can now include a fix into CKB-next since now we have found the origin of the issue.

tatokis, 2018-08-03:
Sadly it's not as easy as that. We need a way of knowing when the system resumes from suspend, and as far as I am aware there is no linux kernel userspace api for it.

It's why I suggested the "b" usbcore quirk.
```
b = USB_QUIRK_RESET_RESUME (device can't resume
	correctly so reset it instead);
```
That would make the daemon reinitialise the device on resume.

tatokis, 2018-08-03:
Only way I can think of detect the device is in hardware mode is if we receive two sequential 0x01-2 HID packets instead of 0x01-2 followed by 0x03

RoroTiti, 2018-08-03:
Ahh OK I see... Of course, my method use systemd hook, but of course, it's not a good solution for everyone since it's not universal... :disappointed:  But at least, as a workaround for me, it's working ! :smile:

RoroTiti, 2018-08-03:
The trick would be to listen to USB events... But I don't have the ability to help you on this side, sadly :stuck_out_tongue_closed_eyes: ...

TheYokai, 2019-01-03:
I'm having this issue on my Ubuntu system as well. Here's my system details:

```
 OS: Ubuntu 18.10 cosmic
 Kernel: x86_64 Linux 4.18.0-13-generic
 Uptime: 19h 51m
 Packages: 3117
 Shell: bash 4.4.19
 Resolution: 1920x1080
 DE: GNOME 
 WM: GNOME Shell
 Font: Ubuntu 11
 CPU: Intel Core i7-4770K @ 8x 3.9GHz [27.8°C]
 GPU: GeForce GTX 760
 RAM: 3979MiB / 11945MiB
```

TheYokai, 2019-01-03:
So I went ahead an tried to make a script like the one above, it didn't seem to work. Mine looks like the following:

```
#!/bin/sh
case $1/$2 in
  pre/*)
    echo "Going to $2..."
    ;;
  post/*)
    echo "Waking up from $2..."

    echo idle > /dev/input/ckb1/cmd
    echo active > /dev/input/ckb1/cmd
    ;;
esac
```
Interestingly, my installation only has ckb0 and ckbl1 directories in input. ckb0 has no ```cmd``` entry though.

partyonthelake, 2019-05-08:
> So I went ahead an tried to make a script like the one above, it didn't seem to work. Mine looks like the following:
>
> ```
> #!/bin/sh
> case $1/$2 in
>   pre/*)
>     echo "Going to $2..."
>     ;; [...]

did you ever manage to fix this on ubuntu? im running pop os and having the exact same problem

ZirconiumX, 2019-05-08:
Can you test with the `b` and/or `m` usbcore quirks? This appears to be a Corsair firmware bug we have problems working around.

partyonthelake, 2019-05-08:
thanks for the fast reply. im sorry, how exactly do i do what you're asking? i read the usbcore quirks section of the troubleshooting page, do i just run the command listed there?

ZirconiumX, 2019-05-08:
You put them on your kernel command line; they're not commands.

partyonthelake, 2019-05-08:
i ran the command and the output was "usbcore.quirks=1b1c:1b2d:gn". im guessing i need to change the "gn" to "bm" then add it to my kernel cmdline and reboot to test if the problem persits? is that what you mean by "Can you test with the b and/or m usbcore quirks?" sorry im trying my best to understand here

ZirconiumX, 2019-05-08:
Replace "gn" with "bgmn", and test it.

partyonthelake, 2019-05-09:
i did that, the issue still persists. here's my logs if that helps 
[ckb-next-report-1557359978.gz](https://github.com/ckb-next/ckb-next/files/3159672/ckb-next-report-1557359978.gz)

TheYokai, 2019-05-09:
> did you ever manage to fix this on ubuntu? im running pop os and having the exact same problem

No, I was never able to solve this problem. If you can find out a consistent script location that ubuntu will run on wake up, that should actually let it work. I recently switched to manjaro gnome. 

For the short term, my solution was to simply prevent my computer from sleeping. This is definitely something that should be resolved at some point.

czarnyckm, 2019-05-18:
> OKAY, changed ckb1 to ckb2 (since the number change among reboots...) and now the keyboard works at wake from sleep ! Thank you very much for you help !
> 
> You can now include a fix into CKB-next since now we have found the origin of the issue.
> 
> The final working script (in my case) :
> 
> ```
> #!/bin/sh
> case $1/$2 in
>   pre/*)
>     echo "Going to $2..."
>     ;;
>   post/*)
>     echo "Waking up from $2..."
> 
>     echo idle > /dev/input/ckb1/cmd
>     echo idle > /dev/input/ckb2/cmd
> 
>     echo active > /dev/input/ckb1/cmd
>     echo active > /dev/input/ckb2/cmd
>     ;;
> esac
> ```
> 
> The script is placed in `/usr/lib/systemd/system-sleep/`

Under debian/ubuntu the script should be placed in  /lib/systemd/system-sleep/

partyonthelake, 2019-05-19:
thank you so much, it works now and i dont have to restart ckbnext everytime i resume from sleep now!

czarnyckm, 2019-05-22:
I had still problems with the script above. Now I additional reconnect all usb devices in resume step. This fix all issues with the led effects and keyboard is always after resume working
The final working script (in my case) :
```
#!/bin/sh
case $1/$2 in
  pre/*)
    echo "Going to $2..."
    ;;
  post/*)
    echo "Waking up from $2..."

    for i in /sys/bus/pci/drivers/[uoex]hci_hcd/*:*; do
        [ -e "$i" ] || continue
        echo "${i##*/}" > "${i%/*}/unbind"
        echo "${i##*/}" > "${i%/*}/bind"
    done

    echo idle > /dev/input/ckb1/cmd
    echo idle > /dev/input/ckb2/cmd

    echo active > /dev/input/ckb1/cmd
    echo active > /dev/input/ckb2/cmd

    ;;
esac
```
The script is placed in /usr/lib/systemd/system-sleep/ or under debian/ubuntu the script should be placed in /lib/systemd/system-sleep/

tatokis, 2019-06-05:
This should be fixed in master

closed 2019-06-05 by tatokis

RoroTiti, 2019-06-06:
Glad to see that :D !

czarnyckm, 2019-06-06:
I do have still the problem with the k95 keyboard. First resume the keyboard works.. second resume not responsible. So I use  the script from my previous post again.

tatokis, 2019-06-06:
Generate a report and attach it, after it fails

czarnyckm, 2019-06-06:
> Generate a report and attach it, after it fails
I reconnected keyboard after resume to be able to login.. I hope it is ok
[ckb-next-report-1559854509.gz](https://github.com/ckb-next/ckb-next/files/3263459/ckb-next-report-1559854509.gz)

maetthew, 2019-06-13:
Also periodically having issues after waking from suspend, seems to be random though, sometimes it works but on occasion it doesn't. I've also experienced being able to type in my password to login to the desktop after which the keyboard stops working.

Here's a report from when it didn't work at all after waking from suspend:

[ckb-next-report-1560422242.gz](https://github.com/ckb-next/ckb-next/files/3285507/ckb-next-report-1560422242.gz)

tatokis, 2019-06-19:
This seems like a different issue, but I have no idea what could be causing it...