ckb-next/ckb-next #220: Windows lock and mute key color dosn't change

Wisssard, 2018-08-07:
I've been using my strafe mk.2 for a couple of days and I realized I can't change the color of the mute button or the windows lock key. They show up as black on the ckb-next gui ,but even though i try to change them nothing changes. Switching between profiles causes it to become the correct color on the gui ,but no color is displayed on the keyboard.

https://i.imgur.com/dtxeNr7.jpg
https://i.imgur.com/VicVSwF.png
https://i.imgur.com/3zsdClG.png
https://i.imgur.com/ZBwqVFE.png
 
[ckb-next-report-1533650115.gz](https://github.com/ckb-next/ckb-next/files/2266651/ckb-next-report-1533650115.gz)

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2266630/ckb-next-dev-detect-report.gz)

ZirconiumX, 2018-08-07:
First off, is pulseaudio installed? The mute key requires pulseaudio and uses pulseaudio's mute status.

Wisssard, 2018-08-07:
Yes but i mean the color dosn't change

ZirconiumX, 2018-08-07:
@tatokis also mentioned that the Performance Tab overrides the colour of the Winlock and Mute keys because otherwise it wouldn't display winlock or mute.

Wisssard, 2018-08-07:
oh ok sorry for the trouble

closed 2018-08-07 by Wisssard

ZirconiumX, 2018-08-07:
Don't be. If you enable winlock, does the light show up on your keyboard? If not, that's a bug. Same for mute, if you have pulseaudio installed and enabled.