ckb-next/ckb-next #222: How to run ckb-next on Kali Linux?

takochako, 2018-08-09:
Sorry if this isn't the right place to be asking this. I don't know if issues are for bugs only or bugs and questions. I'm trying to run the ckb-next GUI in Kali Linux, but I'm the root user so it doesn't let me run it. I created a new non-root user, but that doesn't work either. This is the message it spits out when I run `ckb-next`:
`QObject::startTimer: Timers cannot have negative intervals
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-nonroot'
No protocol specified
qt.qpa.screen: QXcbConnection: Could not connect to display :1
Could not connect to any X display.`

Does anyone know how I can fix this? I really want to get the RGB on my STRAFE keyboard working with Linux.

Once again, sorry if I shouldn't be asking questions here.

ZirconiumX, 2018-08-09:
What's happening there is that you have X running - presumably as root -
but you're trying to launch ckb-next from what might be a Linux console.

The immediate hack is `DISPLAY=:0 ckb-next`, where you tell it to display
on your root X session.

The better and safer way is to log out of X as root and then log into your
non-root user and launch X through that.

ZirconiumX, 2018-08-09:
No. Running things as root unnecessarily is a pretty terrible idea for a
wide variety of reasons, and I'd have expected a security-focused distro
such as Kali to not make such a major blunder.

OvermindDL1, 2018-08-09:
> Running things as root unnecessarily is a pretty terrible idea for a
wide variety of reasons

_/me coughs about not using systemd properly to lock down the /dev keyboard/mouse entries to necessary physically logged in users only_

^.^

tatokis, 2018-08-09:
>No. Running things as root unnecessarily is a pretty terrible idea for a wide variety of reasons

Yeah, it definitely is, but it's different when the whole desktop runs as root _by default_. We can't tell Kali devs not to do it. People expect software to work, and this is $bad_experience++;

It's the same as telling mac people to not use mac. It's not happening.

>I'd have expected a security-focused distro such as Kali to not make such a major blunder.

I'd say more pentesting than anything. 

@OvermindDL1 please remain on topic, but to respond to your message, yeah. I totally agree.
Now make sure the implementation works with sysvinit, upstart, runit and openrc.

Once you do that, I'll be okay with it.

OvermindDL1, 2018-08-09:
> Now make sure the implementation works with sysvinit, upstart, runit and openrc.

Should be a simple test to see 'what' it is running on and then set the appropriate flags to run certain code or not.

But yeah, this is my big annoyance, the others logged in there can just read the keyboard inputs freely...  ^.^;

Either way, the `DISPLAY=:0` seems entirely reasonable as it is the forced override of what to choose.

takochako, 2018-08-10:
I get the same error with `DISPLAY=:0 ckb-next`.
`QObject::startTimer: Timers cannot have negative intervals
Invalid MIT-MAGIC-COOKIE-1 keyqt.qpa.screen: QXcbConnection: Could not connect to display :0
Could not connect to any X display.`
When I do `DISPLAY=:1 ckb-next` I get this instead:
`QObject::startTimer: Timers cannot have negative intervals

(ckb-next:18775): Gtk-WARNING **: 20:25:43.924: Theme parsing error: gtk-widgets.css:1476:22: Not a valid image
The ckb-next GUI should not be run as root.`

The Gtk error is probably because I have a theme that isn't fully up to date. I'm not sure what the negative intervals thing is about, though.

I get another _different_ error when I do `DISPLAY=:1-next` or `DISPLAY=:0 ckb-next` as the non-root user:
`QObject::startTimer: Timers cannot have negative intervals
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-nonroot'
No protocol specified
qt.qpa.screen: QXcbConnection: Could not connect to display :1
Could not connect to any X display.`

It seems like it's a problem on my end but I'm not sure. Just in case this is relevant, I'm using the Cinnamon desktop environment. I was using Gnome originally, but I've been trapped by the world of 144hz monitors and it was just too slow for me.

OvermindDL1, 2018-08-13:
@takochako You are running that ckb-next command with the DISPLAY environment as root?

takochako, 2018-08-14:
I tried running it as root and as non-root by switching with `su - nonroot`. "nonroot" is the standard account that I made. I'm not sure if that's how I'm supposed to be doing it though.

OvermindDL1, 2018-08-14:
I 'think' it needs to be run as whatever user that X is, so if X is root then ckb-next needs to be root too.

ThoreauHenry, 2018-08-21:
Kali is not a distro.  It's a diagnostic tool running as root.  If you are worried about configuring led's on that OS, you are doing it wrong.  I swear, sometimes I wish Mr. Robot burned in a fire.

tatokis, 2018-08-21:
As a sidenote, this software is needed for devices like the glaive to even move the pointer.

Anyway, since others disagree with me, I'm going to suggest you can either figure out how to run it as a normal user, or remove those lines and recompile.
https://github.com/ckb-next/ckb-next/blob/master/src/gui/main.cpp#L171-L175

Running the GUI as root is not supported, and if something breaks due to that, you get to keep all the pieces.

closed 2018-08-21 by tatokis

Unexpected-Development, 2024-06-20:
This one liner typically FIXES software that treats you like an idiot.  Works for ckb-next as well.  Examples below.

sed -i 's/getuid/getpid/' /usr/bin/ckb-next
sed -i 's/geteuid/getppid/' /usr/bin/vlc