ckb-next/ckb-next #223: Corsair Scimitar Pro bricked updating firmware from 1.09 to 1.10

wyqydsyq, 2018-08-09:
---

# Devices information
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2274577/ckb-next-dev-detect-report.gz)

# Bug
I just got a new SteelSeries Scimitar Pro, plugged it in and it worked out of the box so I thought I'd try ckb-next to customize the bindings. It popped up telling me there was a firmware update (from 1.09 to 1.10) and I clicked OK to install it. Installation seemed to go fine but then the lights went out and my computer no longer detected the mouse, only CKB. I tried following some guides online to reset the firmware manually by pressing the pin to activate the CRP DISABLD mode, but it seems to be completely bricked now. I can't delete the existing firmware.bin file, it looks like its deleted and I can add the new firmware but when I disconnect and reconnect the mouse it still has the original firmware from 2008

tatokis, 2018-08-09:
Please join us on irc. There's a link in the readme.

ZirconiumX, 2018-08-09:
First off, your mouse isn't entirely bricked. The bootloader ensures it can
be saved.

Don't try to delete or flash the firmware through Linux fat, because the
internal flash isn't an actual FAT filesystem but a virtual one; the write
reordering that the kernel does messes up the firmware.

But yeah, come join us in IRC.

tatokis, 2018-08-09:
For context, the firmware update most likely failed due to the Scimitar Pro not being in the V2_OVERRIDE macro. Firmware 1.XX is P04.

Should be fixed by https://github.com/ckb-next/ckb-next/commit/6d921ab337664f02126bb27ccb463608959a2cac

Device was restored by flashing an ISP image.

closed 2018-08-09 by tatokis