ckb-next/ckb-next #225: K70 Keyboard constant 'No devices connected'

iancox890, 2018-08-09:
I've done everything I can think of, but no matter what my K70 RGB keyboard won't show up as connected. The rainbow effect will activate, or the standard, and it will switch back and forth. Besides that, there's nothing.

[ckb-next-report-1533841062.pdf](https://github.com/ckb-next/ckb-next/files/2275574/ckb-next-report-1533841062.pdf)

ZirconiumX, 2018-08-09:
Mac OS X isn't detecting your keyboard. Check if it's plugged in and try a
different port. If nothing works, you might need to RMA the keyboard.
Either way, CKB is correct this time.

iancox890, 2018-08-09:
Thank you for the quick response. I have the keyboard plugged into a USB hub. I've tried other ports and it seems like the issue still persists.

iancox890, 2018-08-09:
I have used it in my PC before hand and everything worked great.

ZirconiumX, 2018-08-09:
A few ideas from people who have K70 RGBs - out of the factory the K70 RGB
has a "demo mode" that sounds exactly like what you're describing, and the
keyboard physically stops responding, requiring a replug.

As a side note, USB hubs tend to cause more harm than good.

iancox890, 2018-08-09:
I was forced into using the hub due to a lack of ports with my laptop. Do think there is anything else I can try in order to resolve the issue?

tatokis, 2018-08-09:
If your K70 has multiple USB plugs, make sure both are plugged in.

iancox890, 2018-08-09:
Yes, both are plugged in.

tatokis, 2018-08-09:
Yeah, I'm not sure what can be done. It's not a software issue. The demo mode, as mentioned by @ZirconiumX, means that your keyboard can't talk to your computer.
The report also proves that since there is nothing under "IOReg".

It's either a usb hub issue, or the wire on the keyboard is broken somewhere.

iancox890, 2018-08-09:
Do you think I should try seeing if it still works with my PC?

tatokis, 2018-08-09:
Yes. That will rule out the possibility of a broken keyboard.

theupriser, 2018-08-09:
What happens if you reinstall ckb? seems to bee the same issue as I have

iancox890, 2018-08-09:
I will bring my pc out now and try. I've reinstalled ckb once already and nothing changed.

tatokis, 2018-08-09:
@theupriser it's not the same issue. The device is not being enumerated. It would show up in IOReg regardless of ckb.

iancox890, 2018-08-09:
I got my PC hooked up and the keyboard isn't working.

ZirconiumX, 2018-08-09:
That is not a good sign.

tatokis, 2018-08-09:
What if you unplug it, hold down the Esc key, and while holding it down, plug the keyboard in?

iancox890, 2018-08-09:
Regardless, the lights are working. So when it detects that it's connected, the lights pop up. Is that the demo mode you're referring to?

iancox890, 2018-08-09:
It works! The ESC key trick did it.

tatokis, 2018-08-09:
Update the firmware since you're at it. You're probably on firmware v1.

iancox890, 2018-08-09:
Yes, definitely. Thank you so much, you were a huge help. Jumped in joy when it worked. :)

iancox890, 2018-08-09:
Keyboard is now working on my mac. Thank you once again, incredible! :)