ckb-next/ckb-next #229: K70 RGB randomly not seen by Linux until re-plugged

exblack, 2018-08-15:
# Devices information

Custom Built PC

Linux Mint 19 Tara
Linux Mint 18.3
Ubuntu

Corsair K70 RGB (firmware 2.05)

---

# Bug report

## General information

### Source
  * https://github.com/ckb-next/ckb-next
  * built manually using quickinstall script

## Logs & crash reports
logs generated when the keyboard is connected
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2291301/ckb-next-dev-detect-report.gz) 

Let me know if another set when it is not seen by os is needed

## What is the program's current behavior?
Ever since I got into Linux I am experiencing the same issue with my K70 keyboard. From time to time when I boot into Linux after restart or shutdown the keyboard is not seen by the OS and I have to re-plug it to have it working.

I can use the keyboard normally at all times until I boot to Linux. When Linux boots the keyboard doesn't work in 20% of cases. It is lit up according to saved hardware configuration; Numlock light is off and no input is made from it. Re-plugging the keyboard at this state forces the OS to see the device in 90% of cases. I noticed that waiting for 2-3 sec before plugging it back has a higher success rate but that might be just my imagination.

On other 80% of cases or after I re-plug it I see the Numlock light blink after a second or two and then turn on indicating the keyboard is working properly. 

When the keyboard is not seen it remains like that until it is re-plugged or Linux is rebooted (up to chance afterwards again). Logging in using a different keyboards (for example on-screen one) doesn't change its state and ckb shows "no devices connected" - haven't tested that in ages though.

I have tried switching to different USB ports (including between USB 2 and 3) to no effect.
Adding USB quirks seems to have increased the change it will work on boot (before it was 20% success change instead of 80%) but that might have been a coincidence.
I created a dupe boot entry in grub-customizer with usbquirks set to 0x4 but didn't notice any difference on that boot. Not sure if they kick in when added like that.
Re-plugging it with ESC pressed didn't do anything but resent my color profile

To clarify:
- Keyboard always worked on Windows
- Keyboard always worked on Mac (hackingtosh) with no ckb installed
- Keyboard has the same behaviour with all Linux distros I tried
- Keyboard always works in UEFI - issue only occurs once Linux is booted (login screen or not)

## What is the expected behavior of the program?
For the keyboard to be detected by the OS on all boots

## Any other notes & comments?
I've seen a few similar issues raised but they are not quite the same so I will do my best to provide as much info as possible. Also not sure if this can be addressed with ckb-next at all. In general it feels like the kernel is not seeing the device on 2/10 boots.
If this is not something that can be addressed via ckb please let me know and I will post this on Linux mint forums.

On rarer occasions I've noticed that when I reboot or power down from login screen in Linux when keyboard is not working the light stay on even if I turn the PC off. Even re plugging it on a turned off PC light the keys. Not sure if that is helpful.
---

# Something else
I will be more than happy to do any test scenarios and provide additional reports.
I am quite new to Linux though so might need some instructions on more advanced stuff.

ZirconiumX, 2018-08-15:
First off, thank you **very** much for the detail, this is very useful. For the record you have a K70 LUX RGB, rather than the original K70 RGB.

Your ckb-next-dev-detect has the following log lines:
```
Aug 15 19:32:43.306981 artem-mint ckb-next-daemon[1001]: [I] Connecting  at /dev/input/ckb1
Aug 15 19:32:43.306981 artem-mint ckb-next-daemon[1001]: [E] os_setupusb (usb_linux.c:594): Unable to read endpoint count from udev, assuming 0
Aug 15 19:32:43.306981 artem-mint ckb-next-daemon[1001]: [I] Attempting reset...
Aug 15 19:32:43.793606 artem-mint ckb-next-daemon[1001]: [E] os_usbsend (via firmware.c:24):  No such process, res = 0xffffffff
Aug 15 19:32:43.793606 artem-mint ckb-next-daemon[1001]: [W] _start_dev (device.c:43): Unable to load firmware version/poll rate
Aug 15 19:32:43.793606 artem-mint ckb-next-daemon[1001]: [I] Reset success
[repeated a bunch of times]
```
and I'm guessing your keyboard was not working at that time. What that indicates is that Linux was having difficulty talking to your keyboard, because if you compare that with a successful initialisation:
```
Aug 15 19:33:03.891667 artem-mint ckb-next-daemon[1001]: [I] Connecting Corsair Gaming K70 LUX RGB Keyboard at /dev/input/ckb1
Aug 15 19:33:03.893575 artem-mint ckb-next-daemon[1001]: [I] Starting input thread for /dev/input/ckb1
Aug 15 19:33:04.009964 artem-mint ckb-next-daemon[1001]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 05 02 14 00 1c 1b 33 1b 02 01 01 06 c0 ff 40 01 03 00 00 00 02 04 02 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Aug 15 19:33:05.986044 artem-mint ckb-next-daemon[1001]: [I] Setup finished for /dev/input/ckb1
```
you can see the `Connecting` line is empty on the failure, and the `couldn't read endpoint count from udev` line, which is an error.

To be blunt, this looks like your keyboard is dying, because that is not normal behavior, but we do have one more kernel quirk up our sleeve; could you try `usbcore.quirks=1b1c:1b33:g`?

exblack, 2018-08-15:
Thanks for the quick reply. Yes on the session I had to re-plug the keyboard on login screen.

I have changed usbhid quirks to usbcore as instructed but on boot all my usb devices weren't working. Logs tell that I have entered an incorrect parameter:
```
Dmesg on last boot:
Aug 15 22:00:26.475329 artem-mint kernel: Command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-29-generic root=UUID=ada844fe-f5bc-4120-8a68-dcf263eee083 ro quiet splash usbcore.quirks=1b1c:1b33:g vt.handoff=1
Aug 15 22:00:26.476612 artem-mint kernel: Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-29-generic root=UUID=ada844fe-f5bc-4120-8a68-dcf263eee083 ro quiet splash usbcore.quirks=1b1c:1b33:g vt.handoff=1
Aug 15 22:00:26.476617 artem-mint kernel: Booting kernel: `1b1c:1b33:g' invalid for parameter `usbcore.quirks'
```
Isn't usbcore for Kernels 4.17 an up? I am on 4.15 if not mistaken.
Anyways I couldn't even login as no USBs were working so rebooted with usbhid.quirks=0x4 and keyboard connected on that one without replug - log: 
[ckb-next-dev-detect-report-1.gz](https://github.com/ckb-next/ckb-next/files/2291744/ckb-next-dev-detect-report-1.gz)

Then I assumed maybe the parameter should start with 0x so I tried that and had the same result. Afterwards booted with regular quirks I had initially and had to re-plug keyboard on login screen (tnx for the code line see the difference quite clearly now) - log:
[ckb-next-dev-detect-report-2.gz](https://github.com/ckb-next/ckb-next/files/2291752/ckb-next-dev-detect-report-2.gz)

So bottom line usbcore.quirks=1b1c:1b33:g just disabled USB for me altogether. I will try to boot with this parameter via usbhid as I feel usbcore is not supported by my kernel with and without 0x in the beginning and update if anything is different.

Lastly would like to mention that I have a USB audio interface connected - Focusrite Scarlet 2i2 - I know that it used to cause some problems with USB on a Mac so mentioning in case it might be the cause here. Regarding the keyboard dying - I had this issue pretty much since I bought it from Amazon, but I guess faulty unit in not impossible. Weird that only Linux cannot handle it though. Anyways I will also do some additional tests to exclude some things over the week and update here as well.

exblack, 2018-08-15:
3
Curent Boot: usbhid.quirks=0x1b1c:1b33:g 
had to replug twice on loging for it to start working
[ckb-next-dev-detect-report-3.gz](https://github.com/ckb-next/ckb-next/files/2291797/ckb-next-dev-detect-report-3.gz)

4
Curent Boot: usbhid.quirks=1b1c:1b33:g 
1st boot keyboard normal
2nd boot took quite long but still keyboard normal
3rd boot had to replug for keyboard to work
[ckb-next-dev-detect-report-4.gz](https://github.com/ckb-next/ckb-next/files/2291798/ckb-next-dev-detect-report-4.gz)

tatokis, 2018-08-16:
@exblack Your kernel is indeed too old to support usbcore quirks over cmdline.
However, it doesn't look like you are doing it correctly either.
On your tests earlier you used the usbcore quirk as "usbhid". Do note they are different.
https://github.com/ckb-next/ckb-next/wiki/Troubleshooting

You have two options.
1. Update to kernel 4.18 through the ubuntu mainline kernel repo (http://kernel.ubuntu.com/~kernel-ppa/mainline/) or ukuu and use both the **usbcore** and **usbhid** quirks.

2. Join us on freenode (link in the readme) and we can try an easier alternative.

exblack, 2018-09-22:
Sorry for such a long delay in replying. Was running out of storage and needed to reinstall Linux on a different SSD so decided to do further tests on a new install. So to sum up:

- I have completely reinstalled Linux Mint 19 to another SSD
- Updated kernel to 4.18 using ukuu
- Installed ckb-next
- added both usb quirks like this "usbcore.quirks=1b1c:1b33:g usbhid.quirks=0x1b1c:0x1b33:0x20000408"

Prior to adding quirks all was as usual, chance on Linux boot not to recognize the keyboard.
After adding both quirks the boot was noticeably longer and on Linux boot none of the USB devices were recognized.
Logs shows: Booting kernel: `1b1c:1b33:g' invalid for parameter `usbcore.quirks' 
not sure why usbcore quirks do not recognize it even though I do have kernel 4.18. Any ideas how I can get those to work?

Had to reboot via older kernel to change boot parameters back and get the log: 
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2407859/ckb-next-dev-detect-report.gz)

I am wondering at this point if the issue is indeed with the keyboard or could it be my motherboard or something else?
I would like to reiterate that only after Linux boot the issue sometimes occurs. both UEFI and Windows recognize the keyboard normally 100% of the time.

PS
Does the position of the switch at the back of the keyboard matter (obviously besides the bios one)? I tied several with no difference so I am sticking with position 2 at the moment. Maybe one works better in Linux than others?

tatokis, 2018-09-22:
The issue is with Corsair's firmware.
The position of the switch doesn't really matter in this case, as long as it is not in the BIOS position.
Do join us on IRC though

exblack, 2018-09-22:
Just joined IRC. My cmdline below:

recordfail
savedefault
load_video
gfxmode $linux_gfx_mode
insmod gzio
if [ x$grub_platform = xxen ]; then insmod xzio; insmod lzopio; fi
insmod part_gpt
insmod ext2
set root='hd3,gpt3'
if [ x$feature_platform_search_hint = xy ]; then
  search --no-floppy --fs-uuid --set=root --hint-bios=hd3,gpt3 --hint-efi=hd3,gpt3 --hint-baremetal=ahci3,gpt3  e7618114-29a4-4ec0-8172-f8015c3fbe30
else
  search --no-floppy --fs-uuid --set=root e7618114-29a4-4ec0-8172-f8015c3fbe30
fi
        linux	/boot/vmlinuz-4.18.8-041808-generic root=UUID=e7618114-29a4-4ec0-8172-f8015c3fbe30 ro  quiet splash $vt_handoff usbcore.quirks=1b1c:1b33:g usbhid.quirks=0x1b1c:0x1b33:0x20000408
initrd	/boot/initrd.img-4.18.8-041808-generic

thespyboss, 2019-06-17:
Can i reopen this issue? im having this problem on Solus os. Do i need to post my report from ckb-next?

exblack, 2019-06-17:
@thespyboss - my issue was solved by firmware update released by Corsair (i did manage to get one before official release but official now works fine)
Firware version 3.08 works fine now and was ever since I got the update.
You you have the latest firware (3.08) and still having issues I would imagine it is a different issue and would better be recorded in a saparate ticket