ckb-next/ckb-next #230: Strafe RGB freze completely after wake up

splatch, 2018-08-17:
I've hit a mysterious issue with Strafe and CKB. I use Arch and ckb-next-git package, so I am close to latest releases.
When error occurs keyboard, meaning its hardware part, is simply hanged. I can not even change back light on it.
I am writing this bug report on second device. Versions used:
- Linux arch 4.17.13-2-MANJARO
- ckb-next 0.3.0-1

Software itself ends up with cores dumped:
```aug 17 10:41:43 arch systemd-coredump[18168]: Process 3854 (ckb-next) of user 1001 dumped core.
                                              
                                              Stack trace of thread 3854:
                                              #0  0x00007fb2c7007b00 _ZN10QByteArray11reallocDataEj6QFlagsIN10QArrayData16AllocationOptionEE (libQt5Core.so.5)
                                              #1  0x00007fb2c71bb84f n/a (libQt5Core.so.5)
                                              #2  0x00007fb2c71bc0b9 n/a (libQt5Core.so.5)
                                              #3  0x00007fb2c71c2416 _ZN11QMetaObject19normalizedSignatureEPKc (libQt5Core.so.5)
                                              #4  0x00007fb2c71e183f _ZN7QObject10disconnectEPKS_PKcS1_S3_ (libQt5Core.so.5)
                                              #5  0x00007fb2c82cbf75 n/a (libQt5Network.so.5)
                                              #6  0x00007fb2c82cc13c n/a (libQt5Network.so.5)
                                              #7  0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #8  0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #9  0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #10 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #11 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #12 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #13 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #14 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #15 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #16 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #17 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #18 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #19 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #20 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #21 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #22 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #23 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #24 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #25 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #26 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #27 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #28 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #29 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #30 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #31 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #32 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #33 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #34 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #35 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #36 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #37 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #38 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #39 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #40 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #41 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #42 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #43 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #44 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #45 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #46 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #47 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #48 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #49 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #50 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #51 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #52 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #53 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #54 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #55 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #56 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #57 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
                                              #58 0x00007fb2c82cc153 n/a (libQt5Network.so.5)
                                              #59 0x00007fb2c82cc588 n/a (libQt5Network.so.5)
[... 99 lines cut ...]
                                              #7  0x00007fb2c69e0a8d start_thread (libpthread.so.0)
                                              #8  0x00007fb2c6910823 __clone (libc.so.6)
```
It is not possible to restart ckb-next daemon. Interesingly enough, even `kill -9` does not shut a process.
```
aug 17 10:44:40 arch systemd-udevd[313]: seq 6254 '/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1B1C:1B20.0005/input/input19' killed
aug 17 10:44:40 arch systemd-udevd[313]: worker [18144] terminated by signal 9 (KILL)
aug 17 10:44:40 arch systemd-udevd[313]: worker [18144] failed while handling '/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1B1C:1B20.0005/input/input19'

aug 17 10:47:40 arch systemd-udevd[313]: seq 6255 '/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1B1C:1B20.0005/input/input19/event17' killed
aug 17 10:47:40 arch systemd-udevd[313]: worker [18147] terminated by signal 9 (KILL)
aug 17 10:47:40 arch systemd-udevd[313]: worker [18147] failed while handling '/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1B1C:1B20.0005/input/input19/event17'
aug 17 10:47:40 arch upowerd[752]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1B1C:1B20.0005

aug 17 10:48:49 arch ckb-next-daemon[18493]:     ckb-next: Corsair RGB driver 0.3.0
aug 17 10:48:49 arch ckb-next-daemon[18493]: [F] ckb-next-daemon is already running (PID 455). Try `killall ckb-next-daemon`.
aug 17 10:48:49 arch ckb-next-daemon[18493]: [F] (If you're certain the process is dead, delete /dev/input/ckb0/pid and try again)
aug 17 10:50:19 arch systemd[1]: ckb-next-daemon.service: State 'stop-sigterm' timed out. Killing.
aug 17 10:50:19 arch systemd[1]: ckb-next-daemon.service: Killing process 2331 (ckb-next-daemon) with signal SIGKILL.
aug 17 10:50:41 arch systemd-udevd[313]: seq 6264 '/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0' killed
aug 17 10:50:41 arch systemd-udevd[313]: worker [18356] terminated by signal 9 (KILL)
aug 17 10:50:41 arch systemd-udevd[313]: worker [18356] failed while handling '/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0'
```

The only way to recover is shutdown so keyboard gets a normal power cycle.

ZirconiumX, 2018-08-17:
A few things that stick out before I look deeper into this:

- We have ckb-next 0.3.1 out now, so could you upgrade to that?
- I'm running Arch with that exact keyboard and so far it's been fine.

splatch, 2018-08-17:
Sure, I will do system upgrade and see if it gets back. It is non deterministic at all - meaning doesn't happen all the time. With older version of kernel (and possibly ckb) I had minor issues which I could solve by restarting daemon.

Will upload system report too.

splatch, 2018-08-17:
Here is system report [ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2296632/ckb-next-dev-detect-report.gz)

tatokis, 2018-08-21:
If the GUI is crashing in libQt5Network.so.5 and/or libQt5Core.so.5, then there's not much we can do about it. Your best bet is to up/downgrade Qt.
Not sure about the daemon though. Maybe you need the usbcore quirks?
https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#usbcore-quirks

Is this still happening after updating your system?

splatch, 2018-08-21:
I updated system right after previous occurrence of error and till this evening it was fine. However I experienced issue again. I was able to open GUI, but hardware itself is not responsible - daemon restart failed:

```
aug 21 23:14:49 arch kernel: input: Corsair Corsair STRAFE RGB Gaming Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1B1C:1B20.0005/input/input19
aug 21 23:14:49 arch kernel: hid-generic 0003:1B1C:1B20.0005: input,hidraw1: USB HID v1.11 Keyboard [Corsair Corsair STRAFE RGB Gaming Keyboard] on usb-0000:00:14.0-7/input0
aug 21 23:14:49 arch kernel: usbhid 1-7:1.2: couldn't find an input interrupt endpoint
aug 21 23:18:51 arch systemd-udevd[365]: seq 5461 '/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1B1C:1B20.0005/input/input19/event17' is taking a long time
aug 21 23:19:03 arch ckb-next-daemon[649]: [I] Caught signal 15
aug 21 23:19:03 arch systemd[1]: Stopping Corsair Keyboards and Mice Daemon...
-- Subject: Unit ckb-next-daemon.service has begun shutting down
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit ckb-next-daemon.service has begun shutting down.

aug 21 23:20:33 arch systemd[1]: ckb-next-daemon.service: State 'stop-sigterm' timed out. Killing.
aug 21 23:20:33 arch systemd[1]: ckb-next-daemon.service: Killing process 649 (ckb-next-daemon) with signal SIGKILL.
aug 21 23:20:37 arch NetworkManager[653]: <warn>  [1534886437.2412] device (wlp0s20u13): set-hw-addr: new MAC address 6E:0C:FA:C0:37:0C not successfully set (scanning)
aug 21 23:20:51 arch systemd-udevd[365]: seq 5461 '/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1B1C:1B20.0005/input/input19/event17' killed
aug 21 23:20:51 arch systemd-udevd[365]: worker [21575] terminated by signal 9 (KILL)
aug 21 23:20:51 arch systemd-udevd[365]: worker [21575] failed while handling '/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1B1C:1B20.0005/input/input19/event17'
aug 21 23:20:51 arch upowerd[826]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1B1C:1B20.0005
aug 21 23:21:52 arch systemd-udevd[365]: seq 5470 '/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0' is taking a long time
aug 21 23:22:03 arch systemd[1]: ckb-next-daemon.service: Processes still around after SIGKILL. Ignoring.
aug 21 23:23:33 arch systemd[1]: ckb-next-daemon.service: State 'stop-final-sigterm' timed out. Killing.
aug 21 23:23:33 arch systemd[1]: ckb-next-daemon.service: Killing process 649 (ckb-next-daemon) with signal SIGKILL.
aug 21 23:23:33 arch systemd[1]: ckb-next-daemon.service: Killing process 768 (ckb-next-daemon) with signal SIGKILL.
aug 21 23:23:52 arch systemd-udevd[365]: seq 5470 '/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0' killed
aug 21 23:23:52 arch systemd-udevd[365]: worker [22039] terminated by signal 9 (KILL)
aug 21 23:23:52 arch systemd-udevd[365]: worker [22039] failed while handling '/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0'
aug 21 23:23:52 arch upowerd[826]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0
aug 21 23:25:04 arch systemd[1]: ckb-next-daemon.service: Processes still around after final SIGKILL. Entering failed mode.
aug 21 23:25:04 arch systemd[1]: ckb-next-daemon.service: Failed with result 'timeout'.
aug 21 23:25:04 arch systemd[1]: Stopped Corsair Keyboards and Mice Daemon.
-- Subject: Unit ckb-next-daemon.service has finished shutting down
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit ckb-next-daemon.service has finished shutting down.
aug 21 23:25:04 arch systemd[1]: ckb-next-daemon.service: Found left-over process 768 (ckb-next-daemon) in control group while starting unit. Ignoring.
aug 21 23:25:04 arch systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
aug 21 23:25:04 arch systemd[1]: Started Corsair Keyboards and Mice Daemon.
-- Subject: Unit ckb-next-daemon.service has finished start-up
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit ckb-next-daemon.service has finished starting up.
-- 
-- The start-up result is RESULT.
aug 21 23:25:04 arch ckb-next-daemon[22241]:     ckb-next: Corsair RGB driver 0.3.1
aug 21 23:25:04 arch ckb-next-daemon[22241]: [F] ckb-next-daemon is already running (PID 649). Try `killall ckb-next-daemon`.
aug 21 23:25:04 arch ckb-next-daemon[22241]: [F] (If you're certain the process is dead, delete /dev/input/ckb0/pid and try again)
aug 21 23:26:34 arch systemd[1]: ckb-next-daemon.service: State 'stop-sigterm' timed out. Killing.
aug 21 23:26:34 arch systemd[1]: ckb-next-daemon.service: Killing process 768 (ckb-next-daemon) with signal SIGKILL.
```

There is still a process which can not be terminated:
> root       649  0.4  0.0      0     0 ?        Zsl  aug20   9:41 [ckb-next-daemon] <defunct>

Will try to tweak usb settings and report back if it improves. I will close issue if it will solve my trouble. Generally I see "couldn't find an input interrupt endpoint" on every system boot but I am not sure why and if its something worrying.

tatokis, 2018-08-21:
>Generally I see "couldn't find an input interrupt endpoint" on every system boot but I am not sure why and if its something worrying.

That's normal, and nothing to worry about. It'll go away once Corsair release the new firmware images.

I'm not really sure why the daemon is hanging though...

splatch, 2020-05-06:
I am pulling this up, since I began to run into rather systematic freeze after system wake up. Restart of daemon doesn't work. I noticed that sometimes after wakeup keyboard has a lag or small hang causing back light to flicker. If such thing happens - it works. Overall I could not determine reason of why it does hang.

Some logs from `journalctl`:
```
maj 05 10:39:42 arch ckb-next-daemon[693]: ckb-next: Corsair RGB driver 0.4.2
maj 05 10:39:42 arch ckb-next-daemon[693]: [I] Root controller ready at /dev/input/ckb0
maj 05 10:39:47 arch ckb-next-daemon[693]: [I] Connecting Corsair STRAFE RGB Gaming Keyboard at /dev/input/ckb1
maj 05 10:39:47 arch ckb-next-daemon[693]: [I] Starting input thread for /dev/input/ckb1
maj 05 10:39:47 arch ckb-next-daemon[693]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 08 03 14 00 1c 1b 20 1b 08 01 01 08 c0 ff 40 00 03 00 00 00 02 04 01 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
maj 05 10:39:49 arch ckb-next-daemon[693]: [I] Setup finished for /dev/input/ckb1
maj 06 11:00:54 arch ckb-next-daemon[693]: [I] System has woken from sleep
maj 06 11:00:55 arch ckb-next-daemon[693]: [E] os_sendindicators (usb_linux.c:278): Device or resource busy
maj 06 11:00:55 arch ckb-next-daemon[693]: [I] Attempting reset...
```

closed 2022-10-03 by tatokis