ckb-next/ckb-next #232: ckb next not detecting my mouse anymore

octillery143, 2018-08-17:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?
does not detect mouse
## What would you like it to do instead?
to detect mouse
## Any other notes & comments?
i accidentally pulled the power cord out of my pc and after i loaded up again and ckb and it doesnt detect my mouse anymore (corsair m65 pro rgb mouse)
---

# Bug report

## General information

### Source

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

octillery143, 2018-08-17:
um, feature request is supposed to be in bug report, sorry about that

tatokis, 2018-08-19:
Please make sure the mouse is plugged in, open a terminal, run `curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash` and attach the file.

Gcx44, 2018-10-03:
Hi there,

Same mouse and same error. No device detected, but it worked a week ago.
Here is the report.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2442541/ckb-next-dev-detect-report.gz)

tatokis, 2018-10-04:
That is very odd. Have you tried unplugging the mouse, powering the computer off, plugging it in and then turning it back on?

Sparkplug1034, 2019-03-22:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2994788/ckb-next-dev-detect-report.gz)
Here is my report. I am having the issue still, even on the latest version (that dnf gave me).

zladuric, 2019-09-22:
Hi,

I'm also seing no mouse - I have a keyboard and a mouse, but ckb does not see the mouse, just the keyboard - no matter what combination of "plugged in/unplugged" I use. Under "Other OS", the mouse back button works properly.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/3639645/ckb-next-dev-detect-report.gz)

buddybuddybuddy, 2019-12-16:
hey same problem, here is report 
it's doing my head in
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/3966186/ckb-next-dev-detect-report.gz)

tatokis, 2019-12-16:
Try running `sudo kextcache --clear-staging` and rebooting

closed 2022-03-05 by tatokis