ckb-next/ckb-next #241: Strafe RGB MK.2 Problem

EricGao888, 2018-08-27:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

### Source

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

Does the latest version of ckb-next support Strafe RGB MK.2? It shows no devices connected.

ZirconiumX, 2018-08-27:
Yes, it does.

Can you please upload a ckb report, like it tells you to do?

aydex, 2018-08-27:
[ckb-next-report-1535370215.gz](https://github.com/ckb-next/ckb-next/files/2323709/ckb-next-report-1535370215.gz)
Having the same issue as @EricGao888, report attached

tatokis, 2018-08-27:
Try this http://ckb-next.info/mac/ckb-next_v0.3.1-41-gb25a611_10.11-SDK.dmg

aydex, 2018-08-27:
Works for me. Thank you!

ZirconiumX, 2018-08-27:
Can we close this, then?

tatokis, 2018-08-27:
I'd like to leave it open until the next release, so that people can more easily see it.

EricGao888, 2018-08-27:
@tatokis I've tried a lot but only the method you provide works for me! Much appreciation!

tatokis, 2018-08-27:
It'll be fixed in the next release. Apologies for that.

MikaelAbehsera, 2018-09-02:
@tatokis I'm on a k70 rapidF RGB and my x seems to decide its own colors and does not listen, this just started lately, and also ckb stops recognizing my keyboard after couple of times of disconnecting it, I have now downloaded the ckb above and can note back to you if anything has been fixed, thanks for making this software your a lifesaver!

Vermyndax, 2018-09-24:
FYI this also fixed my issue (using macOS Mojave).

ZirconiumX, 2019-01-09:
v0.3.2 is out, so I'm closing this.

closed 2019-01-09 by ZirconiumX