ckb-next/ckb-next #243: M65 Pro Back/Forward Binding Bug

Jass360, 2018-08-27:
Forwards and backwards buttons on the Corsair M65 Pro do not bind properly. Can this be a fixed in a future version?

tatokis, 2018-08-27:
What does "do not bind properly" mean.
Which OS?

Jass360, 2018-08-27:
Sorry - Mac OS High Sierra. Binding as in I select the next and back functions to the forwards and backwards buttons on the side of the mouse. The functions do not seem to be binding, with the standard left click left as default for both buttons.

tatokis, 2018-08-27:
Can you generate a report from the settings tab and attach it?

tatokis, 2018-08-27:
Do keep in mind macOS does not have support for "forwards" and "backwards" by default.

Jass360, 2018-08-27:
[ckb-next-report-1535404389.gz](https://github.com/ckb-next/ckb-next/files/2325649/ckb-next-report-1535404389.gz)

Jass360, 2018-08-27:
So if Mac OS does not have support for these binds, i'm guessing this isn't classed as a bug or anything that could be fixed?

Jass360, 2018-08-27:
Thank you very much for those macros. They do exactly what I wanted the binds to do. From what you're saying, I would say that this is a bug from my set functions not being performed. Thank you for your help - I look forward to future releases!

tatokis, 2018-08-27:
If the macros work when bound to the mouse buttons, then it's not a bug.

Judging by the `They do exactly what I wanted the binds to do.`, it sounds like it does work.
It would've been a bug if pressing the buttons with the macros did absolutely nothing.

Either way, this will be the default behaviour on the next release, so I'm closing this.
If I misunderstood you, then do reply and I will reopen the issue.

closed 2018-08-27 by tatokis