ckb-next/ckb-next #247: Gentoo Packages

orangeturtle739, 2018-09-04:
# Packages

I made Gentoo Ebuild scripts for the new CMake-based releases in my [panther overlay](https://github.com/orangeturtle739/panther). The overlay provides:
* `app-misc/ckb-next-9999`, the latest git sources
* `app-misc/ckb-next-0.3.1`, the latest release

The official Gentoo repositories also have the pre-cmake version, 0.2.9, under `app-misc/ckb`: https://packages.gentoo.org/packages/app-misc/ckb.

I'm willing to keep my packages up to date. Would you mind mentioning them on the [Linux Installation](https://github.com/ckb-next/ckb-next/wiki/Linux-Installation) page?

orangeturtle739, 2018-09-20:
@tatokis interested in this (you seem to be the last one who touched the Linux Installation page on the wiki)?

tatokis, 2018-09-20:
Sorry, I completely forgot about this.
I'll add these to the wiki and possibly advise against using the official ones, as they are horribly out of date.

There's nothing worse than receiving reports about fixed bugs, or people bricking their devices installing new firmware because they were running a broken old version of the software.

closed 2018-09-20 by tatokis

tatokis, 2018-09-20:
Updated. Hopefully I got it right, as I know nothing about gentoo overlays.

orangeturtle739, 2018-09-20:
Thanks!  It looks good. Gentoo overlays are roughly similar to the AUR. Users can add my overlay to their systems with:
```
layman -a panther
```
And then proceed to install the package as usual:
```
emerge app-misc/ckb-next
```