ckb-next/ckb-next #249: K95 platinum RGB visuals after sleep not working

Kallestofeles, 2018-09-06:
# Bug report

## General information
K95 platinum RGB will not work after resuming from sleep. The keys work and the ckb-next utility can be used to modify the keyboard however the visuals (rgb) will be the ones stored on the keyboard memory and not from the application. Selecting another visualization from the app does nothing.
Solution is to restart the ckb-next-daemon.service via systemd after every sleep - after daemon restart it starts functioning as expected.

### Source
  * ckb-next-git package in AUR

## Logs & crash reports

Upload:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2358003/ckb-next-dev-detect-report.gz)

## What is the program's current behavior?
The visuals are forgotten and not grabbed from the ckb-next utility after the computer resumes from sleep.

## What is the expected behavior of the program?
Visuals are used from the utility just as they were before the computer went to sleep.

## Any other notes & comments?
Restarting ckb-next-daemon.service via systemd after waking up resolves the issue until next sleep.

## Other
System:
  Host: shoemaster Kernel: 4.18.6-1-MANJARO x86_64 bits: 64 
  Desktop: Gnome 3.28.3 Distro: Manjaro Linux 

If any other piece of intel is required, I'm more than happy to provide.

ZirconiumX, 2018-09-06:
There's no clean way for us to discover we've woken from suspend, aside
from a quirk in the Corsair HID protocol which doesn't really work for the
Voids.

tatokis, 2018-10-07:
Really late, but it looks like a duplicate of https://github.com/ckb-next/ckb-next/issues/215

czarnyckm, 2019-05-23:
I provided systemd script in https://github.com/ckb-next/ckb-next/issues/215#issuecomment-494959496 which fix this issue with k95 keyboard

closed 2022-06-27 by tatokis