ckb-next/ckb-next #251: X key is deciding it’s on colours

MikaelAbehsera, 2018-09-10:
Excuse is deciding it’s own colours no matter what I try and it’s not listening to ckb, 

On MacBook Pro High Sierra 10.13.6 model early 2015

MikaelAbehsera, 2018-09-20:
@tatokis Only the x key, I'm not sure if it happens on windows, I currently only use the mac side, and currently not using windows on Bootcamp, it's a k70 RGB rapid fire, let me know if I can do anything else to help.

tdis8629, 2018-09-21:
Quick questions: What color is the key when you set everything to white, and does it go dark when you set it to a primary color (R, B, or G)?

tdis8629, 2018-10-03:
If red doesn't turn on for the key, it's probably a hardware issue - specifically - the LED. My Strafe RGB has a bad Blue and Red LED for the Up arrow and F7, respectively, which isn't seen in software.

tatokis, 2018-10-04:
I know old firmwares had such an issue. Assuming you're running the latest firmware, try holding down esc and then while holding it down, plug in your keyboard to your computer.
Keep it held for a couple of seconds and then release it.

That should reset the onboard storage of your keyboard (so you'll lose any settings saved to hardware), but it might help.

I can only suggest to try the stock profiles that come with ckb and check if the issue happens there too.
If it does, your last option is to check it on windows with the official software. Either way, it sounds like a hardware fault to me.

MikaelAbehsera, 2018-11-14:
@tatokis  I tried to do what you said, and now my keyboard won't even get recognized.