ckb-next/ckb-next #252: k65 + sabre issue

wannesvc, 2018-09-14:
Hi,

when I plug in my K65 keyboard an sabre mouse at the same time, The ckb-next daemon crashes and my peripherals stop responding. I get the following error message:

```
-- The start-up result is RESULT.
Sep 14 09:10:25 vancawa.cloudar ckb-next-daemon[7808]:     ckb-next: Corsair RGB driver 0.3.1-48-ge8d644f
Sep 14 09:10:25 vancawa.cloudar ckb-next-daemon[7808]: [I] Root controller ready at /dev/input/ckb0
Sep 14 09:10:25 vancawa.cloudar ckb-next-daemon[7808]: [I] Connecting Corsair Gaming SABRE RGB Mouse at /dev/input/ckb1
Sep 14 09:10:25 vancawa.cloudar ckb-next-daemon[7808]: [I] Connecting Corsair K65 RGB Gaming Keyboard at /dev/input/ckb2
Sep 14 09:10:25 vancawa.cloudar kernel: input: ckb2: Corsair K65 RGB Gaming Keyboard vKB as /devices/virtual/input/input75
Sep 14 09:10:25 vancawa.cloudar kernel: input: ckb1: Corsair Gaming SABRE RGB Mouse vKB as /devices/virtual/input/input76
Sep 14 09:10:25 vancawa.cloudar kernel: input: ckb2: Corsair K65 RGB Gaming Keyboard vM as /devices/virtual/input/input77
Sep 14 09:10:25 vancawa.cloudar kernel: input: ckb1: Corsair Gaming SABRE RGB Mouse vM as /devices/virtual/input/input78
Sep 14 09:10:25 vancawa.cloudar ckb-next-daemon[7808]: [I] Starting input thread for /dev/input/ckb2
Sep 14 09:10:25 vancawa.cloudar ckb-next-daemon[7808]: [I] Starting input thread for /dev/input/ckb1
Sep 14 09:10:25 vancawa.cloudar systemd-logind[1194]: Watching system buttons on /dev/input/event18 (ckb1: Corsair Gaming SABRE RGB Mouse vKB)
Sep 14 09:10:25 vancawa.cloudar systemd-logind[1194]: Watching system buttons on /dev/input/event20 (ckb2: Corsair K65 RGB Gaming Keyboard vM)
Sep 14 09:10:25 vancawa.cloudar systemd-logind[1194]: Watching system buttons on /dev/input/event19 (ckb1: Corsair Gaming SABRE RGB Mouse vM)
Sep 14 09:10:25 vancawa.cloudar systemd-logind[1194]: Watching system buttons on /dev/input/event17 (ckb2: Corsair K65 RGB Gaming Keyboard vKB)
Sep 14 09:10:26 vancawa.cloudar ckb-next-daemon[7808]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 04 02 01 01 1c 1b 2f 1b 01 01 01 30 c1 ff 40 03 5e 00 01 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 01 01 00 01 00 00 6c >
Sep 14 09:10:26 vancawa.cloudar ckb-next-daemon[7808]: [I] Setup finished for /dev/input/ckb1
Sep 14 09:10:26 vancawa.cloudar audit[7808]: ANOM_ABEND auid=4294967295 uid=0 gid=0 ses=4294967295 subj=system_u:system_r:unconfined_service_t:s0 pid=7808 comm="ckb-next-daemon" exe="/usr/local/bin/ckb-next-daemon" sig=11 res=1
Sep 14 09:10:26 vancawa.cloudar ckb-next-daemon[7808]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 05 02 11 00 1c 1b 17 1b 01 01 01 06 c0 ff 40 00 03 00 00 00 02 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 >
Sep 14 09:10:26 vancawa.cloudar kernel: ckb-next-daemon[7811]: segfault at 0 ip 00000000004038ba sp 00007f84e7ffed40 error 4 in ckb-next-daemon[400000+18000]
Sep 14 09:10:26 vancawa.cloudar systemd[1]: ckb-next-daemon.service: Main process exited, code=killed, status=11/SEGV
Sep 14 09:10:26 vancawa.cloudar systemd[1]: ckb-next-daemon.service: Failed with result 'signal'.
Sep 14 09:10:26 vancawa.cloudar audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=ckb-next-daemon comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=fa>
Sep 14 09:10:26 vancawa.cloudar systemd[1]: ckb-next-daemon.service: Service hold-off time over, scheduling restart.
Sep 14 09:10:26 vancawa.cloudar systemd[1]: ckb-next-daemon.service: Scheduled restart job, restart counter is at 5.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Automatic restarting of the unit ckb-next-daemon.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Sep 14 09:10:26 vancawa.cloudar systemd[1]: Stopped Corsair Keyboards and Mice Daemon.
-- Subject: Unit ckb-next-daemon.service has finished shutting down
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit ckb-next-daemon.service has finished shutting down.
Sep 14 09:10:26 vancawa.cloudar systemd[1]: ckb-next-daemon.service: Start request repeated too quickly.
Sep 14 09:10:26 vancawa.cloudar systemd[1]: ckb-next-daemon.service: Failed with result 'signal'.
Sep 14 09:10:26 vancawa.cloudar systemd[1]: Failed to start Corsair Keyboards and Mice Daemon.
-- Subject: Unit ckb-next-daemon.service has failed
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit ckb-next-daemon.service has failed.
```

When I use each device separately I don't experience any issues. Could you please help me out? (thanks for the work on ckb-next!!)

tatokis, 2018-09-18:
Make sure you have all the dependencies from the wiki.
You will also need to install gdb using your package manager.

Plug in another mouse and KB, as your devices will most likely not function during this.
```
sudo systemctl stop ckb-next-daemon
git clone https://github.com/ckb-next/ckb-next.git
cd ckb-next
mkdir build && cd build
cmake .. -DCMAKE_BUILD_TYPE=Debug && make -j4 daemon
sudo gdb ./bin/ckb-next-daemon
```

Then in the (gdb) prompt:
```
(gdb) run
```
Do what is needed to make the software crash.
Once gdb says that a `SIGSEGV` was received, run this to get a backtrace:
```
(gdb) bt full
```
When you do that, paste the backtrace here so that I can have a look at what went wrong.
You can then type
```
(gdb) quit
sudo systemctl start ckb-next-daemon
```
to restore the daemon.

wannesvc, 2018-09-18:
I don't really know what I'm doing, but is this what you need?

```
[vancawa@vancawa] 15:31 $ sudo coredumpctl gdb
           PID: 987 (ckb-next-daemon)
           UID: 0 (root)
           GID: 0 (root)
        Signal: 11 (SEGV)
     Timestamp: Tue 2018-09-18 15:22:49 CEST (9min ago)
  Command Line: /usr/local/bin/ckb-next-daemon
    Executable: /usr/local/bin/ckb-next-daemon
 Control Group: /system.slice/ckb-next-daemon.service
          Unit: ckb-next-daemon.service
         Slice: system.slice
       Boot ID: d96b087f693f4ab1bcc865eaa00e8c11
    Machine ID: 6100b669ce604d99bc2fcd32481d47b8
      Hostname: vancawa.cloudar
       Storage: /var/lib/systemd/coredump/core.ckb-next-daemon.0.d96b087f693f4ab1bcc865eaa00e8c11.987.1537276969000000.lz4
       Message: Process 987 (ckb-next-daemon) of user 0 dumped core.
                
                Stack trace of thread 996:
                #0  0x00000000004038ba _start_dev (ckb-next-daemon)
                #1  0x0000000000403970 start_dev (ckb-next-daemon)
                #2  0x000000000040e401 _setupusb (ckb-next-daemon)
                #3  0x00007fc1083b8594 start_thread (libpthread.so.0)
                #4  0x00007fc1088efe6f __clone (libc.so.6)
                
                Stack trace of thread 987:
                #0  0x00007fc1088e77c7 __select (libc.so.6)
                #1  0x000000000040ffa8 usbmain (ckb-next-daemon)
                #2  0x00000000004027a1 main (ckb-next-daemon)
                #3  0x00007fc10881911b __libc_start_main (libc.so.6)
                #4  0x0000000000402b1a _start (ckb-next-daemon)
                
                Stack trace of thread 995:
                #0  0x00007fc1088e0e54 __read (libc.so.6)
                #1  0x000000000040539c readlines (ckb-next-daemon)
                #2  0x000000000040e479 _setupusb (ckb-next-daemon)
                #3  0x00007fc1083b8594 start_thread (libpthread.so.0)
                #4  0x00007fc1088efe6f __clone (libc.so.6)
                
                Stack trace of thread 1090:
                #0  0x00007fc1088e6c57 ioctl (libc.so.6)
                #1  0x000000000040f5f5 os_inputmain (ckb-next-daemon)
                #2  0x00007fc1083b8594 start_thread (libpthread.so.0)
                #3  0x00007fc1088efe6f __clone (libc.so.6)
                
                Stack trace of thread 1092:
                #0  0x00007fc1088e0e54 __read (libc.so.6)
                #1  0x00000000004101c1 _ledthread (ckb-next-daemon)
                #2  0x00007fc1083b8594 start_thread (libpthread.so.0)
                #3  0x00007fc1088efe6f __clone (libc.so.6)
                
                Stack trace of thread 1091:
                #0  0x00007fc1083c188a __lll_unlock_wake (libpthread.so.0)
                #1  0x00007fc1083bc421 __pthread_mutex_unlock_usercnt (libpthread.so.0)
                #2  0x000000000040891e process_input_urb (ckb-next-daemon)
                #3  0x000000000040f66d os_inputmain (ckb-next-daemon)
                #4  0x00007fc1083b8594 start_thread (libpthread.so.0)
                #5  0x00007fc1088efe6f __clone (libc.so.6)

GNU gdb (GDB) Fedora 8.1.1-3.fc28
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
[... 48 lines cut ...]
[New Thread 0x7ffff5450700 (LWP 4807)]
[I] Starting input thread for /dev/input/ckb1
[New Thread 0x7ffff4c4f700 (LWP 4810)]
[I] Starting input thread for /dev/input/ckb2
[New Thread 0x7fffe7fff700 (LWP 4811)]
[I] Received identification packet: 0e 01 00 00 01 01 00 01 04 02 01 01 1c 1b 2f 1b 01 01 01 30 c1 ff 40 03 5e 00 01 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 01 01 00 01 00 00 6c dc 02 00 00 00 00 00 00 00 00 00 00 00 00 00
[I] Setup finished for /dev/input/ckb1
[I] Received identification packet: 0e 01 00 00 01 01 00 01 05 02 11 00 1c 1b 17 1b 01 01 01 06 c0 ff 40 00 00 00 00 00 02 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Thread 3 "ckb-next-daemon" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff5c51700 (LWP 4782)]
0x00000000004038ba in _start_dev ()
(gdb) bt full
#0  0x00000000004038ba in _start_dev ()
No symbol table info available.
#1  0x0000000000403970 in start_dev ()
No symbol table info available.
#2  0x000000000040e401 in _setupusb ()
No symbol table info available.
#3  0x00007ffff75d9594 in start_thread () from /lib64/libpthread.so.0
No symbol table info available.
#4  0x00007ffff7b10e6f in clone () from /lib64/libc.so.6
No symbol table info available.
(gdb) quit
```

tatokis, 2018-09-18:
That's not too helpful because there are no debug symbols. 
`No symbol table info available.`

Try installing the debug binary you compiled earlier system wide(sudo make install should do) and try again.

wannesvc, 2018-09-19:
well... looks like the issue fixed itself. Recompiling and reinstalling did it, no idea why! Sorry to bother you guys.

closed 2018-09-19 by wannesvc

tatokis, 2018-09-19:
Make sure you recompiled and installed a release build (as that has optimizations enabled).
It's possible the bug doesn't happen on Debug builds but only Release ones, which would make it even harder to track down.