ckb-next/ckb-next #254: Mouse drag not working.

LiamGiiV, 2018-09-17:
## What is the program's current behavior?

Click and drag seems to stutter as though the mouse click is being input many times across the drag  instead of being held in the down position. 

## What would you like it to do instead?

Click and drag like it's the good old days.

## Any other notes & comments?

This issue happens if CKB-NEXT is running or not! And thanks for making this fun and cool thing at all.

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2389467/ckb-next-dev-detect-report.gz)

tatokis, 2018-09-17:
Try installing this and rebooting afterwards http://ckb-next.info/mac/ckb-next_v0.3.1-41-gb25a611_10.11-SDK.dmg

LiamGiiV, 2018-09-17:
Yeah, that seems to have worked! :0 are you some kind of science wizard? For the record I did try to update my copy of ckb-next before posting my issue but it's 100% possible that I messed that up because I am potato.

tatokis, 2018-09-18:
Don't worry about it, you didn't do anything incorrectly. It's a bug that was fixed in code but the fix isn't included in the latest release.
I'll leave this open so that people can see this until we get the new release out.

closed 2018-10-09 by ZirconiumX