ckb-next/ckb-next #256: Kubuntu 18.04 - Keyboard always freezes on return from sleep since update to 18.04

bobbruno, 2018-09-23:
# Bug report

I had Kubuntu 17.04 64 bits, and I did notice a couple of times when the keyboard would just freeze after waking up from hybernation. Waking up was hitting some key, so I guess it's not a hardware incompatibility. But already at the password screen, the custom backlight would load and freeze, no animation (before that point the keyboard was backlit the default white). I suppose this indicates that CKB managed to load the color profile into the keyboard but then somehow lost communication. Mouse,  (not Corsair) connected to the same  USB, works just fine. I did manage to open ckb-next using mouse, and it wouldn't notice that the keyboard was non-responsive - everything seemed normal. Shutting ckb-next and restarting it would not help. I didn't try restarting the daemon. Rebooting solves the problem.

This was intermittent, not enough to be a nuisance. Then I upgrade to Kubuntu 18.04, and it became a constant problem - AFAIK, every time I hybernate the PC, I have to restart it to get the keyboard working again. I tried a git pull, new make and install from commit e8d644 (August 27th), but it didn't solve the problem I also could not see any bugs or PRs that seemed related to this since then.

If I can provide any additional information let me know, I'll try to get it.

## General information

### Source
git clone from official repo and local build.

## Logs & crash reports
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2408786/ckb-next-dev-detect-report.gz)
As requested.
[ckb-next-report-1537712473.gz](https://github.com/ckb-next/ckb-next/files/2408787/ckb-next-report-1537712473.gz)
Generated by the settings window on ckb-next. Unfortunately, the  keyboard was obviously working at that point. Next time it happens, I'll try

## What is the program's current behavior?

The program is fine, but the keyboard gets unresponsive. 

## What is the expected behavior of the program?
For the keyboard to get responsive, or at least for the program to recognize the problem and log enough info to allow further debugging.

## Any other notes & comments?

bobbruno, 2018-09-24:
No. I forgot to mention that I tried it, but it didnt react. The keyboard doesn't even switch back to the default white backlighting, it just goes dark and back to the starting colors of the animated pattern. But they don't animate, and the keyboard remains unresponsive.

tatokis, 2018-10-07:
Have a look at https://github.com/ckb-next/ckb-next/issues/215

bobbruno, 2018-10-08:
Thanks, I'll give it a try. It's a bit lower level than I'm familiar with,
I'll come back here with my results as soon as I get it working.

M4SC, 2019-06-06:
Hey there, I'm coming from Manjaro and I am experiencing the same behaviour mentioned here now after every wakeup from hibernation at my current boot after switching to the 5.1.4 kernel.

[ckb-next-report-1559832509.gz](https://github.com/ckb-next/ckb-next/files/3262151/ckb-next-report-1559832509.gz)

Aditionally, what's missing in the report file, are some lines from journalctl:

> Jun 06 16:34:41 manjaro-max kernel: usb 5-4: usbfs: process 24872 (ckb1 led) did not claim interface 0 before use
> Jun 06 16:34:41 manjaro-max kernel: usb 5-4: usbfs: process 24871 (ckb1 input) did not claim interface 0 before use

After plugging both usb cables out and in again it works fine though. Maybe someone has a thought here.

tatokis, 2019-06-06:
Try the latest code master (or ckb-next-git from AUR). It should fix this issue.

hippo-o-matic, 2019-08-12:
I was having the same problem on both Kde Neon and Manjaro, the keyboard would freeze up and the only cure was to reboot. I noticed that you said 
> After plugging both usb cables out and in again it works fine though. Maybe someone has a thought here.
so I went to try that and noticed both cables were plugged into usb 3.0 ports. Apparently usb 3.0 has issues with usb devices sometimes, so I switched the keyboard cable to a 2.0, and it quit freezing after suspend. Unfortunately I don't know if this is something that can be fixed in software, so I'm just mentioning it here in case anyone else has this problem.

jms1989, 2019-10-15:
My K55 keyboard is unresponsive on resume from sleep with this software enabled and running. Works fine without it, just don't get the custom stuff. I have to unplug and replug the keyboard to get it to respond again (sometimes twice).

I found this was getting repeated a lot in my journald
```
Oct 15 11:05:03 terminator upowerd[2816]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0/0003:1B1C:1B3D.000D
Oct 15 11:05:03 terminator upowerd[2816]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0
Oct 15 11:05:03 terminator upowerd[2816]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5
```

along with stuff like this;
```Oct 15 11:05:03 terminator ckb-next-daemon[1227]: [I] Connecting Corsair Gaming K55 RGB Keyboard at /dev/input/ckb1
Oct 15 11:05:03 terminator ckb-next-daemon[1227]: [E] os_setupusb (usb_linux.c:629): Unable to read endpoint count from udev, assuming 0
Oct 15 11:05:03 terminator ckb-next-daemon[1227]: [I] Attempting reset...
Oct 15 11:05:03 terminator upowerd[2816]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0/0003:1B1C:1B3D.000D
Oct 15 11:05:03 terminator upowerd[2816]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.0
Oct 15 11:05:03 terminator upowerd[2816]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5
Oct 15 11:05:03 terminator ckb-next-daemon[1227]: [E] os_resetusb (via usb.c:613): resetusb failed: No such device
Oct 15 11:05:03 terminator ckb-next-daemon[1227]: [E] usb_tryreset (usb.c:621): Reset failed. Disconnecting.
Oct 15 11:05:03 terminator ckb-next-daemon[1227]: [I] Disconnecting /dev/input/ckb1
Oct 15 11:05:03 terminator ckb-next-daemon[1227]: [I] Removed device path /dev/input/ckb1
```

gravityfargo, 2024-05-01:
This issue has been marked as stale due to age and or lack of activity. 

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-01 by gravityfargo