ckb-next/ckb-next #257: K70 RGB not connecting under macOS 10.14 Mojave

cloone8, 2018-09-25:
# Bug report

## General information

Keyboard won't connect, although it lights up. Reconnecting doesn't work.
This happened right after the upgrade to macOS Mojave.

Debug info:

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2414248/ckb-next-dev-detect-report.gz)
[ckb-next-report-1537861755.gz](https://github.com/ckb-next/ckb-next/files/2414249/ckb-next-report-1537861755.gz)

### Source
Pre built package for macOS
Version 0.3.1

ZirconiumX, 2018-09-25:
Mojave isn't currently supported because we have no way of testing it.

tatokis, 2018-09-25:
`[F] os_inputopen (input_mac.c:64): Unable to open VirtualHIDDevice`
@cloone8 Did you allow the system extension to be loaded?

cloone8, 2018-09-25:
No it's fine, I get that Apple might randomly change or break things seeing as they're Apple. I am running a stock 2017 iMac 5k, nothing hacked about it.

tatokis, 2018-09-25:
That is very interesting. What happens if you run `sudo kextutil "/Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext"` in such a state?

cloone8, 2018-09-25:
That returns 0

tatokis, 2018-09-25:
What about `sudo kextload "/Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext"` and then unplug and replug the keyboard?

cloone8, 2018-09-25:
That returns 0 again, but the replugging doesn't seem to do anything

cloone8, 2018-09-25:
That returns the following:

`  152    0 0xffffff7f80f11000 0x5000     0x5000     org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800 (6.8.0) 19C2E27F-C5B5-334B-9F70-8C9A51814EA3 <44 5 4 3 1>`

tatokis, 2018-09-25:
So, the kext is loaded but we can't talk to it. That is very odd.
What if you do 
```
sudo kextunload "/Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext" && sudo kextload "/Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext"
```
 and then unplug and replug your keyboard.

The unplug/replug is so that ckb tries to talk to the kext again.

cloone8, 2018-09-25:
Nope, still nothing. Is there any way to help you with debugging or to make it easier?

I took some logs from the macOS console btw, filtered in "kext". This is the output starting with the commands and then me replugging everything.
[log.txt](https://github.com/ckb-next/ckb-next/files/2416708/log.txt)

tatokis, 2018-09-25:
>Nope, still nothing. Is there any way to help you with debugging or to make it easier?

Not really, sadly. I'm stuck on Sierra, so I can't test anything.
Thank you for being patient though.

I also don't see anything out of place in the logs, but I'm not too familiar with the internal workings of the VirtualHIDDevice kext.

Last thing I can think of. Can you download IORegistryExplorer (should be able to get it from apple's developer resources) and look for this entry?
[image]

It may provide more info.

cloone8, 2018-09-25:
[image]
 Here it is!

tatokis, 2018-09-25:
Hmmm. That also looks okay. Makes me wonder why the daemon can't talk to it.
I'll add a few debugging lines and send you a build to test, to maybe see if we can get more info from there.

cloone8, 2018-09-25:
That would be great! Let me know if I can do anything to help. Thanks for your time

cloone8, 2018-09-25:
Another small detail I just noticed: The keyboard works on the macOS lock screen

tatokis, 2018-09-25:
Works as in, ckb can connect to it?

cloone8, 2018-09-25:
Works as in I can type on it, but ckb doesn't detect it

cloone8, 2018-09-26:
Okay, I'll at least try to reinstall it again, see if that changes anything.

cloone8, 2018-09-26:
Okay I've done some more experimenting, it seems that the first time that CKB starts up it doesn't work, but when I kill ckb-next and the daemon from the activity monitor and then restart it, the keyboard is detected.

I would understand if this is kind of impossible to debug for you, as you have no access to a "faulty" machine.

tatokis, 2018-09-26:
Interesting. I wonder if something fails much earlier and ends up causing this instead.

tatokis, 2018-09-26:
http://ckb-next.info/mac/ckb-next-daemon-kextdbg
```
8da6c470c10431b38f8b590ccc12002db512f86c2fb265ab00d8fdcf98a082a9  ckb-next-daemon-kextdbg
```
Download that, try these commands, and reboot.
```
sudo launchctl unload /Library/LaunchDaemons/org.ckb-next.daemon.plist
sudo cp ckb-next-daemon-kextdbg /Library/Application\ Support/ckb-next-daemon
```

Generate another report from the GUI and attach it after the boot.
(ckb-next-dev-detect is the same as the Generate Report button in the GUI, so only one is enough)

cloone8, 2018-09-26:
There you go

[ckb-next-report-1538000240.gz](https://github.com/ckb-next/ckb-next/files/2421752/ckb-next-report-1538000240.gz)

joemulray, 2018-09-28:
> @tatokis what do we need to do to get it working on mojave i have K65 and M65 neither of them functioning

[ckb-next-report-1538138574.gz](https://github.com/ckb-next/ckb-next/files/2428197/ckb-next-report-1538138574.gz)

tatokis, 2018-09-28:
@joemulray Install ckb-next 0.3.1.

tatokis, 2018-10-04:
@cloone8 I had a thought. It's possible the kext takes longer to initialise.
In the upcoming days, I will send you a build that will attempt to wait until the kext is available, and hope that works.
Apologies for the inconvenience

cloone8, 2018-10-05:
That would be great, thanks! No problem at all, I understand that this is probably not your job and you don't get paid for it at all

tatokis, 2018-10-06:
This build should try every half a second for 10 seconds to talk to VirtualHIDDevice.

http://ckb-next.info/mac/ckb-next-daemon-kextwait
```
86d240155afdec1a6c6d6211c9e9aa2250ae8bdfcbb0ea48b1a7456fb2f17daa  ckb-next-daemon-kextwait
```
Same procedure as earlier. Upload another report afterwards too.

cloone8, 2018-10-07:
After doing multiple reboots to test, this works on 100% of them. Totally seems to have fixed the issue! 

Thanks man, you've helped tremendously

[ckb-next-report-1538898356.gz](https://github.com/ckb-next/ckb-next/files/2453696/ckb-next-report-1538898356.gz)

tatokis, 2018-10-07:
Glad to hear it. I just merged the code to fix it.

closed 2018-10-07 by tatokis

tatokis, 2018-10-09:
We just released 0.3.2 with the fix included. You can update to it so that you run an optimised/non-debug version of the software.
https://github.com/ckb-next/ckb-next/releases/download/v0.3.2/ckb-next_v0.3.2_1011.dmg

cloone8, 2018-10-09:
Thanks, you're the best

commit 2447df07 (2018-10-07): Wait for the kext to be ready after loading it

Closes #257