ckb-next/ckb-next #259: Binding applications to volume keys stops keyup messages

9ParsonsB, 2018-09-28:
# Devices information

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2428134/ckb-next-dev-detect-report.gz)
---

# Bug report

Whenever the GUI binds an application to the volume key (volume wheel in my case), the daemon stops sending keyup messages

## General information

I am trying to run mpc volume +/- on rolling the volume wheel

### Source

  * ckb-next-git package in AUR

## What is the program's current behavior?
The command runs once and then continuing to roll the wheel does nothing.

## What is the expected behavior of the program?
Run the command on each roll of the volume wheel

## Any other notes & comments?
It runs the command again after pressing any other key, regardless of time.

9ParsonsB, 2018-10-05:
Seeming no change.

9ParsonsB, 2018-10-06:
ckb-next: Corsair RGB driver 0.3.1-63-ga2fc54c

tatokis, 2018-10-06:
Yeah, you're not running the version from the branch.
It should say `ckb-next: Corsair RGB driver 0.3.1-64-g2c39451`

commit 5416a684 (2018-10-07): Merge branch 'wheel-unbind'

Closes #259
closed 2018-10-07 by tatokis