ckb-next/ckb-next #261: STRAFE - unable to change colors

SonnySelhorst, 2018-09-30:
Hello,

I have just bought the STRAFE corsair keyboard. Tried installing a few different versions of cbk (including the newest) and nothing seems to be working for me. 

THE ISSUE: can't change colors, it doesn't show in the app, nor does it show on the keyboard (obviously) - the keys in the app remain a different tint of grey, regardless of color selected. The different tints ARE shown on the keyboard and translate into different tints of red (the standard color).

Any help would be greatly appreciated! 

Thanks for having made this :) I hope I can get it to work. 

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2431553/ckb-next-dev-detect-report.gz)

Cheers,
Sonny[](url)

tatokis, 2018-09-30:
I'm not sure what the problem is.
> The different tints ARE shown on the keyboard and translate into different tints of red (the standard color).

You bought a monochrome strafe. It can't do any colour other than shades of red.

SonnySelhorst, 2018-09-30:
Hey Tatokis,

Quick reply: fml... 

I was convinced it could change colors, but you're right, it can't. 

Have a great life my man, thanks for the quick reply. 

Cheers,
Sonny

closed 2018-10-03 by ZirconiumX