ckb-next/ckb-next #262: Respecting the "Adjusting Key Repeat" and "Delay Until Repeat" Values

AaronMartone, 2018-10-01:
I upgraded to Mojave 10.14 and thanks to 0.3.1, the keyboard is seen and working again. Awesome turn around time, too, despite not having that test environment!

My only issue now is that the "Key Repeat" and "Delay Until Repeat" features of the Keyboard section of System Preferences don't seem to be honored; and I see no such place in CKB to adjust these.

I set Key Repeat to: Fast and Delay Until Repeat to: Short. If I hold down a key, it takes about 1/4th of a second before it starts repeat, and it repeats at a rate of about 3 characters per second. Expected behavior would be like 1/20th of a second before repeating and like 20 characters per second (estimate).

I will search online to see if there are any apple preferences I can write to and report what findings I have.

I have attached the device information per your request.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2434189/ckb-next-dev-detect-report.gz)

tatokis, 2018-10-01:
I just tested this on a clean Mojave installation, and I can't reproduce the issue.

I also don't see why it would happen, as we are no longer managing key repeating ourselves, but only just tell the OS that the key is held down. Does it work with other keyboards, or if the `ckb-next-daemon` is not running?

Long shot, but maybe try `sudo rm /Library/Application\ Support/ckb-daemon` and reboot.

AaronMartone, 2018-10-01:
When I attempted to run your commend, it informed me there was no such directory.

I then ran the uninstallation commands and (as noted) some errors indicated:

`No matching processes belonging to you were found
No matching processes belonging to you were found
/Library/LaunchDaemons/com.ckb.daemon.plist: No such file or directory
/Library/LaucnhDaemons/org.next.ckb.daemon.plist: No such file or directory`

I went to a friend who was using OSX 10.13 (not 10.14, like I am) and 0.2.4 of your software, and we noticed that it inherited the changes he had from the OS. I also had him run some commands like `sudo defaults read -g` and we noted it was clean, reporting no matching domains. So I ran `sudo defaults find ApplePressAndHoldEnabled` (again 2 more times for `KeyRepeat` and `InitialKeyRepeat`), then ran `sudo defaults delete` to remove all those entries in all domains they were found. Afterwards I rebooted, noted that, despite what slider I set it to, the input rate was still horribly slow (via the Macbook Pro keyboard). I plugged in an Apple keyboard, and it was still the same slow speed.

I reinstalled 0.3.1, rebooted, and the sliders all show at max performance (fastest options), but despite that, the repeat is still slow.

AaronMartone, 2018-10-01:
I may have made some progress. It seems there's this thing called "Slow Keys" 

`System Preferences > Accessibility > Keyboard`

And the moment I turned that off (and restarted), I got SIGNIFICANTLY better repeat (starts faster, more characters per second). Now, oddly, if I change values for the `NSGlobalDomain`, I see them get updated, but even after restarts, it seems changing either value doesn't adjust what I see. The sliders in the `Keyboard` section of `System Preferences` show maxed out.

ZirconiumX, 2018-10-01:
This is not directly related, but is worth mentioning:

>0.2.4 of your software

Please **do not** bother with 0.2.4; not only is it *not* our software, but it's *ancient*, not even supporting 2.x firmwares, let alone 3.x firmwares.

I'm pretty sure it does not even support 10.13, given the [issues](https://github.com/ckb-next/ckb-next/issues?q=is%3Aissue+sierra) we have had with Sierra, which was 10.12.

AaronMartone, 2018-10-01:
I appreciate you saying so, but that was the version of software **my friend** was on, I was using it for as best a comparison as I could. I was on 0.3.0 before I noted 10.14 was incompatible, updated to 0.3.1, restored functionality, but lost respect for OS-level key repeat settings.

closed 2018-10-03 by AaronMartone

tatokis, 2018-10-03:
Did you get this fixed?