ckb-next/ckb-next #264: Problem with daemon and passing USB device to VM

GloriousEggroll, 2018-10-03:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?

I've been using ckb-next for the past few months, and every time I go to pass through my keyboard and mouse to a virtual machine, I have to stop ckb-next before starting the VM, otherwise it tries to reclaim the devices and they never get passed to the VM.

For example my mouse will disconnect for a moment as it tries to pass to the vm, then it reconnects to the host instead.

A good example may be to have a look at libratbag - as it also runs as a daemon, but has no issues with passing logitech devices to a vm and reclaiming them after the vm closes

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

### Source

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

ZirconiumX, 2018-10-03:
If you need to use a VM, then stopping and starting the daemon is the
easiest way, although I question why you're passing through the USB devices
themselves and not using a virtual device provided by VirtualBox or QEMU.

Here's the thing: ratbag has an entirely different driver model compared to
us.

For Logitech devices, the Linux kernel talks to the device through the
hid-logitech-hidpp module, and so when you want to pass through the device,
hid-logitech-hidpp detaches itself from the device and hands it over to the
VM, pretty colours intact. This works because Logitech's protocol is
persistent, and so ratbag only has to talk to the device when it needs to
change the colour.

On Corsair devices, the situation is not so convenient; we have to take
control of the device ourselves because the protocol is stateless, and it
has a proprietary HID format for things like keypresses that would confuse
the kernel. An approach like that is awkward for situations like yours, but
for (buggy)  devices like the Glaive, it's the only way to make them work.

GloriousEggroll, 2018-10-03:
Ah I see, this is understandable. I pass the device through because I run a gpu passthrough setup for native performance and virtual device latency is not decent enough for gaming. I am unable to pass the entire usb controller through because I have a large amount of USB devices connected to the controller (including already using a hub and a pci card). 

I was not aware that logitech used its own kernel module, which indeed does make things easier on that front.  I'll attempt to pass the entire controller through as a test to see if that helps with passing it through without starting/stopping the service.

thanks for the input

closed 2018-10-03 by GloriousEggroll