ckb-next/ckb-next #265: Problem compile ckb-next

nero08, 2018-10-05:
Hello, I have a problem to compile CKB-NEXT, I have this error:

-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- ckb-next version: 0.3.1-63-ga2fc54c (Non-release)
-- Performing Test C_COMPILER_SUPPORTS_-Og
-- Performing Test C_COMPILER_SUPPORTS_-Og - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Og
-- Performing Test CXX_COMPILER_SUPPORTS_-Og - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST - Failed
-- Found Iconv: /usr/lib/x86_64-linux-gnu/libc.so
-- Checking for module 'libudev'
--   Found libudev, version 237
-- Found UDEV: /lib/x86_64-linux-gnu/libudev.so  
-- Searching for running ckb and/or ckb-next GUI
-- NOTE:
    Privileged access is required for operations upon the daemon at
    configure time as it is owned by root. If this is unacceptable,
    disable SAFE_INSTALL and prepare the filesystem manually.
-- upstart detected
-- Generating and importing ckb-next-daemon.conf (upstart)
Unit ckb-daemon.service could not be found.
initctl: Unable to connect to Upstart: Failed to connect to socket /com/ubuntu/upstart: Connection refused
Unit ckb-next-daemon.service could not be found.
initctl: Unable to connect to Upstart: Failed to connect to socket /com/ubuntu/upstart: Connection refused
-- systemd detected
-- Generating and importing ckb-next-daemon.service (systemd)
inactive
Failed to get unit file state for ckb-daemon.service: No such file or directory
inactive
Failed to get unit file state for ckb-next-daemon.service: No such file or directory
-- Found QuaZip: /usr/lib/x86_64-linux-gnu/libquazip5.so  
-- Found PulseAudioSimple: /usr/lib/x86_64-linux-gnu/libpulse-simple.so  
-- Found PulseAudio: /usr/lib/x86_64-linux-gnu/libpulse.so  
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Generating binary.qrc
[... 99 lines cut ...]
AutoUic subprocess error
------------------------
uic failed for
  "/home/riname/Téléchargements/ckb-next/src/gui/animadddialog.ui"
included by
  "/home/riname/Téléchargements/ckb-next/src/gui/animadddialog.cpp"

Command
-------
/usr/lib/qt5/bin/uic -o /home/riname/Téléchargements/ckb-next/build/src/gui/ckb-next_autogen/include/ui_animadddialog.h /home/riname/Téléchargements/ckb-next/src/gui/animadddialog.ui

Output
------
Could not create output file

src/gui/CMakeFiles/ckb-next_autogen.dir/build.make:78: recipe for target 'src/gui/CMakeFiles/ckb-next_autogen' failed
make[2]: *** [src/gui/CMakeFiles/ckb-next_autogen] Error 1
CMakeFiles/Makefile2:803: recipe for target 'src/gui/CMakeFiles/ckb-next_autogen.dir/all' failed
make[1]: *** [src/gui/CMakeFiles/ckb-next_autogen.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
An error occurred,
press enter to exit.

I installed all required lib and my Qt version is 5.9.5

nero08, 2018-10-05:
Solved: just move to home folder

light2yellow, 2018-10-09:
This is an interesting issue that should never ever happen.

After poking Arch Linux for a few hours, I finally managed to reproduce this issue on openSUSE Tumbleweed with:
```
% locale
LANG=ru_RU.UTF-8
LC_CTYPE="ru_RU.UTF-8"
LC_NUMERIC="ru_RU.UTF-8"
LC_TIME="ru_RU.UTF-8"
LC_COLLATE="ru_RU.UTF-8"
LC_MONETARY="ru_RU.UTF-8"
LC_MESSAGES="ru_RU.UTF-8"
LC_PAPER="ru_RU.UTF-8"
LC_NAME="ru_RU.UTF-8"
LC_ADDRESS="ru_RU.UTF-8"
LC_TELEPHONE="ru_RU.UTF-8"
LC_MEASUREMENT="ru_RU.UTF-8"
LC_IDENTIFICATION="ru_RU.UTF-8"
LC_ALL=

% LC_ALL=C locale
LANG=ru_RU.UTF-8
LC_CTYPE="C"
LC_NUMERIC="C"
LC_TIME="C"
LC_COLLATE="C"
LC_MONETARY="C"
LC_MESSAGES="C"
LC_PAPER="C"
LC_NAME="C"
LC_ADDRESS="C"
LC_TELEPHONE="C"
LC_MEASUREMENT="C"
LC_IDENTIFICATION="C"
LC_ALL=C
```

by building ckb-next in a directory whose absolute path contains "Новая папка" ('New folder'), and getting same `uic` errors.

The differences in Arch and SUSE locale handling are to be determined, if at all, since I tried to reproduce on Arch by changing the locale dynamically (i.e. generating locale, unsetting `LANG`, sourcing `/etc/profile.d/locale.sh` and starting a new shell instance) without complete relogin. Maybe doing so triggers this magic bug.

I took a glance at `uic` source code, read Qt5 documentation about the QFile and QString functions it uses for encoding and decoding, and just a little bit `icu` source code. So far no success. In `uic` the `open()` function is the one that fails, but it can be due to setting an incorrect filename previously and not failing early.

commit 44d253f4 (2018-12-22): quickinstall: handle multibyte characters in path (#265)

Problem: Qt5 UIC cannot create a file if its absolute path contains
multibyte characters and LC_CTYPE is set to "C".

Solution: set all (but LC_CTYPE and LC_COLLATE) POSIX locale environment
variables to "C". LC_CTYPE and LC_COLLATE should be set to the original
locale via LANG. Avoid LC_ALL.

Closes: #265
closed 2018-12-22 by tatokis