ckb-next/ckb-next #266: driver is not taking control of my scimitar pro rgb

davidbdeath, 2018-10-07:
The driver is not taking control of my scimitar pro RGB, but it is disabling my onboard profiles. I would just use the onboard profiles but a while back Cue broke them and they have not fixed that issue.

ZirconiumX, 2018-10-07:
The on-board profiles are disabled because Corsair changed the format, and I haven't had the motivation to fully reverse the format. So until that time, rather than risk corrupting the mouse, we just disabled the feature.

However, you'll need to more clearly define "not taking control". Could you make a report for me?

davidbdeath, 2018-10-07:
Wow, thank you so much for getting back with me so quickly.
I sometimes struggle to describe things, if I'm not making sense I'm sorry.

I am unable to configure keybindings, lightning and DPI settings in the CKB software. Enough control is present that my onboard presets are disabled, and I'm unable to click and drag most UI elements. My onboard profiles don't work correctly and are not needed if the CKB software could replace them. My SIPS is disabled if that makes a difference.

[ckb-next-report-1538940631.gz](https://github.com/ckb-next/ckb-next/files/2454251/ckb-next-report-1538940631.gz)

ZirconiumX, 2018-10-07:
I just spent the past five minutes rewriting a reply and then realised you have the 0.3.1 Scimitar bug.

[Try this version that fixes the bug.](http://ckb-next.info/mac/ckb-next_v0.3.1-41-gb25a611_10.11-SDK.dmg)

DarkLogicX, 2018-10-07:
I just updated to 0.3.1 on CentOS 7 via Yum Update, and now I'm unable to use my Scimitar mouse at all.

ZirconiumX, 2018-10-07:
Build latest master and you'll also get the fix.

davidbdeath, 2018-10-07:
Is the bug a hardware issue with my mouse? One last questions bindings that I regularly use like "command+shift+[" I am unable to create.

ZirconiumX, 2018-10-07:
>Is the bug a hardware issue with my mouse?

It's a firmware issue.

Truthfully it's because Corsair have very inconsistent firmware naming that we plan to fix at some point.

Imagine that we talk to the mouse through sending messages through mailboxes. For the original series of K70 RGB amongst others, they had five letterboxes, where messages would go into mailbox no.1 and different messages would come out of 2-5 depending on their type. The firmware described this as P01, and labelled it as the 1.xx series.

Later, they switched to a different number of mailboxes, and changed which mail goes in and out of which mailbox, described it as P04, and labelled it as the 2.xx series.

The Scimitar Pro labels itself as 1.xx (you are running firmware v1.10), but is actually P04, AKA 2.xx. However, the daemon would talk to the Scimitar Pro as if it was P01, resulting in no mail getting through either side.

That's what the latest (as of 6d921ab3) Mac and Linux builds fix.

ZirconiumX, 2018-10-07:
> ./quickinstall: line 28: cmake: command not found

At the risk of sounding obvious, it's because you don't have CMake installed. You can find that through `yum` too.

DarkLogicX, 2018-10-07:
Well any other packages that are required to run the manual install?
Also any way of knowing if/when the repo will get the fixed version?

DarkLogicX, 2018-10-07:
Well now it seems the Cmake that YUM provides is not a new enough one for the script.

DarkLogicX, 2018-10-07:
At least my K70 keyboard is working on this version still.
But seems the quick install script is a bit to current in it's dependencies for CentOS7.

ZirconiumX, 2018-10-07:
`cmake --version`?

ZirconiumX, 2018-10-07:
The version of CMake we require is two years old. It's not like we're asking for cutting edge tooling.

DarkLogicX, 2018-10-07:
The error from trying to run quickinstall the last time gave the versioon info.

CMake Error at CMakeLists.txt:30 (cmake_minimum_required):
  CMake 3.5.1 or higher is required.  You are running version 2.8.12.2

That's with freshly installed cmake via (yum install cmake)

ZirconiumX, 2018-10-07:
Try [this guide](https://stackoverflow.com/questions/48831131/cmake-on-linux-centos-7-how-to-force-the-system-to-use-cmake3).

DarkLogicX, 2018-10-07:
Might be that Fedora is more current in it's repo for cmake, but with CentOS being based on RHEL it's a bit longer term before those trickle down.

DarkLogicX, 2018-10-07:
So I removed cmake, installed cmake3 then ran.

alternatives --install /usr/local/bin/cmake cmake /usr/bin/cmake3 20 \
--slave /usr/local/bin/ctest ctest /usr/bin/ctest3 \
--slave /usr/local/bin/cpack cpack /usr/bin/cpack3 \
--slave /usr/local/bin/ccmake ccmake /usr/bin/ccmake3 \
--family cmake

from a root prompt. (IE after sudo su - ing into root)
And it got further, but still failed with the following as the last output.

[100%] Linking CXX executable ../../bin/ckb-next
[100%] Built target ckb-next
sudo: cmake: command not found
An error occurred,
press enter to exit.

ZirconiumX, 2018-10-07:
You aren't running `quickinstall` as root, are you?

As an aside, `sudo -i` is better and shorter than `sudo su -`.

DarkLogicX, 2018-10-07:
Yes I was running it as root.

ZirconiumX, 2018-10-07:
Don't run it as root; it'll prompt you for `sudo` when it needs to, because building as root is unnecessary and inconvenient.

DarkLogicX, 2018-10-07:
Anything I need to do to clean up what's done so far?  Or just move the CKB-next dir to my normal user home and go again?

DarkLogicX, 2018-10-07:
I used "rm -rf ckb-next" to delete the clone that was in root's home.
then exited our to my normal user and repeated the steps.

It failed at the same point.

tatokis, 2018-10-09:
We just released 0.3.2 which contains a fix. It should be packaged soon.

DarkLogicX, 2018-10-11:
Thank you, I'll be looking forward to it.
I've been using some old HP mouse in the mean time.

tatokis, 2018-10-13:
@hevanaa Can you please update the build on your repo?

hevanaa, 2018-10-13:
I've updated to latest release in the Fedora copr repo. Please notify me of any problem. I don't have access to the Corsair keyboard any more, so I can't check much more than that the installation works.

hevanaa, 2018-10-13:
As mentioned above, there are some differences between Fedora and CentOS/RHEL. You can see the workarounds in the rpm build [here](https://github.com/hevanaa/ckb-spec/blob/master/SPECS/ckb-next.spec) (the %if 0%{?rhel} parts and the stupid desktop icon thing).

closed 2018-11-02 by tatokis