ckb-next/ckb-next #270: AIO Compatibility

jbonnett92, 2018-10-08:
Hi All,
Could we add compatibility with AIO's? https://www.corsair.com/us/en/Categories/Products/Liquid-Cooling/c/Cor_Products_Cooling

Thanks,
Jamie

KamranMackey, 2018-10-08:
I don't think supporting AIOs is within the realm of what `ckb-next` wants to do. It's named `ckb` for a reason, as so far its only goal is to support keyboards (and mice) from what I can tell. But, let's just see what @tatokis has to say about this, but I think it may be a no.

ZirconiumX, 2018-10-09:
As I understand it, Tasos thinks we should try to keep our options open,
but I personally think it's out of scope for ckb-next, especially because
https://github.com/audiohacked/OpenCorsairLink/ exists and is fairly
mature. We could consider using OCL as a backend for the GUI if there's
suitable demand for it.

davidbdeath, 2018-12-22:
Can I ask how do the corsair AIO that are software not motherboard controlled work in Mac OS? Can I make a hardware profile in Windows that works in Mac OS without iCUE?

tatokis, 2018-12-26:
@davidbdeath I don't think anyone here knows, but my guess is that it only uses its defaults without any software.

gravityfargo, 2023-04-19:
There are breakout requests for individual devices now. Seeing as this is a generalized request, I'll close this.

closed 2023-04-19 by gravityfargo

jbonnett92, 2023-04-19:
This was 5 years ago lol

gravityfargo, 2023-04-19:
I don't want to close old issues just because they're old. Some are still valid bugs or requests. This one was an obvious close :smiley: