ckb-next/ckb-next #274: "Screen Brightness Down/Up" for LG UltraFine 4K

xuguangyuansh, 2018-10-11:
# Problem
G1, G2 binding "Screen Brightness Down/Up", but the LG UltraFine 4K is invalid.

tatokis, 2018-10-11:
The brightness up/down bindings are the same as if you pressed the actual brightness up/down keys on your mac keyboard. Do they work if you press them on your mac keyboard?

Do you have a brightness slider on “System Preferences” -> “Displays” for your LG monitor?
If neither, then there's not much we can do about it. You will need to use third party software to dim it.

xuguangyuansh, 2018-10-12:
After the test，Press brightness up/down keys of apple keyboard is ok, but when press brightness up/down bindings, it always adjust the macbook pro 15 's screen brightness, rather than LG Ultrafine 4 k.

Yes I have a brightness slider on “System Preferences” -> “Displays” for my LG monitor, and work well. But that's not very convenient, and I need to adjust the brightness frequently.

tatokis, 2018-10-12:
I don't think there's a way to explicitly do that. Which version of ckb-next are you running?

tatokis, 2018-10-13:
Also, try binding a G key to "Fn" and then press G+f2 and see if that changes anything

tatokis, 2018-10-24:
I meant for example to bind the G1 key to Fn and then use G1 + F2 to see if that changes the brightness.

tatokis, 2018-11-02:
Just a quick update. I think I found the issue and I'm working on implementing it. Turns out there are multiple keycodes to change the brightness.

xuguangyuansh, 2018-11-07:
> I meant for example to bind the G1 key to Fn and then use G1 + F2 to see if that changes the brightness.

First of all, thank you for your help. I am currently using ckb-next v0.3.2, but i could't find the 'Fn' key from  Binding -> keyboard. I've also tried Macro,  which doesn't tie the macbook keyboard's Fn to Corsair's G*.

xuguangyuansh, 2018-11-07:
> Just a quick update. I think I found the issue and I'm working on implementing a fix. Turns out there are multiple keycodes to change the brightness.

Great! I'm really looking forward to it!!

xuguangyuansh, 2018-11-08:
> http://ckb-next.info/mac/ckb-next_v0.3.2-12-g0a9ca07_avtopcase.dmg
> Try this one, and use the Screen Brightness Up/Down bindings.
> 
> ```
> 1eaa979fe5e81f77560c8fc03657931db282da398b72c5abd248366bd8c535d7  ckb-next_v0.3.2-12-g0a9ca07_avtopcase.dmg
> ```

I have tried the "ckb-next_v0.3.2-12-g0a9ca07_avtopcase.dmg", and use the Screen Brightness Up/Down bindings, but still can only adjust the macbook pro Screen.

closed 2022-03-05 by tatokis