ckb-next/ckb-next #275: keypresses not recognized during system boot (alt)

mattbehnken, 2018-10-13:
--

# Something else

ZirconiumX, 2018-10-13:
I have a hunch about this. Corsair use a nonstandard USB endpoint configuration for their keyboards, and perhaps the Apple EFI doesn't recognise it because normal Apple keyboards use the standard endpoint configuration.

So let's try using BIOS mode on your keyboard. When you boot up, hold winlock and F1 for about three seconds and release, which should make your scroll lock light blink if you did it correctly. Then try holding alt.

Hopefully you still have time left in the EFI to press alt.

tatokis, 2018-11-02:
Closing due to lack of reply from OP

closed 2018-11-02 by tatokis

mattbehnken, 2018-11-04:
Thank you, I missed the reply email from 22 days ago. I'll try that today, very  much appreciate any tips. Currently I have another keyboard tucked away but still plugged via usb for switching back to mac os (alt during boot), will circle back.e